The-OpenROAD-Project/OpenROAD #4821: Fix problem or remove unactionable warning

oharboe, 2024-03-20:
### Describe the bug

Fix the cause of the following errors warnings or remove them if they are bogus or inactionable:

- [ ] `Error: duplicated via definition for via1_2_70092_18_1_1947_36_36`
- [ ] `[WARNING DRT-0240] CUT layer V3 does not have square single-cut via, cut layer width may be set incorrectly. `

To reproduce:

The file is big, but running the test until you get the warnings only takes a minute:

untar https://drive.google.com/file/d/1nHLO6_kBF4hhUBYYEDDPbirIYIGbMu7b/view?usp=sharing

```
$ ./run-me-BoomTile-asap7-base.sh
OpenROAD v2.0-12661-g64b9d673c
[deleted]
[INFO DRT-0150] Reading design.
Error: duplicated via definition for via1_2_70092_18_1_1947_36_36
Error: duplicated via definition for via5_6_120_288_1_2_58_322
[deleted]
[WARNING DRT-0240] CUT layer V3 does not have square single-cut via, cut layer width may be set incorrectly.
[WARNING DRT-0240] CUT layer V5 does not have square single-cut via, cut layer width may be set incorrectly.
[deleted]
```

### Expected Behavior

Only actionable real warnings

### Environment

```shell
OpenROAD v2.0-12661-g64b9d673c
```

### To Reproduce

See above

osamahammad21, 2024-03-25:
>  Error: duplicated via definition for via1_2_70092_18_1_1947_36_36

This via appears in both the tech lef and the def. I don't see it in the original asap7 tech lef but it appears when I write lef from 5_2_fillcell.odb. An earlier step in the flow did something wrong by adding it to the tech. I am thinking of making this an actual error. However it seems more fit to catch it in odb not drt so that it could be handled as early as possible.

> [WARNING DRT-0240] CUT layer V3 does not have square single-cut via, cut layer width may be set incorrectly.

This should be handled in the specified commit in this issue. I am removing this warning as well as the dependency on the cut layer width.

maliberty, 2024-03-28:
@oharboe can you find which step produces that via in the tech lef and make a test case for it.

osamahammad21, 2024-03-28:
@maliberty @oharboe It happens at PDN in hierarchical designs. PDN creates its via definitions for the sub-modules (block vias). Then when the block is hardened into a macro, the pdn via moves to the tech of the top module. PDN then creates the same via definition for the top module.
Solution would be to check if the same via definition exists in the tech at pdn via creation step. (this would require some changes in PDN structure since it assumes it uses dbVia not dbTechVia)
Another solution would be to append the block name to the via definition. That way both vias would exist but they would be redundant.
Which seems more reasonable?

maliberty, 2024-03-28:
Can we just check if the vias are identical?  In that case it doesn't matter and we can just keep one.

I expect we are mixing up things a bit.  The tech LEF comes from the PDK.  A LEF abstract is generated by OR.  They both go into odb but are distinct data sources.

rovinski, 2024-03-28:
I can imagine a process where the same via is generated in two different blocks which is not in the tech LEF. These are done in parallel, so it is reasonably necessary that each one has to have it's own via definition. But when they are assembled at the top level, they appear as duplicate vias.

A foolproof solution would be to, like @maliberty said, append a hash like the block name. Another method might be to check if the definitions are the same and then don't warn if so.

osamahammad21, 2024-03-31:
> Can we just check if the vias are identical? In that case it doesn't matter and we can just keep one.

@maliberty  Keep one where? at ODB or DRT?

maliberty, 2024-04-01:
One in drt since it is what is complaining - or just don't warn if duplicates don't affect drt.