The-OpenROAD-Project/OpenROAD #4828: Speed up reporting in CTS, SKIP_REPORT_METRICS=1 takes cts from 4 hours to 10 minutes

oharboe, 2024-03-21:
### Description

untar https://drive.google.com/file/d/1i1t-JcIebV9CiMOm55G2RezttqwH74DD/view?usp=sharing

```
$ SKIP_REPORT_METRICS=1 time ./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-12680-gc2b1d565b 
[deleted]
```

### Suggested Solution

Make reporting fast enough that it isn't worth disabling. This would make it easier to learn OpenROAD, less fiddly bits that one has to learn about.

maliberty, 2024-03-21:
This test case has macros but no .lib (or .lef outside of the .odb) are included.  Is this intentional?

oharboe, 2024-03-22:
> This test case has macros but no .lib (or .lef outside of the .odb) are included. Is this intentional?

No, sorry about that. Will try to fix it.

oharboe, 2024-03-22:
> This test case has macros but no .lib (or .lef outside of the .odb) are included. Is this intentional?

Fixed, uploaded a new version, same link as above.

Added a PR for review in bazel-orfs w.r.t. the root cause of why .lib files were not included https://github.com/The-OpenROAD-Project/bazel-orfs/pull/18

oharboe, 2024-03-27:
@maliberty fixed by #4852?

oharboe, 2024-03-27:
@maliberty I think #4852 would have caught this and stopped in resizing, before CTS.

Look at the pathological slack due to macros being created at floorplan instead of global route.

So close? This PR is moot?

[image]

closed 2024-03-27 by maliberty