The-OpenROAD-Project/OpenROAD #4830: ifp: initialize_floorplan should accept lower bound for relative sizing

proppy, 2024-03-21:
### Description

When using floorplan `relative` floorplan
```
initialize_floorplan -utilization
```
for "very small" design pdngen or placement can fails with the following errors:
```
[ERROR PDN-0185] Insufficient width (16.128 um) to add straps on layer K3 in grid "top" with total strap width 13.108 um and offset 6.4 um.
```
```
[ERROR GPL-0302] Use a higher -density or re-floorplan with a larger core area.
Given target density: 0.80
Suggested target density: 0.98
Error: place.tcl, 455 GPL-0302
```

Looking at the implementation:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/c2b1d565be57e0d41cb488d78ed5e4977e0c4235/src/ifp/src/InitFloorplan.cc#L119-L123 it doesn't appear that there is a way to specify a lower bound for automated sizing.

### Suggested Solution

It would be nice if `initialize_floorplan` accepted two additional parameter for specifying lower bound of core area width and height.

maliberty, 2024-03-21:
How would a user know what to set those args to?

proppy, 2024-03-21:
> How would a user know what to set those args to?

I thought those could be derived from some multiple of the average standard cell size for a given PDK?

maliberty, 2024-05-21:
No updates - please reopen if there is something actionable.

closed 2024-05-21 by maliberty