The-OpenROAD-Project/OpenROAD #4832: Add timing report columns for routing and buffers

oharboe, 2024-03-21:
### Description

In architectural exploration, I'm interesting in separating the concerns of quality of results in placement&routing and logic.

Separating the concerns speeds up the architectural exploration: we can immediately know how much better the logic has to become even with perfect CTS, place&route.

When creating histograms and timing reports, it would be nice to get the top paths for logic only.

[image]

```
>>> report_checks -from {core/rob/io_rob_head_idx\[2\]$_SDFFE_PP0P_/QN} -to {lsu/stq_31_bits_uop_lrs1\[4\]$_SDFFE_PP0P_/D}
Startpoint: core/rob/io_rob_head_idx[2]$_SDFFE_PP0P_
            (rising edge-triggered flip-flop clocked by clock)
Endpoint: lsu/stq_31_bits_uop_lrs1[4]$_SDFFE_PP0P_
          (rising edge-triggered flip-flop clocked by clock)
Path Group: clock
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock clock (rise edge)
1374.85 1374.85   clock network delay (propagated)
   0.00 1374.85 ^ core/rob/io_rob_head_idx[2]$_SDFFE_PP0P_/CLK (DFFHQNx2_ASAP7_75t_R)
  92.59 1467.44 v core/rob/io_rob_head_idx[2]$_SDFFE_PP0P_/QN (DFFHQNx2_ASAP7_75t_R)
  80.02 1547.46 ^ core/rob/_100770_/CON (HAxp5_ASAP7_75t_R)
  26.77 1574.24 ^ load_slew124092/Y (BUFx12f_ASAP7_75t_R)
  37.92 1612.16 ^ max_cap124090/Y (BUFx10_ASAP7_75t_R)
  49.11 1661.27 v core/rob/_055791_/Y (NOR2x2_ASAP7_75t_R)
  97.41 1758.68 ^ core/rob/_055972_/Y (NAND2x2_ASAP7_75t_R)
  33.88 1792.56 ^ max_length108163/Y (BUFx16f_ASAP7_75t_R)
  63.31 1855.88 ^ max_length108162/Y (BUFx16f_ASAP7_75t_R)
  25.21 1881.09 v core/rob/_055973_/Y (OAI22x1_ASAP7_75t_R)
  46.67 1927.76 v core/rob/_055974_/Y (OR4x2_ASAP7_75t_R)
  76.43 2004.19 v core/rob/_055977_/Y (OR5x2_ASAP7_75t_R)
  71.68 2075.87 v load_slew88761/Y (BUFx16f_ASAP7_75t_R)
  34.67 2110.54 ^ core/rob/_055979_/Y (NOR2x2_ASAP7_75t_R)
  47.21 2157.75 ^ core/rob/_055980_/Y (AND2x6_ASAP7_75t_R)
  64.33 2222.07 ^ core/rob/_056042_/Y (AND3x4_ASAP7_75t_R)
  49.00 2271.08 v core/_08190_/Y (NAND2x2_ASAP7_75t_R)
  66.16 2337.24 v wire80404/Y (BUFx16f_ASAP7_75t_R)
  70.70 2407.94 v wire80403/Y (BUFx16f_ASAP7_75t_R)
  62.76 2470.70 v core/_08191_/Y (OA21x2_ASAP7_75t_R)
  47.13 2517.83 v core/_08194_/Y (OR3x4_ASAP7_75t_R)
  51.30 2569.13 ^ core/_08195_/Y (NOR2x2_ASAP7_75t_R)
  55.07 2624.20 ^ core/_11301_/Y (AND2x6_ASAP7_75t_R)
  28.96 2653.16 ^ wire68158/Y (BUFx16f_ASAP7_75t_R)
  94.62 2747.78 ^ wire68156/Y (BUFx16f_ASAP7_75t_R)
  85.95 2833.73 ^ max_length68155/Y (BUFx12_ASAP7_75t_R)
  48.23 2881.96 ^ wire68154/Y (BUFx16f_ASAP7_75t_R)
  90.44 2972.40 ^ lsu/_368744_/Y (AND3x4_ASAP7_75t_R)
  86.57 3058.97 ^ lsu/_368769_/Y (XNOR2x2_ASAP7_75t_R)
  33.62 3092.59 ^ wire64163/Y (BUFx16f_ASAP7_75t_R)
  83.48 3176.08 v lsu/_703495_/CON (HAxp5_ASAP7_75t_R)
  38.08 3214.15 ^ lsu/_645235_/Y (INVx1_ASAP7_75t_R)
  18.69 3232.85 v lsu/_703493_/CON (HAxp5_ASAP7_75t_R)
  16.04 3248.89 ^ lsu/_645236_/Y (INVx1_ASAP7_75t_R)
  21.41 3270.30 v lsu/_703491_/CON (HAxp5_ASAP7_75t_R)
  27.33 3297.62 ^ lsu/_703491_/SN (HAxp5_ASAP7_75t_R)
  31.28 3328.91 ^ lsu/_660138_/Y (XNOR2x1_ASAP7_75t_R)
  55.44 3384.35 ^ lsu/_660140_/Y (AND5x2_ASAP7_75t_R)
  42.10 3426.45 ^ wire46466/Y (BUFx16f_ASAP7_75t_R)
  64.56 3491.01 ^ wire46465/Y (BUFx16f_ASAP7_75t_R)
  66.92 3557.93 ^ core/_08202_/Y (AND2x2_ASAP7_75t_R)
  23.87 3581.80 ^ core/_08203_/Y (OR5x1_ASAP7_75t_R)
  26.21 3608.01 ^ core/_08204_/Y (OR4x2_ASAP7_75t_R)
  16.73 3624.73 v core/_08206_/Y (OAI21x1_ASAP7_75t_R)
  46.39 3671.13 v core/_08207_/Y (OA21x2_ASAP7_75t_R)
  55.65 3726.78 v core/_11302_/Y (AND2x6_ASAP7_75t_R)
  22.98 3749.76 v wire42467/Y (BUFx16f_ASAP7_75t_R)
 130.71 3880.47 v max_length42465/Y (BUFx16f_ASAP7_75t_R)
  83.54 3964.02 v wire42464/Y (BUFx16f_ASAP7_75t_R)
  96.54 4060.55 v max_length42463/Y (BUFx16f_ASAP7_75t_R)
  48.11 4108.66 v wire42462/Y (BUFx16f_ASAP7_75t_R)
  75.81 4184.47 v lsu/_368760_/Y (AND2x2_ASAP7_75t_R)
  53.39 4237.87 ^ lsu/_368792_/Y (NAND2x2_ASAP7_75t_R)
  42.46 4280.32 v lsu/_368829_/Y (NOR2x1_ASAP7_75t_R)
  54.11 4334.43 v lsu/_368830_/Y (AO21x2_ASAP7_75t_R)
  75.82 4410.25 ^ lsu/_703522_/CON (HAxp5_ASAP7_75t_R)
  19.39 4429.64 v lsu/_368834_/Y (INVx2_ASAP7_75t_R)
  20.14 4449.78 ^ lsu/_703516_/CON (HAxp5_ASAP7_75t_R)
  13.77 4463.55 v lsu/_645234_/Y (INVx1_ASAP7_75t_R)
  58.94 4522.49 v lsu/_703518_/SN (HAxp5_ASAP7_75t_R)
  30.71 4553.21 v lsu/_660144_/Y (XNOR2x1_ASAP7_75t_R)
  35.12 4588.33 v lsu/_660146_/Y (AND5x2_ASAP7_75t_R)
  24.28 4612.61 v wire25820/Y (BUFx16f_ASAP7_75t_R)
  70.04 4682.65 v wire25819/Y (BUFx16f_ASAP7_75t_R)
  84.65 4767.30 v core/_08213_/Y (AO221x2_ASAP7_75t_R)
  44.60 4811.90 v core/_08215_/Y (OR3x1_ASAP7_75t_R)
  70.44 4882.34 v core/_08216_/Y (OR5x2_ASAP7_75t_R)
  93.42 4975.76 ^ core/_11304_/Y (NOR3x2_ASAP7_75t_R)
  43.51 5019.27 ^ wire24788/Y (BUFx16f_ASAP7_75t_R)
  67.05 5086.32 ^ load_slew24787/Y (BUFx16f_ASAP7_75t_R)
  19.21 5105.53 ^ wire24786/Y (BUFx16f_ASAP7_75t_R)
  41.89 5147.41 ^ wire24785/Y (BUFx16f_ASAP7_75t_R)
  87.26 5234.67 ^ max_length24784/Y (BUFx12_ASAP7_75t_R)
  57.80 5292.48 ^ lsu/_368842_/Y (AND2x2_ASAP7_75t_R)
  34.68 5327.16 v lsu/_368843_/Y (NAND2x2_ASAP7_75t_R)
  67.53 5394.68 v lsu/_369154_/Y (XNOR2x2_ASAP7_75t_R)
  44.74 5439.43 v load_slew23779/Y (BUFx16f_ASAP7_75t_R)
  83.73 5523.16 ^ lsu/_703558_/CON (HAxp5_ASAP7_75t_R)
  26.72 5549.88 v lsu/_369182_/Y (INVx2_ASAP7_75t_R)
  23.87 5573.75 v lsu/_369186_/Y (AND3x1_ASAP7_75t_R)
  60.57 5634.32 v lsu/_703564_/SN (HAxp5_ASAP7_75t_R)
  33.03 5667.35 v lsu/_660150_/Y (XNOR2x1_ASAP7_75t_R)
  36.14 5703.49 v lsu/_660152_/Y (AND5x2_ASAP7_75t_R)
  26.36 5729.86 v core/_08220_/Y (AO22x2_ASAP7_75t_R)
  22.53 5752.39 v wire13006/Y (BUFx16f_ASAP7_75t_R)
  70.31 5822.69 v wire13005/Y (BUFx16f_ASAP7_75t_R)
  95.31 5918.01 v core/_08222_/Y (OR4x2_ASAP7_75t_R)
  68.62 5986.63 v core/_08223_/Y (OR5x2_ASAP7_75t_R)
  33.84 6020.47 v core/_08224_/Y (OA21x2_ASAP7_75t_R)
  35.81 6056.27 ^ core/_08225_/Y (AOI21x1_ASAP7_75t_R)
  40.53 6096.81 ^ core/_11305_/Y (AND3x4_ASAP7_75t_R)
  48.13 6144.94 ^ wire13000/Y (BUFx16f_ASAP7_75t_R)
  66.18 6211.12 ^ wire12999/Y (BUFx16f_ASAP7_75t_R)
 108.19 6319.31 ^ wire12998/Y (BUFx16f_ASAP7_75t_R)
 104.73 6424.03 ^ lsu/_389953_/Y (AND2x4_ASAP7_75t_R)
  24.68 6448.71 v lsu/_398123_/Y (NAND2x2_ASAP7_75t_R)
  74.58 6523.29 v lsu/_423337_/Y (OR3x4_ASAP7_75t_R)
  25.24 6548.53 v wire11453/Y (BUFx16f_ASAP7_75t_R)
  92.02 6640.55 v max_length11450/Y (BUFx16f_ASAP7_75t_R)
  88.10 6728.65 v lsu/_423426_/Y (AND2x6_ASAP7_75t_R)
  62.39 6791.05 v max_length7295/Y (BUFx12f_ASAP7_75t_R)
  56.18 6847.23 v max_length7294/Y (BUFx16f_ASAP7_75t_R)
  77.60 6924.83 v max_length7292/Y (BUFx16f_ASAP7_75t_R)
  72.56 6997.38 v max_length7291/Y (BUFx16f_ASAP7_75t_R)
  47.71 7045.10 v wire7290/Y (BUFx16f_ASAP7_75t_R)
  84.83 7129.93 v wire7289/Y (BUFx16f_ASAP7_75t_R)
 147.02 7276.95 v lsu/_423428_/Y (AND3x4_ASAP7_75t_R)
  25.46 7302.40 v max_length3687/Y (BUFx16f_ASAP7_75t_R)
  41.24 7343.65 v wire3679/Y (BUFx16f_ASAP7_75t_R)
  72.87 7416.51 v wire3678/Y (BUFx16f_ASAP7_75t_R)
 122.86 7539.37 v wire3677/Y (BUFx16f_ASAP7_75t_R)
  43.22 7582.60 v max_length3676/Y (BUFx16f_ASAP7_75t_R)
  70.88 7653.48 v wire3675/Y (BUFx16f_ASAP7_75t_R)
  97.68 7751.16 v max_length3674/Y (BUFx16f_ASAP7_75t_R)
  40.86 7792.02 v max_length3673/Y (BUFx16f_ASAP7_75t_R)
  71.25 7863.28 v max_length3671/Y (BUFx16f_ASAP7_75t_R)
  40.36 7903.63 v max_length3670/Y (BUFx16f_ASAP7_75t_R)
  46.40 7950.03 v max_length3668/Y (BUFx16f_ASAP7_75t_R)
  68.40 8018.44 v max_length3666/Y (BUFx16f_ASAP7_75t_R)
  47.09 8065.53 v max_length3665/Y (BUFx16f_ASAP7_75t_R)
  64.88 8130.41 ^ lsu/_644405_/Y (NAND2x1_ASAP7_75t_R)
  22.78 8153.19 ^ lsu/_644406_/Y (OA211x2_ASAP7_75t_R)
   0.01 8153.19 ^ lsu/stq_31_bits_uop_lrs1[4]$_SDFFE_PP0P_/D (DFFHQNx2_ASAP7_75t_R)
        8153.19   data arrival time

8500.00 8500.00   clock clock (rise edge)
1323.45 9823.45   clock network delay (propagated)
 -10.00 9813.45   clock uncertainty
  23.33 9836.78   clock reconvergence pessimism
        9836.78 ^ lsu/stq_31_bits_uop_lrs1[4]$_SDFFE_PP0P_/CLK (DFFHQNx2_ASAP7_75t_R)
  -6.38 9830.41   library setup time
        9830.41   data required time
---------------------------------------------------------
        9830.41   data required time
        -8153.19   data arrival time
---------------------------------------------------------
        1677.21   slack (MET)
```

### Suggested Solution

Perhaps a filtering option in "Timing Report" "Settings" to exclude *BUF* delays and the same for slack histogram?

Also, it would be useful to get worst paths in "Timing Report" with 0 skew between data path and capture path to separate the CTS concern, logic and placement&route.

maliberty, 2024-03-21:
There are a lot of different things mixed in here.  Let's start with what a "top paths for logic only" means outside of the P&R context.  If you want to pretend there are no long wires that need buffering then you could just run with wire load models and skip any buffering.  Does it mean something more than that?

oharboe, 2024-03-22:
> There are a lot of different things mixed in here. Let's start with what a "top paths for logic only" means outside of the P&R context. If you want to pretend there are no long wires that need [...]

I definitely don't want to manually tweak and re-run. That would be enormously time-consuming and error prone. I would have to learn a lot in terms of domain knowledge of what and how to exclude and I would have to get it right on a 24 hour feedback loop. I'd be lucky to get that done in, say, two weeks. Let's say I have some tweak I need to do and I get it wrong(2 days), then I ask a question to someone knowledgable in the field(2 days). That's a working week right there for a single iteration at best.

Instead, I want an ability to present the information from a proper run in various ways. If I make a mistake in what I want to present, that is quickly and interactively corrected in the GUI. Minutes, not weeks.

Example: I want to be able to quantify which paths are worst from a logic only point of view that are worst & I want to look at paths that are worst only from a P&R point of view.

Based upon this, I can set some expectations and make some plans for what changes are needed: in RTL, in OpenROAD config scripts or in OpenROAD itself.

maliberty, 2024-03-22:
That sounds good in theory but I don't know what is means in practice.  What does "worst from a logic only point of view" mean computationally?  Gate delays depend on their loads which depend on P&R or some assumed version like wire load models.  Changing the assumption means recomputing the entire timing to give a reasonable answer (the worst path with P&R is not necessarily the worst path with WL).

oharboe, 2024-03-22:
I see.

oharboe, 2024-03-22:
Reopen when the idea is more cooked and actionable

closed 2024-03-22 by oharboe