The-OpenROAD-Project/OpenROAD #4833: Speed up global route, antenna repair in particular

oharboe, 2024-03-21:
### Description

untar https://drive.google.com/file/d/1YjhsuuE8w0GLaL_opM6-s5izSQq0ML4H/view?usp=sharing

Run:

```
./run-me-BoomTile-asap7-base.sh 
```

global route finishes in an hour or so:

```
OpenROAD v2.0-12680-gc2b1d565b 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
global_route -guide_file bazel-out/k8-fastbuild/bin/results/asap7/BoomTile/base/route.guide -congestion_report_file bazel-out/k8-fastbuild/bin/reports/asap7/BoomTile/base/congestion.rpt -congestion_iterations 30 -congestion_report_iter_step 5 -verbose
[INFO GRT-0020] Min routing layer: M2
[INFO GRT-0021] Max routing layer: M7
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer M1      Track-Pitch = 0.0360  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M2      Track-Pitch = 0.0390  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M3      Track-Pitch = 0.0360  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M4      Track-Pitch = 0.0480  line-2-Via Pitch: 0.0480
[INFO GRT-0088] Layer M5      Track-Pitch = 0.0480  line-2-Via Pitch: 0.0480
[INFO GRT-0088] Layer M6      Track-Pitch = 0.0640  line-2-Via Pitch: 0.0640
[INFO GRT-0088] Layer M7      Track-Pitch = 0.0640  line-2-Via Pitch: 0.0640
[INFO GRT-0019] Found 18847 clock nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 217
[INFO GRT-0003] Macros: 72
[INFO GRT-0004] Blockages: 746004

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
M1         Vertical            0             0          0.00%
M2         Horizontal   95934552      33546961          65.03%
M3         Vertical     110687864      45652441          58.76%
M4         Horizontal   81173092      33753225          58.42%
M5         Vertical     81173092      28729394          64.61%
M6         Horizontal   59034976      21972679          62.78%
M7         Vertical     59034976      27014200          54.24%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0197] Via related to pin nodes: 14672195
[INFO GRT-0198] Via related Steiner nodes: 875677
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 21602716
[INFO GRT-0112] Final usage 3D: 122816530

[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
M1                   0             0            0.00%             0 /  0 /  0
M2            33546961       12998005           38.75%             0 /  0 /  0
M3            45652441       18749496           41.07%             0 /  0 /  0
M4            33753225       11316179           33.53%             0 /  0 /  0
M5            28729394       8018516           27.91%             0 /  0 /  0
M6            21972679       4105254           18.68%             0 /  0 /  0
M7            27014200       2820932           10.44%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total        190668900       58008382           30.42%             0 /  0 /  0

[INFO GRT-0018] Total wirelength: 35929953 um
[INFO GRT-0014] Routed nets: 1953749
[WARNING STA-0450] virtual clock clock_vir can not be propagated.

```

Then antenna repair takes another 2 hours or so:

```
==========================================================================
check_antennas
--------------------------------------------------------------------------
[WARNING GRT-0246] No diode with LEF class CORE ANTENNACELL found.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO ANT-0002] Found 0 net violations.
[INFO ANT-0001] Found 0 pin violations.
Warning: There are 3785 unconstrained endpoints.
[2 hours before completion]
```

I'm aware of SKIP_ANTENNA_REPAIR variable, but of course it would be simpler if antenna repair wasn't so slow that I would have to worry about it.

### Suggested Solution

Speed up antenna repair

maliberty, 2024-03-21:
@luis201420 is reworking antenna repair so he can use this as a test for his work.

eder-matheus, 2024-06-12:
@oharboe We recently added parallelization to check_antennas and repair_antennas. After the PR https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2063, it will be on by default in ORFS. I will try with the provided test case.

closed 2024-06-12 by eder-matheus

eder-matheus, 2024-06-13:
@oharboe As a follow-up on the tests with parallelization, I see that the GRT + the incremental flow runtime is about 53 minutes. The repair_antennas + check_antennas ends in ~4 minutes, using 16 threads. The remaining runtime is spent in estimate_parasitics. I started another run to get the exact value for the estimate_parasitics runtime. I will study if it's possible to speed up this step with parallelization or other optimization.

eder-matheus, 2024-06-13:
@oharboe I found the problem lies in report_metrics, not estimate_parasitics. I didn't notice it when editing the script to get the runtimes, but now I see this:
* 54 minutes in GRT + incremental
* 4 minutes in repair_antennas + check_antennas
* 2.5 minutes in estimate_parasitics

The remaining runtime is only with report_metrics, which is about 2 hours.

oharboe, 2024-06-14:
Same as I saw. Is there a github issue to speed up report_metrics? For now I run with it disabled.

eder-matheus, 2024-06-18:
> Same as I saw. Is there a github issue to speed up report_metrics? For now I run with it disabled.

We have this issue: https://github.com/The-OpenROAD-Project/OpenROAD/issues/4533. I'll check what can be done to improve its runtime.