The-OpenROAD-Project/OpenROAD #4840: sta slow at reporting clock skew

maliberty, 2024-03-22:
### Describe the bug

In this test case report_checks takes ~2 minutes to run while report_clock_skew takes ~25 minutes.  This seems surprisingly different.

This is extracted from https://github.com/The-OpenROAD-Project/OpenROAD/issues/4828 as a standalone sta test case.****

### Expected Behavior

Faster report_clock_skew

### Environment

```shell
Using the sta built from OR:

OpenSTA 2.5.0 073ff2e8c8
```

### To Reproduce

Use https://drive.google.com/file/d/1xqfCArlnRCESUx0uDK77IpEoaadTB0mJ/view?usp=drive_link

% sta sta.tcl

### Relevant log output

```shell
start report_checks Thu Mar 21 14:22:31 PDT 2024
...
start report_clock_skew Thu Mar 21 14:24:26 PDT 2024
...
done Thu Mar 21 14:50:14 PDT 2024
```

QuantamHD, 2024-03-22:
Isn't this the thing Antmicro reported

maliberty, 2024-03-22:
I recall them reporting an issue with threading, was there a similar performance issue reported?  Is there an issue or PR?

QuantamHD, 2024-03-22:
https://github.com/The-OpenROAD-Project/OpenSTA/pull/220

maliberty, 2024-03-22:
That represents a possible solution by threading.  I'm still surprised that the single threaded time spread is so large.  In any case it is a useful example.