The-OpenROAD-Project/OpenROAD #4841: Use some reasonable number of buckets in slack histogram

oharboe, 2024-03-22:
### Description

Run default target in ORFS `make && make gui_final` and view the slack histogram.

Only two buckets are created. Perhaps because the units are nanoseconds and all the slack is within 1 nanosecond?

[image]

### Suggested Solution

Use some better heuristic for number of buckets, say minimum 4-8?

AcKoucher, 2024-03-25:
Would a solution like changing the scale for improving the visualization be acceptable here? Instead of using decimal numbers in ns we can just use µs:

[image]

oharboe, 2024-03-25:
I would prefer a good heuristic and fewer options generally. Note that us, ns, ps are all reasonable units, depending on the design...

AcKoucher, 2024-03-26:
@oharboe Keeping the unit from sta and using somewhat decimal values:

[image]

Would it be reasonable?

oharboe, 2024-03-26:
Very nice.

PR #4858 (2024-03-26): gui: create minimum number of bins for Charts
Fix #4841
closed 2024-03-26 by maliberty