The-OpenROAD-Project/OpenROAD #4843: Improve repair_timing TNS_END_PERCENT=0 after CTS test case

oharboe, 2024-03-23:
### Description

Another test case that includes .lib files for macros.

Repar timing takes ca. 10 minutes and repair_timing has not completed after 10 minutes. Creating feature request, even if duplicate, to save off this test-case in case it is useful.

untar https://drive.google.com/file/d/1O-WZx1yoG4V3p4vkqRwbk-dIaItKTqeQ/view?usp=sharing

This takes ca. 10 minutes:

```
$ ./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-12709-g48675cbda 
[deleted]
[INFO CTS-0101]  Average sink wire length 1413.18 um
[INFO CTS-0102]  Path depth 35 - 35
[WARNING STA-0450] virtual clock clock_vir can not be propagated.
[INFO RSZ-0058] Using max wire length 162um.
[INFO RSZ-0047] Found 16 long wires.
[INFO RSZ-0048] Inserted 58 buffers in 16 nets.
Placement Analysis
---------------------------------
total displacement      10750.2 u
average displacement        0.0 u
max displacement           28.0 u
original HPWL        25963874.1 u
legalized HPWL       26193076.1 u
delta HPWL                    1 %
```

Not finished after 10 hours:

```
repair_timing -repair_tns 0
[INFO RSZ-0094] Found 4039 endpoints with setup violations.
[INFO RSZ-0043] Swapped pins on 1 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0046] Found 8552 endpoints with hold violations.
[WARNING RSZ-0066] Unable to repair all hold violations.
[INFO RSZ-0032] Inserted 2383080 hold buffers.
[still running after 10 hours]
```

### Suggested Solution

Make repair_timing fast.

oharboe, 2024-03-23:
I only read `[INFO RSZ-0032] Inserted 2383080 hold buffers.` now...

"Well, there's your problem", I believe the expression is...

I believe this is a case of where repair_timing should just fail and put itself out of it's missery...

maliberty, 2024-03-23:
There is -max_buffer_percent (default 20%) which should have stopped it unless this is a huge design.  How many instances are there?

oharboe, 2024-03-23:
> There is -max_buffer_percent (default 20%) which should have stopped it unless this is a huge design. How many instances are there?

2-3m for MegaBoom, if I recall.

oharboe, 2024-03-24:
@maliberty Changed the title of this feature request: this is not a problem in CTS or repair timing as such, but rather than it would be nice if repair timing failed doomed runs early with an actionable error message.

I'll continue to work on the megaboom design. I am updating the abstracts for the macros that BoomTile uses to run through global routing so that the timing should be reasonable at the BoomTile CTS level.

mock abstracts from the floorplan stage are useful for iterating on floorplan and placement in BoomTile, but not through CTS.

maliberty, 2024-03-24:
It already should stop at the 20% threshold so it isn't an enhancement so much as a potential bug.

oharboe, 2024-03-24:
Just FYI: when I fix the macro abstracts so that they have .lib files from after global routing, CTS, including repair, completes in 15 minutes, whereas I stopped the report_timing after 10 hours without progress with the macro abstracts made after floorplan.

maliberty, 2024-03-26:
This design has tremendous negative slack which you probably don't get without the macro .libs

closed 2024-03-26 by maliberty

oharboe, 2024-03-26:
> This design has tremendous negative slack which you probably don't get without the macro .libs

Indeed. This issue is only about giving actionable information timely. Thanks!

The macros have pathologically negative slack because they have abstracts created after floorplanning. If I create the anbstracts after global route, the timing is reasonable. mock abstracts from floorplanning are useful to iterate early on floorplan and placement on hhigher level macros, but of course it cant work with CTS.

commit 6b6e2ab2 (2024-03-26): rsz: correctly set the percentage for -max_buffer_percent

Fixes #4843