The-OpenROAD-Project/OpenROAD #4860: How do I complete the configuration of the "openroad -python" environment on the IDE?

lazzy-dog, 2024-03-27:
### Description

I can enter the command "./openroad -python ./myself_code.py" on the terminal and import openroad in "myself_code.py" for code development. But I cannot run openroad as the environment in the Python interpreter under pycharm, and then use pycharm to complete the development. In other words, I don't know how to run the code "./openroad -python ./myself_code.py" using the IDE.

### Suggested Solution

I tried asking chatgpt but he didn't seem to have a solution.

maliberty, 2024-03-27:
Openroad is the main program and starts its own interpreter.   I don't know how pycharm works but it may assume that it is the main program in which case it is incompatible.  There has been much discussion of making an OR pip module but it hasn't come to fruition.

lazzy-dog, 2024-03-27:
So, ask the teacher. How can I know which APIs are officially provided by Openroad when developing code that relies on the "openroad -python" environment without an IDE? Because I don't seem to find an introduction to the corresponding API in the official documentation.

maliberty, 2024-03-27:
The python API is newer and not totally complete.  I suggest you look at 

https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/include/ord/Design.h
https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/include/ord/Tech.h

which are available in Python as a starting point.  For most modules there are test in python as well that can be used as a guide.  If you have specific needs I can try to help.

lazzy-dog, 2024-03-27:
Thanks for your reply, that will be of great help to me.
And I'm trying to solve the above problem with IDE, I will come back if there are any results.
I hope that like the introduction of "import time", I can find its member functions when I use '.'.
[image]

maliberty, 2024-03-27:
I agree that would be very nice.  @proppy any thoughts?

oharboe, 2024-03-27:
> Openroad is the main program and starts its own interpreter. I don't know how pycharm works but it may assume that it is the main program in which case it is incompatible. There has been much discussion of making an OR pip module but it hasn't come to fruition.

Idiomatic Python support requires OR being provided as a PIP module or all sorts of use-cases break.... This is issue is just an example, so in that sense, this issue is a duplicate of #1424

maliberty, 2024-03-27:
No disagreement, just a matter of priorities.

maliberty, 2024-03-27:
I'll close as a dup.  Feel free to reopen if a more specific issue is identified.

closed 2024-03-27 by maliberty

lazzy-dog, 2024-03-28:
> Idiomatic Python support requires OR being provided as a PIP module or all sorts of use-cases break.... This is issue is just an example, so in that sense, this issue is a duplicate of #1424
Thanks for your reply, there still doesn't seem to be a convenient solution for issue #1424. I'm trying something new.

lazzy-dog, 2024-03-28:
> I'll close as a dup. Feel free to reopen if a more specific issue is identified.

Thank you for your efforts.