The-OpenROAD-Project/OpenROAD #4872: Create an IP Checker Utility

rovinski, 2024-03-29:
### Description

_Originally posted by @rovinski in https://github.com/The-OpenROAD-Project/OpenROAD/issues/4871#issuecomment-2026137319_

Currently, it is very possible to import an IP block which is poorly designed and will cause downstream issues in OpenROAD, which becomes difficult to root-cause. A checker utility which checks against a list of common problems can go a long way in identifying problems early.

### Suggested Solution

List of common problems to check (far from an exhaustive list):
- LEF
  - [ ] Macro dimensions are a multiple of the manufacturing grid
  - [ ] Pins are aligned to the manufacturing grid
  - [ ] Each signal pin must obey one of the following:
    - Minimum width, and each pin lies on the routing grid (with an offset is okay)
    - Greater than minimum size, has an access point on the routing grid, and connections from a default width wire OR an NDR rule does not cause a DRC violation.
    - Pins are wide enough such that regardless of offset, a connection from a default width wire (OR NDR rule) does not cause a DRC violation.
  - [ ] Power ring or grid straps are accessible (no obstructions in the way)
  - [ ] Signal pins are accessible (no obstructions in the way)
  - [ ] LEF file does not have an excessive number of polygons which will slow down OR
  - [ ] Ports have antenna info
  - [ ] LEFs in finfet processes have a fin grid property
- LIB
  - All quantities (rise, fall, delay, power, cap, etc.) are sane
    - [ ] No instantaneous transitions for reasonable input values
    - [ ] No excessively long transitions for reasonable input values
    - [ ] No excessively large capacitances
  - [ ] No missing properties
  - [ ] Characterization corner(s) match(es) the characterization corner(s) of the design
- LIB+LEF
  - [ ] Every LEF digital pin has lib timing info
  - [ ] LEF power pins defined match lib power pins
  - [ ] LEF and LIB area match

QuantamHD, 2024-03-29:
Huge +1 from me.

maliberty, 2024-03-30:
Thanks for writing this up.  A new comer doesn't have to implement every check in order to contribute.  Ticking any box is helpful.

rovinski, 2024-03-30:
Indeed, I just wanted to list everything I could think of.

razmikarshakyann, 2024-07-26:
Hi everyone,
I am currently working on this subtask: Each signal pin must obey one of the following:
- Minimum width, and each pin lies on the routing grid (with an offset is okay)
- Greater than minimum size, has an access point on the routing grid, and connections from a default width wire OR an NDR rule does not cause a DRC violation
- Pins are wide enough such that regardless of offset, a connection from a default width wire (OR NDR rule) does not cause a DRC violation

Here is what I have gathered during my research:
- As I understood minimum width is defined as MINWIDTH in LEF files.
- I found the checkPin function in io_parser_helper.cpp, which uses frPin to check pin alignment on the routing grid. However, I am aware of different pin types such dbMPin, dbBPin, etc.
- I discovered code in FlexPA_prep.cpp where prepPoint_pin is called and a variable nAps is used to determine if there are access points for a pin.

Despite the progress, I still have some critical questions that I need your expertise to clarify:
- Given that this task pertains to LEF files, which function(s) or method(s) should I use to gather all necessary pins before reading the DEF file? I could not find a specific function for this purpose and would appreciate any guidance
- Is the checkPin function the only method available to verify if a pin is located on the routing grid, or are there other recommended approaches? 
- Additionally, does the part of makeFastrouteNet function in the GlobalRouter serve any role in this validation?

@maliberty, @rovinski, @QuantamHD I appreciate your assistance in this matter. Any recommendations will greatly help me proceed with my task.

maliberty, 2024-07-26:
@rovinski in order to talk about access points we would have to run pin_access.  This can only happen after macro placement at the earliest.  How are you intending for this checker to be used?  

Perhaps it would be easier to check if  the spacing between pins' centers are integral multiples of the track pitch.  Sufficiently wide pins could be allowed to be exceptions.

titan73, 2024-07-26:
+1
We had to develop such a tool. Some checks are listed here and some others are for compatibility with proprietary tools.
And importantly, to check coherency between views (pins most of the time) between .lib, .lef, behavioural models, .cdl, ...
We have a internal standard for IP package structure but I don't know if there is an open standard for that though.

rovinski, 2024-08-06:
Sorry for late reply.

Ideally pin rules could be checked in isolation without having to have the macro placed yet. It would, however, require tracks and NDRs to be created already (NDRs mainly to avoid false positives).

If the macro has to be placed in order to run pin access, maybe is it possible to place it in a virtual block which then gets deleted after pin access runs?

maliberty, 2024-08-06:
If you are just checking that the pins could be placed on the track grid then you don't need to use pin access and can just look at the pin spacing and shapes.

maliberty, 2024-08-06:
Pins are never blocked by obstructions - pin shapes have priority over obstructions.

rovinski, 2024-08-06:
I mean something really dumb like this:

[image]

Also assuming that metals on top and bottom are blocked.

I've seen this happen in blocks that weren't extracted correctly.

MarufHossain-tech, 2025-07-04:
I want to contribute to this issue. I can add as many checks as. I already created a demo. As soon as you assign it to me I will start finisher polish and do the work with in 3-5 days

MarufHossain-tech, 2025-07-08:
### Summary
This PR adds a LEF Pin Checker utility script that performs:
- Pin-to-grid alignment checks
- Layer validation
- Direction rule enforcement

### Motivation
Supports design integrity at the LEF level. Built for robustness, clarity, and OpenROAD code style.

### Usage
```bash
python3 src/pin_checker/pin_checker.py

MarufHossain-tech, 2025-07-12:
I am a newcomer to the GitHub and open-source contribution world. My work involves PRs, but I have a last query: do I need to make changes to the main source code, or can I create a separate code for IP check?  I assumed it to be a seperate code and file . If it isnt it will take me just few hours to fix it

SajalDevX, 2026-09-25:
Opened #11522 for the three LIB transition/capacitance items, @rovinski, starting where you pointed.

One thing worth flagging: I wanted CHK-132 to compare input pin caps against default_max_capacitance, but OpenSTA never reads that attribute. LibertyReader only picks up default_max_transition and nothing calls setDefaultMaxCapacitance, so it always comes back as absent. Instead of touching OpenSTA I used the largest total_output_net_capacitance the loaded libraries are characterized for, since a pin above that can't be driven by any characterized cell anyway. Easy to swap over if that ever gets read upstream.

Happy to keep going on the rest of the LIB list (missing properties, corner matching) or the LEF+LIB items if those are useful next.

mahjiid, 2026-10-03:
I'd like to work on the lingering multi-layer alignment check mentioned in the discussion from #9405: detect cases where a multi-layer pin's shapes cannot share any valid track-grid alignment across the involved routing layers. I'll keep it focused to `check_ip`, add a regression fixture, and avoid the Liberty work already covered by #11522.

mahjiid, 2026-10-03:
I'm also taking the unchecked LEF/LIB area-consistency item as a separate focused change. The implementation compares the LEF macro physical area against the Liberty cell `area` with a small floating-point tolerance and adds a regression case with an intentional mismatch.

GarvKishoreGupta, 2026-11-13:
For the remaining LEF/Liberty power-pin consistency item: at c76253d, `checkLibertyPins()` skips supply MTerms, and `getLibertyPinDirection()` also rejects power/ground ports ([source](https://github.com/The-OpenROAD-Project/OpenROAD/blob/c76253d3953106aae6d992cf7934a2586a0defd2/src/dbSta/src/IpChecker.cc)). Removing only the skip would therefore not check power-vs-ground mismatches.

The pinned OpenSTA already parses `pg_pin` into `LibertyPort` objects and exposes `pwrGndType()` / `isPwrGnd()` ([parser](https://github.com/The-OpenROAD-Project/OpenSTA/blob/2947a0cab3c01713def012381577f1d45decb87c/liberty/LibertyReader.cc#L1211)). A separate supply comparison appears feasible without a new Liberty parser. The existing `check_ip_liberty_pins` fixture could cover matching supplies, missing `pg_pin`, power/ground mismatch, and no-Liberty behavior; well/bias pins need an explicit policy rather than being treated as ordinary power pins.

This is a source review only; I have not built or run OpenROAD.