The-OpenROAD-Project/OpenROAD #4875: ispd15 mgc_superblue16_a [ERROR GPL-0305] RePlAce diverged at newStepLength.

stefanottili, 2024-03-30:
### Describe the bug

global_placement error's out placing ispd15 mgc_superblue16_a
...
[NesterovSolve] Iter: 2820 overflow: 0.122531 HPWL: 81316178979
[NesterovSolve] Iter: 2830 overflow: 0.128394 HPWL: 81464868051
[ERROR GPL-0305] RePlAce diverged at newStepLength.
[ERROR GUI-0070] GPL-0305

Seem to be the same issue as in https://github.com/The-OpenROAD-Project/OpenLane/issues/657 

### Expected Behavior

global_placement without ERROR

### Environment

```shell
OpenROAD v2.0-12841-g2b53980f4 (Mar 29 2024)
MacOS M1 compiled with mpl2/par
```

### To Reproduce

https://www.ispd.cc/contests/15/web/downloads.html

untar mgc_superblue16_a, gzip all files.

read_verilog design.v.gz
read_lef tech.lef.gz
read_lef cells.lef.gz
read_def floorplan.def.gz
global_placement

### Relevant log output

```shell
OpenROAD v2.0-12841-g2b53980f4 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal1 :"SPACING 0.065 ENDOFLINE 0.056 WITHIN 0.045 PARALLELEDGE 0.11 WITHIN 0.11 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal2 :"SPACING 0.065 ENDOFLINE 0.056 WITHIN 0.045 PARALLELEDGE 0.11 WITHIN 0.11 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal3 :"SPACING 0.065 ENDOFLINE 0.056 WITHIN 0.045 PARALLELEDGE 0.11 WITHIN 0.11 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal4 :"SPACING 0.065 ENDOFLINE 0.056 WITHIN 0.045 PARALLELEDGE 0.11 WITHIN 0.11 ;"
[INFO ODB-0227] LEF file: tech.lef.gz, created 17 layers, 276 vias
[INFO ODB-0227] LEF file: cells.lef.gz, created 4634 library cells
[INFO ODB-0127] Reading DEF file: floorplan.def.gz
[INFO ODB-0128] Design: superblue16
[INFO ODB-0094] 		Created 100000 Insts
[INFO ODB-0094] 		Created 200000 Insts
[INFO ODB-0094] 		Created 300000 Insts
[INFO ODB-0094] 		Created 400000 Insts
[INFO ODB-0094] 		Created 500000 Insts
[INFO ODB-0094] 		Created 600000 Insts
[INFO ODB-0097] 		Created 100000 Nets
[INFO ODB-0097] 		Created 200000 Nets
[INFO ODB-0097] 		Created 300000 Nets
[INFO ODB-0097] 		Created 400000 Nets
[INFO ODB-0097] 		Created 500000 Nets
[INFO ODB-0097] 		Created 600000 Nets
[INFO ODB-0130]     Created 17498 pins.
[INFO ODB-0131]     Created 680869 components and 2271804 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 0 connections.
[INFO ODB-0133]     Created 697303 nets and 2262149 connections.
[INFO ODB-0134] Finished DEF file: floorplan.def.gz
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 100 900
[INFO GPL-0004] CoreAreaLxLy: 1800 1800
[INFO GPL-0005] CoreAreaUxUy: 1484700 1611000
[INFO GPL-0006] NumInstances: 629577
[INFO GPL-0007] NumPlaceInstances: 629159
[INFO GPL-0008] NumFixedInstances: 418
[INFO GPL-0009] NumDummyInstances: 0
[INFO GPL-0010] NumNets: 697303
[INFO GPL-0011] NumPins: 2279647
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 1488000 1616000
[INFO GPL-0014] CoreAreaLxLy: 1800 1800
[INFO GPL-0015] CoreAreaUxUy: 1484700 1611000
[INFO GPL-0016] CoreArea: 2386282680000
[INFO GPL-0017] NonPlaceInstsArea: 1156878270000
[INFO GPL-0018] PlaceInstsArea: 576192240000
[INFO GPL-0019] Util(%): 46.87
[INFO GPL-0020] StdInstsArea: 576192240000
[INFO GPL-0021] MacroInstsArea: 0
[InitialPlace]  Iter: 1 CG residual: 0.00329998 HPWL: 69801065376
[InitialPlace]  Iter: 2 CG residual: 0.00071870 HPWL: 41794679501
[InitialPlace]  Iter: 3 CG residual: 0.00021728 HPWL: 38650540308
[InitialPlace]  Iter: 4 CG residual: 0.00016672 HPWL: 36196752556
[InitialPlace]  Iter: 5 CG residual: 0.00012805 HPWL: 34579328342
[InitialPlace]  Iter: 6 CG residual: 0.00008006 HPWL: 33984336254
[InitialPlace]  Iter: 7 CG residual: 0.00007335 HPWL: 33657136033
[InitialPlace]  Iter: 8 CG residual: 0.00007006 HPWL: 33422390684
[InitialPlace]  Iter: 9 CG residual: 0.00005999 HPWL: 33214506821
[InitialPlace]  Iter: 10 CG residual: 0.00004697 HPWL: 33092418765
[InitialPlace]  Iter: 11 CG residual: 0.00004500 HPWL: 32981388403
[InitialPlace]  Iter: 12 CG residual: 0.00004444 HPWL: 32887949555
[InitialPlace]  Iter: 13 CG residual: 0.00006286 HPWL: 32800212673
[InitialPlace]  Iter: 14 CG residual: 0.00004365 HPWL: 32741225394
[InitialPlace]  Iter: 15 CG residual: 0.00003815 HPWL: 32662774636
[InitialPlace]  Iter: 16 CG residual: 0.00003604 HPWL: 32607003044
[InitialPlace]  Iter: 17 CG residual: 0.00003854 HPWL: 32548334721
[InitialPlace]  Iter: 18 CG residual: 0.00003687 HPWL: 32493402310
[InitialPlace]  Iter: 19 CG residual: 0.00002783 HPWL: 32437921451
[InitialPlace]  Iter: 20 CG residual: 0.00002545 HPWL: 32396409787
[INFO GPL-0031] FillerInit: NumGCells: 957289
[INFO GPL-0032] FillerInit: NumGNets: 697303
[INFO GPL-0033] FillerInit: NumGPins: 2279647
[INFO GPL-0023] TargetDensity: 0.70
[INFO GPL-0024] AveragePlaceInstArea: 915813
[INFO GPL-0025] IdealBinArea: 1308304
[INFO GPL-0026] IdealBinCnt: 1823951
[INFO GPL-0027] TotalBinArea: 2386282680000
[INFO GPL-0028] BinCnt: 1024 1024
[INFO GPL-0029] BinSize: 1449 1572
[INFO GPL-0030] NumBins: 1048576
[NesterovSolve] Iter: 1 overflow: 1.02121 HPWL: 20013586272
[NesterovSolve] Iter: 10 overflow: 0.945545 HPWL: 22006482627
[NesterovSolve] Iter: 20 overflow: 0.902363 HPWL: 22439098630
[NesterovSolve] Iter: 30 overflow: 0.88417 HPWL: 22435228015
[NesterovSolve] Iter: 40 overflow: 0.876783 HPWL: 22350627540
[NesterovSolve] Iter: 50 overflow: 0.875639 HPWL: 22279481638
[NesterovSolve] Iter: 60 overflow: 0.876631 HPWL: 22244875493
[NesterovSolve] Iter: 70 overflow: 0.876035 HPWL: 22232145785
[NesterovSolve] Iter: 80 overflow: 0.875769 HPWL: 22219304496
...
[NesterovSolve] Iter: 2770 overflow: 0.122641 HPWL: 80568346067
[NesterovSolve] Iter: 2780 overflow: 0.122649 HPWL: 80719484757
[NesterovSolve] Iter: 2790 overflow: 0.122669 HPWL: 80869879121
[NesterovSolve] Iter: 2800 overflow: 0.122647 HPWL: 81018958720
[NesterovSolve] Iter: 2810 overflow: 0.122564 HPWL: 81167076695
[NesterovSolve] Iter: 2820 overflow: 0.122531 HPWL: 81316178979
[NesterovSolve] Iter: 2830 overflow: 0.128394 HPWL: 81464868051
[ERROR GPL-0305] RePlAce diverged at newStepLength.
[ERROR GUI-0070] GPL-0305
```

stefanottili, 2024-04-07:
One more datapoint with regard to ispd15 global placement contest:

mgc_superblue12 doesn't finish global_place "out of the box" either, but with a different ERROR.

....
[NesterovSolve] Iter: 500 overflow: 0.224655 HPWL: 29136337676
[NesterovSolve] Iter: 510 overflow: 0.218266 HPWL: 26860168087
[NesterovSolve] Iter: 520 overflow: 0.192555 HPWL: 26266994755
[NesterovSolve] Iter: 530 overflow: 0.15962 HPWL: 27260482435
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.
[ERROR GUI-0070] GPL-0307

stefanottili, 2024-04-11:
Seems to be related to
https://github.com/The-OpenROAD-Project/OpenROAD/issues/2580

stefanottili, 2025-08-11:
Revisiting this testcase:
As of August 10 2025, gpl doesn't ERROR out at 530 iterations, it goes all the way up to iteration 5000 without converging.
It would be nice if gpl would print its runtime.
```
     ...
     4850 |   0.3778 |  2.432200e+11 |  +23.75% |  2.76e-07 |      
     4860 |   0.3716 |  2.495252e+11 |   +2.59% |  2.59e-07 |      
     4870 |   0.3594 |  2.507338e+11 |   +0.48% |  2.56e-07 |      
     4880 |   0.3894 |  2.580363e+11 |   +2.91% |  2.35e-07 |      
     4890 |   0.3333 |  2.617515e+11 |   +1.44% |  2.17e-07 |      
     4900 |   0.3357 |  2.521510e+11 |   -3.67% |  2.93e-07 |      
     4910 |   0.3726 |  2.413491e+11 |   -4.28% |  3.13e-07 |      
     4920 |   0.4361 |  3.083909e+11 |  +27.78% |  2.69e-07 |      
     4930 |   0.3757 |  2.958962e+11 |   -4.05% |  2.71e-07 |      
     4940 |   0.3714 |  2.869799e+11 |   -3.01% |  2.65e-07 |      
     4950 |   0.3674 |  2.978962e+11 |   +3.80% |  2.49e-07 |      
     4960 |   0.3503 |  2.795903e+11 |   -6.15% |  2.56e-07 |      
     4970 |   0.3328 |  2.832405e+11 |   +1.31% |  2.63e-07 |      
     4980 |   0.3289 |  2.526391e+11 |  -10.80% |  3.00e-07 |      
     4990 |   0.3352 |  2.454552e+11 |   -2.84% |  3.08e-07 |      
[WARNING GPL-1010] GPL reached the maximum number of iterations for nesterov 5000. Placement may have failed to converge.
[INFO GPL-1014] Final placement area: 576192.24 (+0.00%)
```

gudeh, 2025-08-11:
Hi @stefanottili, could you provide an artifact for the case with ispd15_mgc_superblue12? The new feature for "revert if diverge" usually works fine.

stefanottili, 2025-08-11:
https://www.ispd.cc/contests/15/web/downloads.html

stefanottili, 2026-03-06:
ispd15_mgc_superblue_12 fails in dpl too ...
It's reporting different DPL errors then ispd15_mg_superblue_16a
```
      530 |   0.1764 |  2.809679e+07 |   +0.60% |  1.98e-06 |      
      540 |   0.1506 |  3.207621e+07 |  +14.16% |  2.65e-06 |      
[WARNING GPL-0323] Divergence detected between consecutive iterations
Divergence occured in 1 regions.
[WARNING GPL-0998] Divergence detected, reverting to snapshot with min hpwl.
[WARNING GPL-0999] Revert to iter:  522 overflow: 0.205 HPWL: 27929868406
[INFO GPL-1014] Final placement area: 1144411.58 (+0.00%)
>>> detailed_placement
[INFO DPL-0034] Detailed placement failed on the following 46 instances:
[INFO DPL-0035]  o1281346
[INFO DPL-0035]  o1281345
...
[INFO DPL-0035]  o1281067
[INFO DPL-0035]  o1281068
[ERROR DPL-0036] Detailed placement failed.
[ERROR GUI-0070] DPL-0036
```

stefanottili, 2026-04-18:
@gudeh Since default diamond fails, I tried -use_negotiation, but this seems to be extremely slow, no output for 1/2 hour ...
but then I finally got a flurry of messages, no runtime reported ;-(
And the same 36 violations remain that diamond can't solve either ...
```
[INFO DPL-0006] Core area: 2925951.12 um^2, Instances area: 1685960.73 um^2, Utilization: 57.6%
[INFO DPL-0005] Diamond search max displacement: +/- 500 sites horizontally, +/- 100 rows vertically.
[INFO DPL-1102] Legalizing using negotiation legalizer.
Negotiation iteration 0: total overflow 60906.
Negotiation iteration 1: total overflow 42488.
Negotiation iteration 2: total overflow 39745.
Negotiation iteration 3: total overflow 39010.
Negotiation iteration 4: total overflow 38824.
Negotiation iteration 5: total overflow 38819.
Negotiation iteration 6: total overflow 38778.
Negotiation iteration 7: total overflow 38760.
Negotiation iteration 8: total overflow 38760.
Negotiation iteration 9: total overflow 38760.
Negotiation iteration 10: total overflow 38760.
[WARNING DPL-0700] Negotiation phase 1: overflow stuck at 38760 for 3 consecutive iterations.
Using old diamond search for remaining cells.
diamondRecovery: cell o1208533 recovered at (12594, 368).
diamondRecovery: cell o1281041 recovered at (12328, 333).
diamondRecovery: cell o1281042 recovered at (12320, 332).
diamondRecovery: cell o1208615 recovered at (12194, 367).
diamondRecovery: cell o1208563 recovered at (12195, 366).
....
diamondRecovery: cell o1286361 recovered at (14153, 1727).
diamondRecovery: cell o1286822 recovered at (13823, 1611).
diamondRecovery: recovered 2682/1281607 stuck cells.
Negotiation iteration 400: total overflow 1576.
Negotiation iteration 401: total overflow 1576.
Negotiation iteration 402: total overflow 1576.
Negotiation iteration 403: total overflow 1576.
[WARNING DPL-0702] Negotiation phase 2: overflow stuck at 1576 for 3 consecutive iterations.
diamondRecovery: recovered 0/1281607 stuck cells.
[WARNING DPL-0701] NegotiationLegalizer did not fully converge. Violations remain: 36
Placement Analysis
---------------------------------
total displacement     579285.9 u
average displacement        0.5 u
max displacement            0.9 u
original HPWL        27671503.2 u
legalized HPWL       29395522.9 u
delta HPWL                    6 %
```

stefanottili, 2026-04-30:
with the REGIONS and GROUPS commented out in the floorplan.def:

1) without -use_negotiation, 230s runtime: dpl is instantanious, no errors
2) with -use_negotiation dpl is very slow and doesn't finish clean. 910s total runtime: Violations remain: 49

a) I've use "disjoint regions" on a family of low power chips a decade years ago.
IMHO the ispd15 test cases represent valid use cases for placement with regions.

```
diamondRecovery: cell o613469 recovered at (4494, 100).
diamondRecovery: cell o527259 recovered at (0, 1690).
diamondRecovery: recovered 2963/677746 stuck cells.
Negotiation iteration 400: total overflow 557.
Negotiation iteration 401: total overflow 557.
Negotiation iteration 402: total overflow 557.
Negotiation iteration 403: total overflow 557.
[WARNING DPL-0702] Negotiation phase 2: overflow stuck at 557 for 3 consecutive iterations.
diamondRecovery: recovered 0/677746 stuck cells.
[WARNING DPL-0701] NegotiationLegalizer did not fully converge. Violations remain: 49
Placement Analysis
---------------------------------
total displacement     306406.8 u
average displacement        0.5 u
max displacement            0.9 u
original HPWL        26976251.7 u
legalized HPWL       27468963.1 u
delta HPWL                    2 %

910.24s user 5.81s system 123% cpu 12:19.46 total
```