The-OpenROAD-Project/OpenROAD #4876: MacOS link "undefined symbol" absl::lts_20240116::container_internal::kEmptyGroup, gpl::MBFF::RunILP

stefanottili, 2024-03-30:
### Describe the bug

As of Mar 30, following a
```
git pull https://github.com/The-OpenROAD-Project/OpenROAD.git
```
openroad doesn't link on M1 macOS Sonoma 14.4.1 any more, I've been able to build/link openroad yesterday.
14.4.1 seems to have changed clang, in another project more clang errors showed up.
Blowing away the build dir, rerunning etc/DependencyInstaller.sh and not compiling in mpl2/par doesn't fix this.
Any idea how to debug this ?
```
Undefined symbols for architecture arm64:
  "absl::lts_20240116::container_internal::kEmptyGroup", referenced from:
      gpl::MBFF::RunILP(std::__1::vector<gpl::Flop, std::__1::allocator<gpl::Flop>> const&, std::__1::vector<gpl::Tray, std::__1::allocator<gpl::Tray>> const&, std::__1::vector<std::__1::pair<int, int>, std::__1::allocator<std::__1::pair<int, int>>>&, float, std::__1::vector<int, std::__1::allocator<int>>) in gpl.a[17](mbff.cpp.o)
      gpl::MBFF::RunILP(std::__1::vector<gpl::Flop, std::__1::allocator<gpl::Flop>> const&, std::__1::vector<gpl::Tray, std::__1::allocator<gpl::Tray>> const&, std::__1::vector<std::__1::pair<int, int>, std::__1::allocator<std::__1::pair<int, int>>>&, float, std::__1::vector<int, std::__1::allocator<int>>) in gpl.a[17](mbff.cpp.o)
ld: symbol(s) not found for architecture arm64
```
On Sonoma 14 openroad only links when using or-tools 9.8

etc/DependencyInstaller.sh
```
@@ -203,12 +203,12 @@ EOF
 
 _installOrTools() {
     os=$1
     version=$2
     arch=$3
-    orToolsVersionBig=9.5
-    orToolsVersionSmall=${orToolsVersionBig}.2237
+    orToolsVersionBig=9.8
+    orToolsVersionSmall=${orToolsVersionBig}.3296
 
     rm -rf "${baseDir}"
     mkdir -p "${baseDir}"
     if [[ ! -z "${PREFIX}" ]]; then mkdir -p "${PREFIX}"; fi
     cd "${baseDir}"
@@ -725,11 +725,11 @@ EOF
             _installOrTools "centos" "8" "amd64"
         fi
         ;;
     "Darwin" )
         _installDarwin
-        _installOrTools "macOS" "13.0.1" $(uname -m)
+        _installOrTools "macOS" "14.1" $(uname -m)
         cat <<EOF
```

### Expected Behavior

be able to link openroad on macOS Sonoma 14.4.1

### Environment

```shell
Git commit: 7f55436ed60402b5be173f6d1ec4cd39e3610058
kernel: Darwin 23.4.0
os: macOS 14.4.1
cmake version 3.29.0
CMake Deprecation Warning at third-party/abc/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning at src/CMakeLists.txt:245 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

Number of processor cores: 8
-- The CXX compiler identification is AppleClang 15.0.0.15000309
[... 128 lines cut ...]
-- Found OpenMP: TRUE (found version "5.0")
-- GPU is not enabled
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: TCL_HEADER-NOTFOUND
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found version "1.84.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable version "1.84.0", minimum required is "1.78")
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: TCL_HEADER-NOTFOUND
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found version "1.84.0") found components: serialization system thread
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found version "1.84.0")
-- Removing MPL2 and PAR to avoid run time fatal error.
-- TCL readline disabled
-- Tcl Extended disabled
-- Python3 enabled
-- Removing PAR_PY to avoid run time fatal error.
-- Configuring done (6.6s)
-- Generating done (0.8s)
```

### To Reproduce

./build_openroad --local --latest

maliberty, 2024-04-01:
Would you make a PR for your DependencyInstaller.sh changes and we can see if upgrading presents any issues.

stefanottili, 2024-04-02:
I have no idea how abseil got installed, but removing it and re2 which depends on it solved my link issues.

brew uninstall re2
brew uninstall abseil

closed 2024-04-02 by stefanottili