The-OpenROAD-Project/OpenROAD #4878: Crash in GlobalRouter repairAntennas

MichaelBell, 2024-03-30:
### Describe the bug

While running the Global Routing step, on certain designs a crash occurs in repair antennas.  This only seems to happen if more than one iteration is required.  This seems very similar to #4471, but I have confirmed it still happens with a very recent version of OpenROAD.

### Expected Behavior

The global router should not crash

### OpenROAD Environment

```shell
I'm using OpenROAD through an OpenLane docker image efabless/openlane-tools:openroad_app-9741bc497a3d5f434b0dacdc3a05af4ebe78846c-centos-7-amd64 using OpenROAD 

But this happens with various versions of OpenROAD, including 6ff71c8822a72dc0c33dd5a8a10076d8691b5a4f and 0889970d1790a2617e69f253221b8bd7626e51dc (used in OpenLane tag 2024.03.07 and 2024.03.12).

It does not happen with 6f9b2bb8b808b1bb5831d4525d868212ae50517a (used in OpenLane tag 2023.11.23).
```

### OpenLane Environment

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
Kernel: Linux v5.15.146.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v20.10.25 (OK)
OpenLane Git Version: fix-antenna-crash-dev
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

1648ffd 2024-03-30T19:27:25+00:00 Update OpenROAD again - Mike Bell -  (HEAD -> fix-antenna-crash)
60f6256 2024-03-29T23:08:42+00:00 Update OpenROAD - Mike Bell -  ()
a663df2 2024-03-06T15:47:26+02:00 Update OpenROAD (#2093) - Mohamed Gaber -  (grafted)
---
Git Remotes

mike    git@github.com:MichaelBell/OpenLane.git (fetch)
mike    git@github.com:MichaelBell/OpenLane.git (push)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### To Reproduce

Clone the [openroad-antenna-bug](https://github.com/MichaelBell/tt06-tinyQV/tree/openroad-antenna-bug) branch of https://github.com/MichaelBell/tt06-tinyQV recursively
Follow https://docs.google.com/document/d/1aUUZ1jthRpg4QURIIyzlOaPWlmQzr-jBn3wZipVUPt4/edit#heading=h.wwc5ldl01nl5 to set up the OpenLane flow for TinyTapeout, with the following changes:
- Use the above project instead of the factory test project.
- Use the 2024.03.07 tag of OpenLane

Run:

    tt/tt_tool.py --create-user-config
    tt/tt_tool.py --harden

I've also attached a zip of the issue reproducible folder.
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/14813589/issue_reproducible.zip)

### Relevant log output

```shell
From 19-global.log:

[INFO GRT-0018] Total wirelength: 268085 um
[INFO GRT-0014] Routed nets: 5770
[INFO GRT-0006] Repairing antennas, iteration 1.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 32 antenna violations.
[... 109 lines cut ...]
EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /work/runs/wokwi/issue_reproducible…
Done.
[INFO]: Reproducible packaged at '../work/runs/wokwi/issue_reproducible'.
Traceback (most recent call last):
  File "/home/mdb36/tt/tt06-tinyQV/tt/tt_tool.py", line 150, in <module>
    project.harden()
  File "/home/mdb36/tt/tt06-tinyQV/tt/project.py", line 481, in harden
    with open(os.path.join(self.local_dir, commit_id_json_path), "w") as f:
FileNotFoundError: [Errno 2] No such file or directory: './runs/wokwi/results/final/commit_id.json'
```

### Additional Context

This issue also caused failures for certain Tiny Tapeout 05 projects (that previously succeeded) when testing the 2024.03.12 tag of OpenLane:
https://github.com/TinyTapeout/tinytapeout-05-reharden/actions/runs/8264211005/job/22609263738
https://github.com/TinyTapeout/tinytapeout-05-reharden/actions/runs/8264211005/job/22609256508

maliberty, 2024-04-01:
@luis201420 is working on rewriting this chunk of code.   I believe that will solve you problem but it will be a bit until the new code is ready.

maliberty, 2024-04-01:
The general goal is to no longer call orderWires during antenna repair

dlmiles, 2024-04-05:
When investigating this, the following assertion is triggered.

Is the frame posted before (containting `db::orderWires(...)` call a symptom, while this frame is closer to the cause ?

```
[INFO GRT-0006] Repairing antennas, iteration 3.
[INFO GRT-0043] No OR_DEFAULT vias defined.
openroad: /openroad/src/odb/src/db/dbWireCodec.cpp:393: int odb::dbWireEncoder::addTechVia(odb::dbTechVia*): Assertion `0' failed.
Signal 6 received
Stack trace:
 0# 0x0000000004065175 in openroad
 1# 0x00007F5413F2B400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007F5413F241A6 in /lib64/libc.so.6
 5# 0x00007F5413F24252 in /lib64/libc.so.6
 6# odb::dbWireEncoder::addTechVia(odb::dbTechVia*) in openroad
 7# grt::RepairAntennas::addWireTerms(grt::Net*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&, int, int, int, odb::dbTechLayer*, std::map<grt::RoutePt, grt::RoutePtPins, std::less<grt::RoutePt>, std::allocator<std::pair<grt::RoutePt const, grt::RoutePtPins> > >&, odb::dbWireEncoder&, std::map<int, odb::dbTechVia*, std::less<int>, std::allocator<std::pair<int const, odb::dbTechVia*> > >&, bool) in openroad
 8# grt::RepairAntennas::makeNetWire(odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&, std::map<int, odb::dbTechVia*, std::less<int>, std::allocator<std::pair<int const, odb::dbTechVia*> > >&) in openroad
 9# grt::RepairAntennas::makeNetWires(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int) in openroad
10# grt::RepairAntennas::checkAntennaViolations(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int, odb::dbMTerm*, float) in openroad
11# grt::GlobalRouter::repairAntennas(odb::dbMTerm*, int, float) in openroad
12# grt::repair_antennas(odb::dbMTerm*, int, float) in openroad
...
```

maliberty, 2024-04-05:
The problem is order wires and is understood.

MichaelBell, 2024-04-10:
This is causing problems with a number of Tiny Tapeout 6 designs, now that Tiny Tapeout has moved to an OpenLane version with the bug present.

This patch seems to work (or at least resolve the crash) on at least a couple of designs, suggesting the problem might be happening in makeNetWires, as dlmiles suggested above.  But none of us in the Tiny Tapeout community fully understand the implications.  Any comment on whether this is a good idea would be welcome!

https://github.com/obriensp/OpenROAD/commit/d1f0b60d7b31393ec0afb8f96e1ca02bb465b3fc

maliberty, 2024-04-10:
We are in the midst of fixing another bug in the same spot of code so results will probably change again.  In general such things are prone to fix one design and break another due to the brittle nature of order_wires.   The real solution is still in progress but you should use whatever local methods you can until then.

dlmiles, 2024-04-16:
> [obriensp@d1f0b60](https://github.com/obriensp/OpenROAD/commit/d1f0b60d7b31393ec0afb8f96e1ca02bb465b3fc)

This recent fix just landed in OR looks the same to me as the patch from this comment I quopte here.

https://github.com/The-OpenROAD-Project/OpenROAD/commit/b98aca833f9827e7f20a8fa3c6defa56fa92e506#diff-02f10642344ceceb7764f552b99c58c78482dbd86eff6a8c6c17aae6407b7d70R304

eder-matheus, 2024-06-05:
@MichaelBell Could you try the latest version of OpenROAD? We've merged the new antenna checker code that don't use _orderWires_, so this crash should not be happening anymore.

I've tried to run your attached testcase, but it fails with a missing SDC file.

MichaelBell, 2024-06-05:
Thanks, I'll take a look.

At a guess of what's going wrong with the SDC file - it should be using `base.sdc` from my `src` directory, but that does `read_sdc $::env(OPENLANE_ROOT)/scripts/base.sdc`, maybe `OPENLANE_ROOT` isn't set in your environment?

eder-matheus, 2024-06-05:
> Thanks, I'll take a look.
> 
> At a guess of what's going wrong with the SDC file - it should be using `base.sdc` from my `src` directory, but that does `read_sdc $::env(OPENLANE_ROOT)/scripts/base.sdc`, maybe `OPENLANE_ROOT` isn't set in your environment?

Yeah, that's the case. I tried to find the file under openlane/scripts but couldn't see it.

MichaelBell, 2024-06-05:
Weird, it should be here: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/base.sdc

eder-matheus, 2024-06-12:
> Weird, it should be here: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/base.sdc

@MichaelBell It wasn't in the test case you have attached. Either way, I'll close this issue for now, but if you happen to have any other crashes, feel free to reopen it or create a new issue.

closed 2024-06-12 by eder-matheus