The-OpenROAD-Project/OpenROAD #4886: report_clock_skew doesn't report the largest skew

oharboe, 2024-04-02:
### Description

I don't know if this is a bug, a feature request or just a misunderstanding on my part what the report_clock_skew feature is doing, so filing a feature request.

Why is the particular path reported?

If I look at the clock tree, it is not the largest skew.

```
>>> report_clock_skew
Clock clock
 337.78 source latency ces_4_7/clock ^
  78.36 source clock tree delay
-359.04 target latency io_lsbs_39$_DFF_P_/CLK ^
  -3.01 CRPR
--------------
  54.09 setup skew

Clock clock_vir
No launch/capture paths found.
```

[image]

The test case was produce with `make DESIGN_CONFIG=designs/asap7/mock-array/config.mk`

To launch GUI:

untar [gui.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/14831958/gui.tar.gz)

```
. vars-mock-array-asap7-base.sh
ODB_FILE=results/asap7/mock-array/base/5_route.odb ./run-me-mock-array-asap7-base.sh
OpenROAD v2.0-12845-gd71278409 
[deleted]
```

### Suggested Solution

I expected the largest skew to be reported.

maliberty, 2024-04-02:
The skew is only reported when a path exists between the endpoints.  So the CTS viewer might appear to have a worse value but there may be no path the exercises it.  Do you have an example showing a path with a worse skew?

oharboe, 2024-04-02:
> The skew is only reported when a path exists between the endpoints. So the CTS viewer might appear to have a worse value but there may be no path the exercises it. Do you have an example showing a path with a worse skew?

After the concept of "useful skew" was introduced to, I no longer know what "worse skew" means. It is ambigious.

Just look at some of the other macros in the clock tree view: there are macros with more skew.

maliberty, 2024-04-02:
I think you may be confusing latency with skew.  Skew is the difference in arrival between two endpoints not the absolute time of arrival (latency).

Useful skew usually refers to intentional skew that improves slack.  It is an optimization.

oharboe, 2024-04-02:
Ah, yes, indeed. I will investigate this testcase with that in mind and reopen if I see anything of interest.

oharboe, 2024-04-03:
@maliberty What's going on here. I expected report_clock_skew to be idempotent, meaning same result if invoke twice?

```
>>> report_clock_skew
Clock clock
 337.78 source latency ces_4_7/clock ^
  78.36 source clock tree delay
-359.04 target latency io_lsbs_39$_DFF_P_/CLK ^
  -3.01 CRPR
--------------
  54.09 setup skew

Clock clock_vir
No launch/capture paths found.

>>> report_checks -through {ces_4_7/clock} -to {io_lsbs_39$_DFF_P_}
Startpoint: ces_4_7 (rising edge-triggered flip-flop clocked by clock)
Endpoint: io_lsbs_39$_DFF_P_ (rising edge-triggered flip-flop clocked by clock)
Path Group: clock
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock clock (rise edge)
 337.78  337.78   clock network delay (propagated)
   0.00  337.78 ^ ces_4_7/clock (Element)
 152.65  490.43 ^ ces_4_7/io_lsbOuts_7 (Element)
   0.03  490.47 ^ io_lsbs_39$_DFF_P_/D (DFFHQNx2_ASAP7_75t_R)
         490.47   data arrival time

 250.00  250.00   clock clock (rise edge)
 359.04  609.04   clock network delay (propagated)
 -10.00  599.04   clock uncertainty
   3.01  602.05   clock reconvergence pessimism
         602.05 ^ io_lsbs_39$_DFF_P_/CLK (DFFHQNx2_ASAP7_75t_R)
  -4.63  597.42   library setup time
         597.42   data required time
---------------------------------------------------------
         597.42   data required time
        -490.47   data arrival time
---------------------------------------------------------
         106.95   slack (MET)

>>> report_clock_skew
Clock clock
 332.49 source latency ces_4_6/clock ^
  78.36 source clock tree delay
-356.22 target latency io_lsbs_38$_DFF_P_/CLK ^
  -3.01 CRPR
--------------
  51.61 setup skew

Clock clock_vir
No launch/capture paths found.
```

tspyrou, 2024-04-03:
@oharboe after the report_checks with the -through's, if you do a report_checks with no options, the report_clock_skew will show the original value. report_checks with -through must be leaving behind some data fragments. You can use this as a workaround for now.

jjcherry56, 2024-04-04:
Indeed, -through is the culprit. In this case -from would have sufficed (register clock pins are valid start points) and it would not have tickled the problem. Fixed in OpenSTA commit
80598f83 report_clock_skew after report_checks -through

oharboe, 2024-04-04:
@jjcherry56 @tspyrou Works, I compiled & tested locally; report_clock_skew now reports the same before and after report_checks -through.

I take it @maliberty will update sta in OpenROAD in due course.

```
OpenROAD v2.0-12959-gfb13ef8e6 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0450] virtual clock clock_vir can not be propagated.
Estimating parasitics
[WARNING GUI-0066] Heat map "IR Drop" has not been populated with data.
>>> report_clock_skew
Clock clock
 337.78 source latency ces_4_7/clock ^
  78.36 source clock tree delay
-359.04 target latency io_lsbs_39$_DFF_P_/CLK ^
  -3.01 CRPR
--------------
  54.09 setup skew

Clock clock_vir
No launch/capture paths found.

>>> report_checks -through {ces_4_7/clock} -to {io_lsbs_39$_DFF_P_}
Startpoint: ces_4_7 (rising edge-triggered flip-flop clocked by clock)
Endpoint: io_lsbs_39$_DFF_P_ (rising edge-triggered flip-flop clocked by clock)
Path Group: clock
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock clock (rise edge)
 337.78  337.78   clock network delay (propagated)
   0.00  337.78 ^ ces_4_7/clock (Element)
 152.65  490.43 ^ ces_4_7/io_lsbOuts_7 (Element)
   0.03  490.47 ^ io_lsbs_39$_DFF_P_/D (DFFHQNx2_ASAP7_75t_R)
         490.47   data arrival time

 250.00  250.00   clock clock (rise edge)
 359.04  609.04   clock network delay (propagated)
 -10.00  599.04   clock uncertainty
   3.01  602.05   clock reconvergence pessimism
         602.05 ^ io_lsbs_39$_DFF_P_/CLK (DFFHQNx2_ASAP7_75t_R)
  -4.63  597.42   library setup time
         597.42   data required time
---------------------------------------------------------
         597.42   data required time
        -490.47   data arrival time
---------------------------------------------------------
         106.95   slack (MET)

>>> report_clock_skew
Clock clock
 337.78 source latency ces_4_7/clock ^
  78.36 source clock tree delay
-359.04 target latency io_lsbs_39$_DFF_P_/CLK ^
  -3.01 CRPR
--------------
  54.09 setup skew

Clock clock_vir
No launch/capture paths found.
```

closed 2024-04-04 by oharboe

maliberty, 2024-04-04:
I've started an update/test cycle