The-OpenROAD-Project/OpenROAD #4887: 45 degrees labels to have more space

oharboe, 2024-04-02:
### Description

The labels don't fit

[image]

### Suggested Solution

Make labels 45 degrees to make more space for labels

oharboe, 2024-04-02:
Use 45 degrees only if necessary and add enough space vertically for the labels....

Something  like:
[image]

The naive change didn't work, because it uses elipsis unecessarily:

[image]

stefanottili, 2024-04-02:
Instead of fiddling with the label presentation, fix the uints.
Just write 50ns, it's much more meaningful then 50000ps in 5000ps increments

oharboe, 2024-04-02:
> Instead of fiddling with the label presentation, why not fix the units instead ? 50ns is shorter and more meaningful then 50000ps in 5000ps increments

I disagree. for a design where units are in ps, stay in ps.

The reason why you see 500000ps here is pathological and I dont accie

closed 2024-04-02 by oharboe