The-OpenROAD-Project/OpenROAD #4901: Change resize verbose output to debug info

oharboe, 2024-04-03:
### Description

OpenROAD resizing recently got more verbose, change this output to debug output or redirect into the reports folder.

```
$ cd flow
$ make DESIGN_CONFIG=./designs/nangate45/gcd/config.mk
[deleted]
OpenROAD fe4d260d96b372e70b1abbc43cb26c72923d196a 
[deleted]
[INFO RSZ-0094] Found 35 endpoints with setup violations.
Iteration | Resized | Buffers | Cloned Gates | Pin Swaps |   WNS   |   TNS   | Endpoint
---------------------------------------------------------------------------------------
        0 |       0 |       0 |            0 |         0 |  -0.162 |  -1.429 | resp_msg[15]
       10 |       9 |       0 |            0 |         0 |  -0.132 |  -1.024 | resp_msg[15]
       20 |      14 |       2 |            0 |         4 |  -0.124 |  -1.619 | resp_msg[15]
       30 |      17 |      10 |            0 |         7 |  -0.123 |  -1.719 | resp_msg[15]
       40 |      24 |      15 |            0 |         8 |  -0.113 |  -1.526 | resp_msg[15]
       50 |      30 |      15 |            0 |        12 |  -0.111 |  -1.684 | resp_msg[15]
       60 |      37 |      21 |            0 |        13 |  -0.105 |  -1.430 | resp_msg[15]
       64 |      39 |      21 |            0 |        14 |  -0.104 |  -1.427 | resp_msg[15]
[deleted]
```

### Suggested Solution

Change to debug output

gadfort, 2024-04-03:
@oharboe I added this a while ago and I would be open to suggestions on how to improve this, but I use it frequently, so I would be against dumping it to debug (as it's not a developer option) or to a report.

maliberty, 2024-04-03:
It gives a senses of progress as multiple people have reported 'hangs' that are not unclear if things are stuck or just slow.  I think it is help in the same way all progress output is helpful.

oharboe, 2024-04-03:
A bit overzealousness on my part in swatting down too verbose output. After the last weeks combing through the output for noise and removing it, the output is now generally either progress, reasonably succinct or necessary for a combination of developers/users.

closed 2024-04-03 by oharboe