The-OpenROAD-Project/OpenROAD #4902: make TNS_END_PERCENT=0 faster

oharboe, 2024-04-03:
### Description

TNS_END_PERCENT=0 can take a long time

```
[deleted]
repair_timing -verbose -repair_tns 0
[INFO RSZ-0094] Found 3360 endpoints with setup violations.
Iteration | Resized | Buffers | Cloned Gates | Pin Swaps |   WNS   |   TNS   | Endpoint
---------------------------------------------------------------------------------------
        0 |       0 |       0 |            0 |         0 | -9607.140 | -6816523.500 | io_pair_1_writePort_port_bits_data_bits[52]
       10 |       3 |       4 |            1 |         3 | -9367.296 | -6718992.500 | io_pair_0_writePort_port_bits_data_bits[61]
       20 |       6 |       9 |            3 |         6 | -9316.068 | -6693474.500 | io_pair_1_writePort_port_bits_data_bits[52]
       30 |      12 |      11 |            4 |         8 | -9298.443 | -6687491.500 | io_pair_0_writePort_port_bits_data_bits[61]
       40 |      16 |      11 |            9 |         9 | -9257.050 | -6658809.500 | io_pair_1_writePort_port_bits_data_bits[52]
       50 |      26 |      11 |            9 |         9 | -9245.098 | -6653363.000 | io_pair_1_writePort_port_bits_data_bits[52]
       60 |      33 |      11 |           11 |        10 | -9232.305 | -6627450.000 | io_pair_1_writePort_port_bits_data_bits[52]
       70 |      37 |      14 |           13 |        13 | -9228.910 | -6630023.000 | io_pair_1_writePort_port_bits_data_bits[52]
       80 |      44 |      14 |           16 |        13 | -9181.571 | -6611451.000 | io_pair_1_writePort_port_bits_data_bits[52]
       90 |      47 |      14 |           21 |        15 | -9116.970 | -6570721.000 | io_pair_1_writePort_port_bits_data_bits[52]
      100 |      56 |      14 |           22 |        15 | -9125.853 | -6571472.500 | io_pair_0_writePort_port_bits_data_bits[61]
      110 |      60 |      14 |           28 |        15 | -9081.704 | -6545675.000 | io_pair_1_writePort_port_bits_data_bits[52]
      120 |      66 |      14 |           30 |        17 | -9047.833 | -6537464.500 | io_pair_0_writePort_port_bits_data_bits[61]
      130 |      73 |      14 |           32 |        18 | -9042.597 | -6531245.500 | io_pair_1_writePort_port_bits_data_bits[52]
[deleted]
```

### Suggested Solution

Currently TNS_END_PERCENT repairs the only the worst path. Can it be changed to simply skip this stage?

### Additional Context

Skipping repair in CTS can be useful when doing early iterations and architectural exploration or getting things up and running in CI.

maliberty, 2024-04-04:
> Can it be changed to simply skip the repair stage?

You should skip the command entirely if that's what you want.  The point of the option is to allow repairing more paths, not to disable the normal resizer functionality or specify a certain number of paths to repair.

oharboe, 2024-04-04:
> > Can it be changed to simply skip the repair stage?
> 
> You should skip the command entirely if that's what you want. The point of the option is to allow repairing more paths, not to disable the normal resizer functionality or specify a certain number of paths to repair.

Would you entertain an ORFS pull request to check if TNS_END_PERCENT=0 and skip repair in that case?

maliberty, 2024-04-04:
If we did that then it would be impossible to pass zero to the command.  It seems it should be some other option entirely like SKIP_REPAIR_TIMING.

oharboe, 2024-04-04:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1912

oharboe, 2024-04-09:
@maliberty Close? 

I think this is addressed as well as I can think of by https://github.com/The-OpenROAD-Project/OpenROAD/pull/4926 and https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1912

closed 2024-04-09 by maliberty