The-OpenROAD-Project/OpenROAD #4905: Improve readability of report_clock_skew

oharboe, 2024-04-04:
### Description

Before running example, make sure this fix is in: https://github.com/The-OpenROAD-Project/OpenROAD/issues/4886#issuecomment-2036039360

Output from report_clock_skew is a bit hard to read:

```
>>> report_clock_skew
Clock clock
 337.78 source latency ces_4_7/clock ^
  78.36 source clock tree delay
-359.04 target latency io_lsbs_39$_DFF_P_/CLK ^
  -3.01 CRPR
--------------
  54.09 setup skew

Clock clock_vir
No launch/capture paths found.
```

[image]

The test case was produce with `make DESIGN_CONFIG=designs/asap7/mock-array/config.mk`

To launch GUI:

untar [gui.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/14831958/gui.tar.gz)

```
. vars-mock-array-asap7-base.sh
ODB_FILE=results/asap7/mock-array/base/5_route.odb ./run-me-mock-array-asap7-base.sh
OpenROAD v2.0-12845-gd71278409 
[deleted]
```

### Suggested Solution

- Could report_clock_skew simply report the arguments with which to run

maliberty, 2024-04-04:
> Could report_clock_skew simply report the arguments with which to run

With which to run what?

maliberty, 2024-04-04:
Note that what is maximum for setup may not be for hold so you would have to have some option for that.  I'll leave it to @tspyrou to decide what is practical here.  I am fairly sure no other tool reports skew in this way.