The-OpenROAD-Project/OpenROAD #4923: `DependencyInstaller.sh` touches system files

povik, 2024-04-08:
### Describe the bug

On Ubuntu, `DependencyInstaller.sh` seems to touch files installed by the system's package manager, even if you go the recommended route of `setup.sh` from `OpenROAD-flow-scripts` for a local install. I think it's reasonable to expect any dependency installer script to *not* do things like that.

### Expected Behavior

.

### Environment

```shell
N/A
```

### To Reproduce

Look into https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/etc/DependencyInstaller.sh#L292 where it stands:

    # need the strip "hack" above to run on docker
    strip --remove-section=.note.ABI-tag /usr/lib/x86_64-linux-gnu/libQt5Core.so

This runs for any Ubuntu installs, even local ones. This seems to be run from `OpenROAD-flow-scripts/setup.sh` too.

vvbandeira, 2024-04-09:
@habibayassin, could you move this `strip` command inside the Dockerfile/DockerHelper and make it conditional to run only on Ubuntu?

oharboe, 2025-02-03:
@vvbandeira Is this issue fixed?

closed 2025-02-06 by vvbandeira