The-OpenROAD-Project/OpenROAD #4937: Right click in inspector to articulate "report_checks -through" commands

oharboe, 2024-04-10:
### Description

Example use-case is to find out what it is that is causing congestion for non-experts:

- zoom in on an area of congestion
- pick some instances
- right click on some of the ITerms of this instance and then have a "report_checks -through _353_/ZN" injected into the command line.

[image]

### Suggested Solution

Right click on ITerms and have menu to select "report path", just like in the Timing Report you can right click on a setup/hold path to have a "report_checks" injected into the command line.

maliberty, 2024-04-10:
I'm curious what you wish to learn from a path through an iterm in an area with routing congestion.

oharboe, 2024-04-10:
> I'm curious what you wish to learn from a path through an iterm in an area with routing congestion.

I want to figure out where in Verilog and ultimately where in Chisel this is originating. Timing paths tell me this.

maliberty, 2024-04-10:
I think it is a right click menu type of thing

PR #4975 (2024-04-18): gui: add context menu to write report command based on ITerms in the inspector
Resolve #4937
closed 2024-04-18 by maliberty