The-OpenROAD-Project/OpenROAD #4948: Global route in megaboom has a single DRC error

oharboe, 2024-04-14:
### Description

A single DRC error seems odd... I would have expected no DRC errors in global routing or several.

[image]

To view DRC error:

untar 

```
. vars-BoomTile-asap7-base.sh
ODB_FILE=home/oyvind/.cache/bazel/_bazel_oyvind/7e6ad621f3f951c3ee6f5b179289b54e/execroot/_main/bazel-out/k8-fastbuild/bin/results/asap7/BoomTile/base/5_1_grt.odb openroad -gui scripts/gui.tcl
```

To reproduce:

```
./run-me-BoomTile-asap7-base.sh 
```

### Suggested Solution

Perhaps this is something that can be improved in global route, or perhaps it just happened to be a single global route DRC error and that the congestion is just very slightly too high?

maliberty, 2024-04-14:
I'm sure something can always be improved but I don't see one error as indicating a problem per se.  It could just be at the edge.

oharboe, 2024-04-14:
@maliberty Please close this feature request if you don't think it offers any value as an interesting test case.

stefanottili, 2024-04-14:
Trying to detail route this on M1 Mac this testcase coredumps in drt::FlexGR::initGR().

Something is off with regard to the M2 H Track-Pitch.
```
Generating GCell with size = 15 tracks, using layer M1 pitch  = 0.036
M1 V Track-Pitch = 0.03600  line-2-Via Pitch = 0.03600
M2 H Track-Pitch = 0.27000  line-2-Via Pitch = 0.03600
M3 V Track-Pitch = 0.03600  line-2-Via Pitch = 0.03600
...
Warning: Track pitch is too small compared with line-2-via pitch
```
```
read_db results/asap7/BoomTile/base/5_1_grt.odb
detailed_route
```
```
0   libunwind.dylib               	       0x1a05f4a7c libunwind::UnwindCursor<libunwind::LocalAddressSpace, libunwind::Registers_arm64>::step(bool) + 816
1   libunwind.dylib               	       0x1a05f7df0 _Unwind_Backtrace + 392
2   openroad                      	       0x1040c571c boost::stacktrace::detail::this_thread_frames::collect(void const**, unsigned long, unsigned long) + 56
3   openroad                      	       0x1040c556c boost::stacktrace::basic_stacktrace<std::__1::allocator<boost::stacktrace::frame>>::init(unsigned long, unsigned long) + 84
4   openroad                      	       0x1040c4208 handler(int) + 144
5   libsystem_platform.dylib      	       0x1939eb584 _sigtramp + 56
6   ???                           	0xffff8001047e9e4c ???
7   openroad                      	       0x1047e1834 drt::FlexGR::initGR() + 332
8   openroad                      	       0x1047e0850 drt::FlexGR::main(odb::dbDatabase*) + 80
9   openroad                      	       0x104a65214 drt::TritonRoute::main() + 1308
10  openroad                      	       0x1047d80f8 detailed_route_cmd(char const*, char const*, char const*, char const*, char const*, bool, int, char const*, char const*, int, double, char const*, char const*, int, bool, bool, bool, int, bool, char const*, int) + 1272
11  openroad                      	       0x1047da52c _wrap_detailed_route_cmd(void*, Tcl_Interp*, int, Tcl_Obj* const*) + 1592
12  libtcl8.6.dylib               	       0x108debe24 TclNRRunCallbacks + 80
13  libtcl8.6.dylib               	       0x108decba4 TclEvalEx + 1664
14  libtcl8.6.dylib               	       0x108ded514 Tcl_Eval + 40
15  openroad                      	       0x1046ca33c gui::TclCmdInputWidget::executeCommand(QString const&, bool, bool) + 292
16  openroad                      	       0x1046c443c gui::CmdInputWidget::handleEnterKeyPress(QKeyEvent*) + 480
17  openroad                      	       0x1046c613c gui::TclCmdInputWidget::keyPressEvent(QKeyEvent*) + 88
```
```
OpenROAD v2.0-13282-gee5d3738a 
Features included (+) or not (-):  +Charts +GPU +GUI +MPL2 +PAR +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 8 thread(s).

[WARNING GRT-0300] Timing is not available, setting critical nets percentage to 0.
[WARNING GUI-0066] Heat map "IR Drop" has not been populated with data.
[INFO DRT-0149] Reading tech and libs.

Units:                1000
Number of layers:     21
Number of macros:     231
Number of vias:       77
Number of viarulegen: 11

[INFO DRT-0150] Reading design.

Design:                   BoomTile
Die area:                 ( 0 0 ) ( 1398924 1398924 )
Number of track patterns: 32
Number of DEF vias:       0
Number of components:     1781981
Number of terminals:      388
Number of snets:          2
Number of nets:           1646600

[INFO DRT-0167] List of default vias:
  Layer V1
    default via: VIA12
  Layer V2
    default via: VIA23
  Layer V3
    default via: VIA34
[... 151 lines cut ...]

             Routing   #Avail     #Track     #Total     %Gcell
Layer      Direction    Track    Blocked      Gcell    Blocked
--------------------------------------------------------------
M1              V       38859          0     6708100     97.41%
M2              H       36268          0     6708100      9.30%
M3              V       38859          0     6708100      9.19%
M4              H       29144          0     6708100      9.30%
M5              V       29144          0     6708100      9.23%
M6              H       21858          0     6708100     14.92%
M7              V       21858          0     6708100      0.00%
M8              H       17485          0     6708100      0.00%
M9              V       17485          0     6708100      0.00%
Pad             H           0          0     6708100    100.00%
--------------------------------------------------------------
Total                  250960      0.00%    67081000     24.94%

[INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 6728.25 (MB), peak = 9397.91 (MB)

generating net topology...
done net topology...
Signal 11 received
Stack trace:
zsh: trace trap  openroad -gui -threads max
```

maliberty, 2024-04-15:
I looked briefly and the congestion all around here is at 100% so any extra wires will have to exceed the limit.  More interesting is the macro placement:
[image]

The macros down the middle are blocking all horizontal routing layers and forcing all the signals between them.  That is recipe for congestion.  Is this an mpl2 macro placement?

maliberty, 2024-04-15:
You can see all the congestion is in fitting through the macro gap:
[image]

maliberty, 2024-04-15:
> drt::FlexGR::initGR

This is the global routing in drt not the usual global router.  I don't suggest you use this.

oharboe, 2024-04-15:
@maliberty Yes, placed by mpl2

maliberty, 2024-04-15:
I think this would be more interesting as an mpl test case then.

maliberty, 2024-04-15:
Please make a version from that step

closed 2024-04-15 by oharboe

stefanottili, 2024-04-15:
@maliberty > This is the global routing in drt not the usual global router. I don't suggest you use this.

My assumption was that 5_1_grt.odb was already global routed because I can see the global route heat map after read_db.
Does a global routed odb not contain global route information and I have to run global route on it again ?

Seems like it. At least after global_route detailed_route seems to be running without coredump.

If the "global routing in drt" is not supposed to be used, it might be better to disable it and give the user a
"design not global routed yet" message instead of the current coredump.

maliberty, 2024-04-15:
The db should contain the guides.  I'm not sure what you are doing to trigger its use.

stefanottili, 2024-04-15:
I just used "read_db 5_1_grt.odb" from the testcase.

maliberty, 2024-04-15:
I wonder if this is somehow a mac issue.

oharboe, 2024-04-15:
@stefanottili Nice finds! Although this is not a practical problem for me now, I am sure that unfixed, it is just waiting for a more inoportune time to derail things.

stefanottili, 2024-04-16:
One final set of remarks:

I finally got around using wsl2 and installed ubuntu on a 64GB machine, giving wsl2 56GB to work with.
Which ran out of memory after completing 80% of the 0th detail route optimization claiming 54GB usage, but top showed that openbroad exhausted both 56GB memory and 14GB of swap.

1) @oharboe How much memory is needed to run this testcase all the way thru ?
2) @maliberty not loading the global route info from odb seems to be a mac issue, works on ubuntu.
3) on mac I f(it) and Z(oom out) don't work, works on ubuntu.

klayout shows that code can work the same on all 3 platforms, linux/mac/windows, so 2 and 3 are unfortunate.
But since it seems since I'm the only mac m1 user out there, it's not important.

oharboe, 2024-04-16:
@stefanottili I don't recall, but this PR should increase awareness of memory consumption. You become what you measure... https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1943

This is the machine I build on:

```
$ free -h
               total        used        free      shared  buff/cache   available
Mem:            62Gi        61Gi       564Mi       156Ki       1,0Gi       886Mi
Swap:          257Gi        37Gi       220Gi
```