The-OpenROAD-Project/OpenROAD #4951: improve macro placement in megaboom

oharboe, 2024-04-15:
### Description

The macro placement results are hard to route, ref https://github.com/The-OpenROAD-Project/OpenROAD/issues/4948

To reproduce:

untar https://drive.google.com/file/d/1Qc13d71MCpUMNGOWs1TLq-KYdBuNH2YK/view?usp=sharing

```
./run-me-BoomTile-asap7-base.sh
```

### Suggested Solution

Make macro placement even better

maliberty, 2024-04-15:
This is not the same placement shown in https://github.com/The-OpenROAD-Project/OpenROAD/issues/4948 - is this a different problem?

AcKoucher, 2024-04-15:
Also, the result we see on your image, is it also hard to route as the placement in #4948? It looks like there's a lot of space.

oharboe, 2024-04-15:
> This is not the same placement shown in #4948 - is this a different problem?

Global route works here. I could look at reproducing the exact conditions last time, but I thought this was a useful test case too.

oharboe, 2024-04-16:
will reopen new issue if I can reproduce it

closed 2024-04-16 by oharboe