The-OpenROAD-Project/OpenROAD #4960: [WARNING GUI-0076] QObject::killTimer: Timers cannot be stopped from another thread

oharboe, 2024-04-16:
### Describe the bug

untar https://drive.google.com/file/d/1ddbweLQ1Ev6LveKRxcjlQyqeRaAIm_St/view?usp=sharing

```
. vars-BoomTile-asap7-base.sh
```
```
ODB_FILE=home/oyvind/.cache/bazel/_bazel_oyvind/7e6ad621f3f951c3ee6f5b179289b54e/execroot/_main/bazel-out/k8-fastbuild/bin/results/asap7/BoomTile/base/2_floorplan.odb openroad -gui -no_init scripts/gui.tcl
```

Toggle routing congestion heatmap:

```
v2.0-13298-g9a9521c70
Features included (+) or not (-):  +Charts +GPU +GUI +MPL2 +PAR +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
[WARNING GUI-0066] Heat map "Routing Congestion" has not been populated with data.
[WARNING GUI-0076] QObject::killTimer: Timers cannot be stopped from another thread
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
[WARNING GUI-0066] Heat map "Routing Congestion" has not been populated with data.
[WARNING GUI-0076] QObject::killTimer: Timers cannot be stopped from another thread
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
[WARNING GUI-0066] Heat map "Routing Congestion" has not been populated with data.
[WARNING GUI-0076] QObject::killTimer: Timers cannot be stopped from another thread
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
[WARNING GUI-0066] Heat map "Routing Congestion" has not been populated with data.
[WARNING GUI-0076] QObject::killTimer: Timers cannot be stopped from another thread
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
```

### Expected Behavior

no warning

### Environment

```shell
see above
```

### To Reproduce

see above

AcKoucher, 2024-05-13:
I tried with the same version, but it looks like I can't reproduce.

[image]

The fix for #4806 should work here though.

oharboe, 2024-05-13:
not reproducible with OpenROAD v2.0-13763-g1c9637d0f

closed 2024-05-13 by oharboe