The-OpenROAD-Project/OpenROAD #4970: Create a decap insertion routine

rovinski, 2024-04-18:
OpenROAD doesn't currently have any automated method for inserting decap cells. It could provide a simple routine after static IR drop analysis to insert a user-specified amount of decap (in Liberty capacitance units) around instances with the highest static IR drop because there is a high correlation to the worst dynamic IR drop instances.

I can imagine a simple, greedy approach where decap cells are added within X um of the worst instances until the user-provided threshold is met. Note that decap most useful when it is connected to the power rail of the problematic instance (so same row or +/- 1 row depending on the row orientation).

> @maliberty usually decaps should be added after IR drop analysis and inserted next to instances with the worst IR drop. A loose spec off the top of my head would be:

* list of decap cells to use
* maximum amount of decap to insert (in liberty cap units), or
* voltage margin to fix up to
  * e.g. if worst IR drop is 0.15 V, and the user specifies 0.10 V, the tool would try to insert decap around all failing instances to ensure the worst IR drop is no more than 0.10 V

_Originally posted by @rovinski in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1945#issuecomment-2062955341_

QuantamHD, 2024-04-18:
What's the benefit over just using Decap cells as fill cells? I imagine there's a leakage benefit, but how much?

maliberty, 2024-04-18:
One challenge I see is that the amount of decap provided is not present in the lef or lib (at least in sky130hd).  Where is that typically found?

The amount of leakage is given
```
    cell ("sky130_fd_sc_hd__decap_12") {
        area : 15.014400000;
        cell_footprint : "sky130_fd_sc_hd__decap";
        cell_leakage_power : 0.0032400940;
```

maliberty, 2024-04-18:
I suppose you could read the cdl and spice models and try to figure it out but :anguished:

rovinski, 2024-04-19:
The leakage can be very significant in advanced nodes.

sergeiandreyev, 2024-05-14:
we have the decap values calculated internally for IHP SG13G2 OpenPDK:
|   | slow  | typ  | fast  |
|---|---|---|---|
| sg13g2_decap4 | 11.83 fF  | 12.41 fF  | 13.19 fF  |
| sg13g2_decap8 | 26.97 fF  | 28.6 fF  | 31 fF  |  

please propose a place to store these..
p.s. if I remember correctly, I saw them as a one-line comment in Liberty models from one big technology vendor

maliberty, 2024-05-14:
I assume nobody is taking it from a comment so I still don't know where to store it.  @rovinski or @gadfort any ideas?

maliberty, 2024-05-14:
I suspect it would have to be a command argument as I don't see anything to describe it in Liberty or LEF.

maliberty, 2024-05-14:
I'm thinking something like
```
insert_decap -cap <target> -cells <cell1> <decap_of_cell1> ....
```

A first version could just spread them evenly and later versions could consider IR drop.

maliberty, 2024-05-14:
Having a `-bbox <area>` would allow for manual targeting of the cells.

QuantamHD, 2024-05-14:
Why not just extend liberty to add a new field?

QuantamHD, 2024-05-14:
Or read it out of a lef property

gadfort, 2024-05-14:
you could detect the cells automatically with `is_decap_cell : "true";` from the liberty file.
I don't know where they keep the capacitance information, it's in a commented out block in the PDKs I checked.
As a bypass you could allow the user to set the capacitance per cell (like we do with parasitics).

gadfort, 2024-05-14:
@QuantamHD @maliberty I second the not having to modify the PDK (this would be bad practice).

maliberty, 2024-05-14:
> As a bypass you could allow the user to set the capacitance per cell (like we do with parasitics).

That's essentially what I proposed with `-cells <cell1> <decap_of_cell1>`

gadfort, 2024-05-14:
> > As a bypass you could allow the user to set the capacitance per cell (like we do with parasitics).
> 
> That's essentially what I proposed with `-cells <cell1> <decap_of_cell1>`

Ohh, I missed that. Seems like a reasonable method.

QuantamHD, 2024-05-14:
Yeah, but if the IHP people want to add a property to their LEF that would make setup easier. I think the command is also useful

maliberty, 2024-05-14:
Let's start with the command as that will cover all platforms.   PDK setup is being done by IHP already so I expect it isn't a big help to move the data to Liberty but I'm open.

rovinski, 2024-05-15:
I just checked one of my old projects and decap was added with the following interface:

* decap cell names
* manually specified capacitances in lib units (not sure where I got this number from in the PDK)
* total amount of decap to add to the design

I don't think Lib or LEF have a formal way of specifying the decap capacitance. In fact, it doesn't look like Lib allows custom attributes? If one wanted to specify this in the PDK files, the only way would be through a custom LEF property.

maliberty, 2024-05-15:
You can use `define_group` to add attributes in Liberty

sergeiandreyev, 2024-05-15:
yes, this is the syntax (taken from [here](https://people.eecs.berkeley.edu/~alanmi/publications/other/liberty07_03.pdf)):
[image]

PR #5316 (2024-07-18): Dpl: Inserting decap cells with the insert_decap command
Creating the `insert_decap` command to insert decap cells in the areas with the highest IR DROP.
To use the command you must first execute `analyze_power_grid` for the net VDD, to obtain the areas with the highest IR DROP.
Fixes #4970
closed 2024-07-18 by eder-matheus