The-OpenROAD-Project/OpenROAD #4973: Add "Logic delay" column to Timing Report

oharboe, 2024-04-18:
### Description

It is useful to see the logic delay of a path in the Timing Report when doing triage in architectural exploration.

### Suggested Solution

Add "Logic delay" column to Timing Report

maliberty, 2024-04-18:
This new column would be the sum of the non-buffer gate delays without any wire delay.

maliberty, 2024-04-18:
The idea is to give an indication of paths that have too much logic and probably need an RTL-modification solution.

maliberty, 2024-04-18:
Replaces / simplifies https://github.com/The-OpenROAD-Project/OpenROAD/issues/4832

maliberty, 2024-04-19:
It is the sum of the delays.  I think of logic depth as the number of gates.

maliberty, 2024-04-19:
It does make me think that would also be a good column.  I think we might need to implement column visibility to prevent this from become a super wide table (eg https://ux.stackexchange.com/questions/110077/best-practices-to-allow-user-to-hide-show-columns-in-a-data-table)

maliberty, 2024-04-19:
@oharboe any preference on delay vs depth?

oharboe, 2024-04-19:
> @oharboe any preference on delay vs depth?

I would want to know both. I believe logic depth might be a bit less standard on ASAP7? Isn't ASAP7 missing some things like muxes as primitives? How does a mux count as "levels of logic"? I'm not sure if a mux primitive in the PDK would be enormously much faster than something that is built up of even smaller primitivies. Out of my depth here... Would appreciate some input and guidance...

oharboe, 2024-04-19:
@rovinski Thanks! This makes sense. It sounds like this is complimentary information that helps to create a more fleshed out understanding of what is going on...

maliberty, 2024-04-20:
What is the conclusion in terms of this request?

oharboe, 2024-04-20:
Add both columns. Later on, if there are even more columns, maybe the userinterface should change to  enable/disable columns.

today, startpoint is show, perhaps endpoint column?

maliberty, 2024-04-20:
> today, startpoint is show, perhaps endpoint column?

I don't know what this means

oharboe, 2024-04-20:
> > today, startpoint is show, perhaps endpoint column?
> 
> I don't know what this means

There is a Start column today, and I've highlighted what would be the "End" column in Data Path Details:

[image]

maliberty, 2024-04-20:
So this is a totally unrelated request?

oharboe, 2024-04-20:
> So this is a totally unrelated request?

Yes. I would leave it be for now.

It is just an example of why I think your consideration of what to do if many columns are added makes sense: https://github.com/The-OpenROAD-Project/OpenROAD/issues/4973#issuecomment-2065614093

AcKoucher, 2024-05-03:
@maliberty @oharboe So to be clear, the request here is:

1. Add "Logic Delay" column;
2. Add "Logic Depth" column;
3. Column Visibility.

Is that it?

maliberty, 2024-05-03:
Yes

PR #5072 (2024-05-12): gui: add logic depth, logic delay and implement column visibility in timing report
Resolve #4973 

Add to Timing Report:
- Column Visibility control
- Logic Delay
- Logic Depth 

*Also fix crash when right clicking timing report if no odb is loaded.
closed 2024-05-12 by maliberty