The-OpenROAD-Project/OpenROAD #4979: OR not honouring AREA from techlef in route

hpretl, 2024-04-21:
### Describe the bug

I create a layout with ORFS using platform SG13G2. In the TECHLEF the AREA parameter is set for Metal2, however, when I create a layout I have DRC errors on Metal2 due to metal area too small.

In`sg13g2_tech.lef` the following is contained (I have manually added the MINSIZE setting to test, but it makes no difference):
```
LAYER Metal2
  TYPE		ROUTING ;
  DIRECTION	HORIZONTAL ;
  PITCH		0.42 ;
  OFFSET	0.0 ;
  WIDTH		0.20 ;
  MAXWIDTH	30 ;
  MINIMUMDENSITY        35.0 ;
  MAXIMUMDENSITY        60.0 ;
  DENSITYCHECKSTEP 100 ;
  DENSITYCHECKWINDOW 200 200 ;
  SPACINGTABLE
  PARALLELRUNLENGTH 0.00    1.00    10.00
  WIDTH 0.00        0.21    0.21    0.21
  WIDTH 0.39        0.21    0.24    0.24
  WIDTH 10.0        0.21    0.24    0.60 ;
  MINIMUMCUT 2 WIDTH 1.4 ;
  AREA      0.144 ;
  MINSIZE 0.2 0.72 ;
  HEIGHT 1.880 ;
  #CURRENTDEN 0 ;
  THICKNESS 0.450 ;
  ANTENNACUMAREARATIO 200 ;
  ANTENNACUMDIFFAREARATIO PWL ( ( 0 200 ) ( 0.159 200 ) ( 0.16  3200 ) ( 100 2000000 ) ) ;
  WIREEXTENSION 0.10 ;
  RESISTANCE RPERSQ 0.103 ;
  CAPACITANCE  CPERSQDIST 1.81E-05 ;
  EDGECAPACITANCE  4.47E-05 ;
  DCCURRENTDENSITY AVERAGE 2 ;
END Metal2
```

### Expected Behavior

No Metal2 structures smaller than specificed by `AREA` should be present in the layout (but it is).

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.6.22-linuxkit
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.22.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-12608-g50753a4c5
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
[... 115 lines cut ...]
-- Found OpenMP: TRUE (found version "4.5")  
-- GPU is not enabled
-- TCL library: /usr/lib/aarch64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found version "1.82.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found suitable version "1.82.0", minimum required is "1.78")  
-- TCL library: /usr/lib/aarch64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found version "1.82.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found version "1.82.0")  
-- Found Eigen3: /usr/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.d7cBdqDSgl
```

### To Reproduce

Build one of the stock examples for SG13G2 in ORFS.

hpretl, 2024-04-21:
Maybe this is related, not sure: https://github.com/The-OpenROAD-Project/OpenROAD/pull/4419

maliberty, 2024-04-21:
How are you checking the DRCs?

maliberty, 2024-04-21:
Is it possible to provide the test case?

hpretl, 2024-04-22:
You can use the stock example `gcd` of ORFS by setting in the `Makefile`:
```
DESIGN_CONFIG=./designs/ihp-sg13g2/gcd/config.mk
```

In the resulting `6_final.gds` you can find at x=62.664um/y=138.08um a `Metal2` rectangle of size 0.675um x 0.2um, clearly violating min. area of 0.144um2.

[image]

[6_final.gds.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/15061157/6_final.gds.gz)

[6_final.odb.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/15061159/6_final.odb.gz)

hpretl, 2024-04-22:
I have checked the DRC of my GDS files using the closed-source version of the SG13G2 PDK and a commercial tool. The open-source KLayout DRC deck I did not use.

KrzysztofHerman, 2024-04-25:
Hi All, 
I have reproduced it locally and these are my observations:
- there is nothing at the location `x=62.664um/y=138.08um` 
- I have no DRC's related to min area on metal2 using DRC rule deck of Klayout (maybe not implemented yet)
- I found some Metal2 strips shown on the picture where the metallization is made of two rectangles and the total size fulfils area rule `0.144 = 0.72*0.2`

[image]

hpretl, 2024-04-25:
Interesting, maybe you use a newer OpenROAD version and the issue has been solved meanwhile? @KrzysztofHerman which OR version are you using?

KrzysztofHerman, 2024-04-25:
`openroad -v` gives me this:
`ab5b1ab82e84a85c7d42fb5018de5dbb17d45af8`
Nevertheless I recently have re-bulit the ORFS so the version should be a brand new one.

KrzysztofHerman, 2024-04-25:
Please find attached the `6_final.gds.gz` 
[6_final.gds.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/15105915/6_final.gds.gz)

hpretl, 2024-04-25:
Version `ab5b1ab82e84a85c7d42fb5018de5dbb17d45af8` is a commit from ca. 2024-04-11.

I have version `0889970d1790a2617e69f253221b8bd7626e51dc`, which is from 2024-02-18, so quite a bit older.

So far the theory holds that there was an issue which has been solved meanwhile. I will try a newer build and see how it goes.

osamahammad21, 2024-04-28:
@hpretl Could you check with the latest master? A recent update that addresses this issue was merged.

hpretl, 2024-04-29:
@osamahammad21 I tested the latest OR release. The situation is much better, but there are still a few min.area violations left.

osamahammad21, 2024-05-19:
There was another bug that led to ignoring some violations related to patching. This is fixed now. @hpretl You should give it a try now, I feel it will be solved.
Since there is no reply on this issue for more than 2 weeks, I will close it now. Feel free to re-open if you still find the same issue not solved.

closed 2024-05-19 by osamahammad21