The-OpenROAD-Project/OpenROAD #4983: Unoptimized setup paths in GCD with nangate45 run

pankajmudgil15, 2024-04-22:
### Description

Cells could have been upsized to reduce the cell delay. There are multiple cell with X1 drive strength which are taking higher cell delay. This is leading to setup violation.

### Suggested Solution

Cell driving strength could have been improved to X4 to meet the setup.

maliberty, 2024-04-22:
Have you done an experiment that shows making a specific cell larger actually improves timing?  If the wire is short making the driver large may be no help or even a degradation.  Please provide a specific example of a missed opportunity.

pankajmudgil15, 2024-05-13:
I haven't done the experiment, but found a few places for improvement. Like in attached png, Cell name 352 is being driven by X1 drive strength, and it's slew and delay are comparatively higher than others. Also, clock port can be placed somewhere in middle of left edge, so that clock network delay will be less due to lesser manhatten distance to flop.

Also, C2Q delay of flop also will improve once input clock transition improves.

[image]

maliberty, 2024-06-10:
The problem is that you didn't attach a test case or provide any version information so your results are not reproducible.  I looked at one X1 driver in my current results and manually upsized it.  It made the timing worse as the increase in pin cap caused the input net to slow down more than the increase in drive strength helped the output net.

If you wish to provide a case with a demonstrated benefit that was missed the feel free to reopen, otherwise this is unactionable.

closed 2024-06-10 by maliberty