The-OpenROAD-Project/OpenROAD #4990: Crash in timing report

oharboe, 2024-04-24:
### Describe the bug

1. untar https://drive.google.com/file/d/1lDdIysDfIFES_vrCKpRbrqg5Ie4ebnEw/view?usp=sharing
2. Run: ```
ODB_FILE=results/asap7/mock-array_Element/base/6_final.odb ./run-me-mock-array_Element-asap7-base.sh```
3. Create timing report
4. Select first timing path and first line in data path details, see below.
5. Use keyboard arrows to move/up down until you get a crash

[image]

```
QSocketNotifier: Can only be used with threads started with QThread
OpenROAD v2.0-13378-g6bc228c51 
Features included (+) or not (-):  +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0450] virtual clock clock_vir can not be propagated.
Loading spef
Signal 11 received
Stack trace:
 0# 0x000062BFFC344BB3 in openroad
 1# 0x000075C099A42990 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbNet::getSigType() in openroad
 3# gui::DbNetDescriptor::highlight(std::any, gui::Painter&) const in openroad
 4# gui::TimingPathRenderer::highlightStage(gui::Painter&, gui::Descriptor const*, gui::Descriptor const*) in openroad
 5# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in openroad
 6# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in openroad
 7# gui::RenderThread::run() in openroad
 8# 0x000075C09AADAEB1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
 9# 0x000075C099A97B5A in /lib/x86_64-linux-gnu/libc.so.6
10# 0x000075C099B285FC in /lib/x86_64-linux-gnu/libc.so.6
./run-me-mock-array_Element-asap7-base.sh: line 7: 13940 Segmentation fault      (core dumped) openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl
```

### Expected Behavior

No crash.

### Environment

```shell
OpenROAD v2.0-13378-g6bc228c51
```

### To Reproduce

See above. It takes multiple rounds of navigating with kebyoard arrows to top/bottom, say 10x all the way to the bottom and back up?

maliberty, 2024-04-24:
@AcKoucher please see if you can reproduce this.

AcKoucher, 2024-05-13:
@maliberty @oharboe  I can't reproduce this.

oharboe, 2024-05-13:
> @maliberty @oharboe I can't reproduce this.

What to do now? I dont think it is fixed... Interactive debug session? When would be a good time?

@maliberty Can you reproduce it?

maliberty, 2024-05-13:
No I can't reproduce it either.  Tomorrow 9am?

oharboe, 2024-05-13:
Can't reproduce on OpenROAD v2.0-13763-g1c9637d0f

oharboe, 2024-05-13:
@maliberty @AcKoucher Was anything done to fix this?

Did you try with v2.0-13378-g6bc228c51?

oharboe, 2024-05-13:
@AcKoucher @maliberty My concern is that this crash is still there and by using the latest version, initial conditions were changed and this caused the crash to no longer appear, even if the problem is still there.

oharboe, 2024-05-13:
I can't reproduce it even with OpenROAD v2.0-13378-g6bc228c51  on my home workstation.

oharboe, 2024-05-13:
@maliberty I imagine that timing is critical to reproduce the problem... I tried varying zoom levels, no luck in reproducing.

maliberty, 2024-05-13:
With no way to reproduce it there is no real next step to take.

oharboe, 2024-05-14:
- disable power and ground nets.

Crashes on my laptop:

```
Signal 11 received
Stack trace:
 0# 0x00005BF9239D94A3 in openroad
 1# 0x000076A76AE42990 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbNet::getSigType() in openroad
 3# gui::DbNetDescriptor::highlight(std::any, gui::Painter&) const in openroad
 4# gui::TimingPathRenderer::highlightStage(gui::Painter&, gui::Descriptor const*, gui::Descriptor const*) in openroad
 5# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in openroad
 6# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in openroad
 7# gui::RenderThread::run() in openroad
 8# 0x000076A76BEDAEB1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
 9# 0x000076A76AE97B5A in /lib/x86_64-linux-gnu/libc.so.6
10# 0x000076A76AF285FC in /lib/x86_64-linux-gnu/libc.so.6
./run-me-mock-array_Element-asap7-base.sh: line 7:  4761 Segmentation fault      (core dumped) openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl

```

oharboe, 2024-05-14:
Crashes with latest master too:

```
OpenROAD v2.0-13770-g98c6501d9 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0450] virtual clock clock_vir can not be propagated.
read_spef ./results/asap7/mock-array_Element/base/6_final.spef
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
[WARNING GUI-0076] Wayland does not support QWindow::requestActivate()
[WARNING GUI-0076] Wayland does not support QWindow::requestActivate()
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
[WARNING GUI-0066] Heat map "IR Drop" has not been populated with data.
[WARNING GUI-0076] QObject::killTimer: Timers cannot be stopped from another thread
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
[WARNING GUI-0076] QObject::killTimer: Timers cannot be stopped from another thread
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
[WARNING GUI-0076] QObject::killTimer: Timers cannot be stopped from another thread
[deleted]
[WARNING GUI-0076] QObject::startTimer: Timers cannot be started from another thread
Signal 11 received
Stack trace:
 0# 0x00005D2C048C06E3 in openroad
 1# 0x00007907ADE42990 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbNet::getITerms() in openroad
 3# gui::DbNetDescriptor::highlight(std::any, gui::Painter&) const in openroad
 4# gui::TimingPathRenderer::highlightStage(gui::Painter&, gui::Descriptor const*, gui::Descriptor const*) in openroad
 5# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in openroad
 6# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in openroad
 7# gui::RenderThread::run() in openroad
 8# 0x00007907AEEDAEB1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
 9# 0x00007907ADE97B5A in /lib/x86_64-linux-gnu/libc.so.6
10# 0x00007907ADF285FC in /lib/x86_64-linux-gnu/libc.so.6
./run-me-mock-array_Element-asap7-base.sh: line 7: 10378 Segmentation fault      (core dumped) openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl
```

oharboe, 2024-05-14:
To reproduce, disable as much rendering as possible so that the yellow updates quickly:

[image]

oharboe, 2024-05-14:
I can only make it crash with:

```
./build_openroad.sh --local --latest
```

If I try:

```
./build_openroad.sh --local --latest --openroad-args -DCMAKE_BUILD_TYPE=RelWithDebug
```

No crash...

oharboe, 2024-05-14:
I saw this segfault in the debugger when I clicked Timing Report -> Update:

[image]

Perhaps not related though?

```
$ ODB_FILE=home/oyvind/.cache/bazel/_bazel_oyvind/7e6ad621f3f951c3ee6f5b179289b54e/execroot/_main/bazel-out/k8-fastbuild/bin/results/asap7/BoomTile/base/2_floorplan.odb openroad -gui -no_init scripts/gui.tcl
OpenROAD v2.0-13770-g98c6501d9 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR GUI-0070] Error: gui.tcl, 2 can't read "::env(SCRIPTS_DIR)": no such variable
Signal 11 received
Stack trace:
 0# 0x00005766148776E3 in openroad
 1# 0x0000722C03C42990 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbBlock::getBTerms() in openroad
 3# sta::dbNetwork::pinIterator(sta::Instance const*) const in openroad
 4# sta::Graph::makePinVertices(sta::Instance const*) in openroad
 5# sta::Graph::makeGraph() in openroad
 6# sta::Sta::ensureGraph() in openroad
 7# gui::STAGuiInterface::initSTA() const in openroad
 8# gui::STAGuiInterface::getTimingPaths(std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&, std::vector<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> >, std::allocator<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > > > const&, std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&) const in openroad
 9# gui::TimingPathsModel::populatePaths(std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&, std::vector<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> >, std::allocator<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > > > const&, std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&) in openroad
10# gui::TimingPathsModel::populateModel(std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&, std::vector<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> >, std::allocator<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > > > const&, std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&) in openroad
11# gui::TimingWidget::populateAndSortModels(std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&, std::vector<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> >, std::allocator<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > > > const&, std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&) in openroad
12# gui::TimingWidget::populatePaths() in openroad
13# 0x0000722C04F06312 in /lib/x86_64-linux-gnu/libQt5Core.so.5
14# QAbstractButton::clicked(bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
15# 0x0000722C05C5FE0E in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
16# 0x0000722C05C619EB in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
17# QAbstractButton::mouseReleaseEvent(QMouseEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
19# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
20# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
21# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
22# QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
23# 0x0000722C05BC7DA1 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
24# 0x0000722C05BCB12F in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
25# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
26# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
27# QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
28# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
29# 0x0000722BFFEF9F7E in /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
30# 0x0000722C03F13B2C in /lib/x86_64-linux-gnu/libglib-2.0.so.0
31# 0x0000722C03F6F46F in /lib/x86_64-linux-gnu/libglib-2.0.so.0
32# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
33# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
34# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
35# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
36# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
39# main in openroad
40# 0x0000722C03C28150 in /lib/x86_64-linux-gnu/libc.so.6
41# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
42# _start in openroad
Segmentation fault (core dumped)
```

oharboe, 2024-05-14:
@AcKoucher @maliberty Matt was able to make it crash *once* on his end and I can make it happen fairly regularly. If I use the mouse to highlight up and down across the REG lines, it seems to happen fairly consistently on my machine.

[image]

oharboe, 2024-05-14:
It is possible to get the crash in the debugger with sybmols:

[image]

oharboe, 2024-05-14:
@maliberty So how can `highlight` be nullptr on line 589, when it clearly wasn't nullptr in line 583?

oharboe, 2024-05-14:
@maliberty @AcKoucher Looks like the rendering thread is using the same list that is being manipulated by the main thread and that the "swap" thing is not being done for `highlight_stage_`?

If the list is cleared from the main thread, then I can see that a null pointer exception happening.

[image]

maliberty, 2024-05-14:
threading most likely

maliberty, 2024-05-14:
What is the "swap thing"?

oharboe, 2024-05-14:
> What is the "swap thing"?

See screenshot above from the render loop.

PR #5106 (2024-05-15): gui: lock the updates to the TimingPathRenderer's fields
The main thread could get a key/mouse event that updates the renderer while it is drawing leading to a crash.  This provides locking to avoid this.

Fixes #4990
commit 95e33745 (2024-05-15): gui: lock the updates to the TimingPathRenderer's fields

The main thread could get a key/mouse event that updates the renderer while
it is drawing leading to a crash.  This provides locking to avoid this.

Fixes #4990
closed 2024-05-15 by maliberty