The-OpenROAD-Project/OpenROAD #4993: ORFS is using `report_cell_usage` which my OpenROAD is missing

hpretl, 2024-04-25:
### Describe the bug

As stated above, the ORFS `final_report.tcl` is using `report_cell_usage`, but my build of OR does not have this command. What am I doing wrong?

### Expected Behavior

Either the command `report_cell_usage` should not be used, or it should be available in OR.

### Environment

```shell
Git commit: 391c9dfb8906659b9cc782fafd3a3835a069ca06
kernel: Linux 6.6.22-linuxkit
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.22.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-13379-g391c9dfb8
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
[... 80 lines cut ...]
-- GPU is not enabled
-- TCL library: /usr/lib/aarch64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found version "1.82.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found suitable version "1.82.0", minimum required is "1.78")  
-- TCL library: /usr/lib/aarch64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found version "1.82.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.82.0/BNumber of processor cores: 10
oostConfig.cmake (found version "1.82.0")  
-- Found Eigen3: /usr/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.rUq28pVGL2
```

### To Reproduce

Start OR in interactive mode, check for command `report_cell_usage`.

hpretl, 2024-04-25:
[image]
 Very weird. I have no idea what could have gone wrong...

hpretl, 2024-04-25:
And when I type `report_cell_usage` in OpenROAD, I get this:
[image]

hpretl, 2024-04-25:
OK, now I think I have the full picture. @maliberty I think there might still be something to fix here, although something different that the original case:

- In the new OpenROAD version that I have (see above), the command `report_cell_usage` is indeed there, and it works normally in the ORFS.
- When `openroad` is started, and no design is loaded, when calling `report_cell_usage` a segfault occurs. That is not ideal.
- When `openroad` is run interactively, the command completion typing `report_` and then tab does not show `report_cell_usage` (while bringing up other commands, see above), which is also not ideal, and caused my confusion in the first place.

rovinski, 2024-04-25:
Sounds like 2 separate issues. `report_cell_usage` needs to 1) check that a design is loaded and 2) be included in the Tcl help.

maliberty, 2024-04-28:
I've fixed the crash in in #5010.  Fixing the auto complete looks a harder (I hate tcl).  I'll see if I can find a solution for that as it affects all the sta commands.  Apparently tclreadline doesn't understand 'namespace import'

hpretl, 2024-05-01:
@maliberty BTW, I found meanwhile another OR command that segfaulted when no design is loaded. Unfortunately, I forgot which one. Maybe all commands should be tested for running directly after OR start w/lo loading a design, that they exit gracefully instead of segfaulting. Could/should be added to the regression suit, I guess.

maliberty, 2024-05-01:
Good idea, I'll add one (and I suspect fix a lot of issues).

maliberty, 2024-05-01:
The hall of shame
```
  cluster_flops
  check_placement
  repair_antennas
  triton_part_design
  optimize_mirroring
  evaluate_part_design_solution
  place_endcaps
  highlight_path
  remove_fillers
  check_antennas
  global_placement_debug
  improve_placement
  pdngen
  insert_dft
  set_driving_cell
  tapcell
  tapcell_ripup
  preview_dft
  define_corners
```

hpretl, 2024-05-01:
Oh yes, the `repair_antennas` was the one I found :-)

maliberty, 2024-05-02:
#5032 adds the test but doesn't actually fix the problem yet.  The 'skip' list there needs to be reduced down to just 'exit' (which shouldn't be tested).

maliberty, 2024-05-02:
@arthurjolo please take care of fixing up the commands.

arthurjolo, 2024-05-02:
Sure, will fix them.

closed 2024-05-02 by maliberty

maliberty, 2024-05-02:
All are fixed except define_corners which will require an sta fix/update cycle.