The-OpenROAD-Project/OpenROAD #5057: Incorrect command documentation

nmoroze, 2024-05-07:
### Describe the bug

I've noticed the help strings for several commands are incomplete/incorrect. 

- `add_global_connection` (missing many options)
- `add_pdn_ring` (missing many options)
- `check_antennas` (missing `-report_file`)
- `report_checks` missing (`-fields`, possibly others)
- `with_output_to_variable` (missing positional arguments)

I know some of these commands are defined in OpenSTA, I can open an issue there for those commands if more appropriate.

### Expected Behavior

I'd expect running `help <command>` to display a help string that reflects all options available for a given command. 

### Environment

```shell
Git commit: 6fc686431fbb0dadf024120f912b28d982b899d5
kernel: Linux 5.10.76-linuxkit
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-13652-g6fc686431
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
[... 85 lines cut ...]
-- GPU is not enabled
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread
-- Found BoNumber of processor cores: 4
ost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.zZPFxRs101
```

### To Reproduce

Run `help <command>` in the OpenROAD Tcl shell, and compare the output to the implementation.

maliberty, 2024-05-07:
It is better to file the sta related ones to https://github.com/parallaxsw/OpenSTA

maliberty, 2024-05-07:
@luarss is there a way to check for such missing flags in your docs setup?

luarss, 2024-05-07:
@maliberty There is, but the unit tests/CI are disabled for now due to some bugs.

nmoroze, 2024-05-07:
@luarss I'm curious where these bugs come from. [tclint](https://github.com/nmoroze/tclint) (which I maintain) includes a Tcl parser Python library that I intend to have a stable, documented API for use outside of tclint itself. I noticed the docs tests involve extracting info from Tcl source files via regexp. If that part has been finicky, perhaps using tclint's parser might help? It's fairly robust, and I'd be willing to fast-track a clean-up pass and documentation of the API to facilitate this.

I additionally have some code written (but not yet published) that parses the OpenROAD help strings themselves. I think that piece could also be helpful in putting together automated checks of Tcl-to-docs correspondence, although it's not nearly as close to stable as the Tcl parser.

For context, I'm working on adding a feature to tclint, via a plugin, that allows it to check for correct usage of OpenROAD-specific commands. It does this by parsing the help output of a user's installed OpenROAD (in order to determine the API of the exact version they have installed). I caught these issues while debugging this feature, and this also gives me an interest in these docs being accurate, beyond it being generally helpful for the project :)

closed 2024-05-10 by eder-matheus

luarss, 2024-05-23:
> @luarss I'm curious where these bugs come from. [tclint](https://github.com/nmoroze/tclint) (which I maintain) includes a Tcl parser Python library that I intend to have a stable, documented API for [...]

@nmoroze Sorry for the late reply. I can share with you some of the errors I have faced while building the manpages CI in private. I'd also love to find out more about `tclint`, as currently we enforce some restrictions on the writing of Tcl syntax to make it detectable by our regexp. For starters, the Tcl commands in our codebase denoted with `;# checker off` are usually corner cases our parser cannot catch. Was curious if your framework could catch these?

Examples of Tcl format: https://openroad.readthedocs.io/en/latest/contrib/TclFormat.html

nmoroze, 2024-05-26:
> For starters, the Tcl commands in our codebase denoted with ;# checker off are usually corner cases our parser cannot catch. Was curious if your framework could catch these?

Definitely! tclint's parser includes comments in the returned syntax tree (tclint uses comments to disable lint violations inline, so it does a similar thing to what you're trying to achieve).

Here's a very simple example that prints the location of "checker off" comments in a script:

```python
from tclint.parser import Parser
from tclint.syntax_tree import Visitor

class CommentVisitor(Visitor):
    def visit_comment(self, comment):
        if comment.value.strip() == "checker off":
            print(f"found 'checker off' at {comment.pos}")

script_text = r"""sta::define_cmd_args "check_antennas" { [-verbose]\
                                          [-net net]} ;# checker off"""

parser = Parser()
script = parser.parse(script_text)

visitor = CommentVisitor()
script.accept(visitor, recurse=True)
```

I'm happy to chat more! Feel free to shoot me an [email](mailto:noahmoroze@gmail.com) if you'd like to follow-up in private.