The-OpenROAD-Project/OpenROAD #5083: Add tooltips to Skew column

oharboe, 2024-05-10:
### Description

Add a tooltip to explain in a sentence or two exactly what the Skew column is displaying.

[image]

See also https://github.com/The-OpenROAD-Project/OpenROAD/issues/5002 that has been following some bugfixes in Skew column.

### Suggested Solution

Add a terse exact definition of the Skew column in tooltips.

Using this definition, it should be possible to corroborate that information in the Skew column with report_clock_skew or other sources in mock-array and mock-array/Element from within the OpenROAD GUI.

maliberty, 2024-05-11:
Do you have some text in mind?

oharboe, 2024-05-11:
> Do you have some text in mind?

@tspyrou ?

AcKoucher, 2024-06-19:
Any ideas here?

tspyrou, 2024-06-19:
How about " path clock skew crpr corrected". Is that too long?

oharboe, 2024-06-19:
> How about " path clock skew crpr corrected". Is that too long?

Looks good. What is the equivalent OpenSTA syntax to reproduce this skew number?

AcKoucher, 2024-06-20:
I'll also add the tooltips for the other new columns.

@maliberty Would it be consistent to say:

Logic Delay: Non-buffer gate delays with no wire delay.
Logic Depth: Number of instances apart from buffers and back-to-back inverters.

*One thing I was thinking is that we're also counting registers in the logic depth. Would that be concise?

@oharboe Any opinions on these being too long?

oharboe, 2024-06-20:
Looks good!

PR #5265 (2024-06-26): gui: add tooltips to Timing Report
Resolve #5083 

Adding tool tips to logic delay/depth and skew.
closed 2024-06-26 by maliberty