The-OpenROAD-Project/OpenROAD #5085: global route fails and there's no congestion.rpt

oharboe, 2024-05-10:
### Describe the bug

Unclear why this global route fails.

untar https://drive.google.com/file/d/1ISEUv6mBhmc2FQAYpRdz4Hj35kae0odG/view?usp=sharing

Run:

```
$ ./run-me-L1MetadataArray-asap7-test.sh 
OpenROAD v2.0-13652-g6fc686431 
[deleted]
Repair setup and hold violations...
[INFO RSZ-0094] Found 2 endpoints with setup violations.
Iteration | Resized | Buffers | Cloned Gates | Pin Swaps |   WNS   |   TNS   | Endpoint
---------------------------------------------------------------------------------------
        0 |       0 |       0 |            0 |         0 | -30.120 | -56.365 | io_read_ready
        4 |       0 |       3 |            0 |         2 |   2.896 |   0.000 | tag_array_ext/R0_en
    final |       0 |       3 |            0 |         2 |   2.896 |   0.000 | tag_array_ext/R0_en
---------------------------------------------------------------------------------------
[INFO RSZ-0040] Inserted 2 buffers.
[INFO RSZ-0043] Swapped pins on 2 instances.
[INFO RSZ-0046] Found 199 endpoints with hold violations.
Iteration | Resized | Buffers | Cloned Gates |   WNS   |   TNS   | Endpoint
---------------------------------------------------------------------------
        0 |       0 |       0 |            0 | -376.251 | -37698.145 | tag_array_ext/W0_addr[4]
[ERROR GRT-0232] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: global_route.tcl, 85 GRT-0232
openroad> 
```

### Expected Behavior

The congestion.rpt should be non-zero lines:

```
$ wc -l bazel-out/k8-fastbuild/bin/reports/asap7/L1MetadataArray/test/congestion.rpt 
0 bazel-out/k8-fastbuild/bin/reports/asap7/L1MetadataArray/test/congestion.rpt
```

### Environment

```shell
OpenROAD v2.0-13652-g6fc686431
```

### To Reproduce

See above

oharboe, 2024-05-10:
@eder-matheus I saw your #5086 fix 👍 

Other than that, is this an interesting global route test case?

eder-matheus, 2024-05-10:
> @eder-matheus I saw your #5086 fix 👍
> 
> Other than that, is this an interesting global route test case?

It actually is. The congestion happens during the incremental grt, so there is only a subset of nets being routed. The congested region is pretty small, with only two gcells near the macro:
[image]

Also, there are a few cells illegally placed, so I wonder if there's something wrong with our grt script. I still looking at it.

oharboe, 2024-05-11:
Ref. @eder-matheus, there are two issues here.

1. missing congestion.rpt. fixed.
2. global routing should have worked.

Reopening to make sure the second problem does not slip between the cracks.

oharboe, 2024-05-11:
@eder-matheus Is this a global route or a placement  problem?

eder-matheus, 2024-05-11:
> @eder-matheus Is this a global route or a placement problem?

The problem is that repair_timing inserts lots of buffers between the macro and the die boundary. The buffers close to the macro are especially difficult for grt.

Most of these buffers would be legalized after repair_timing, but during it, we don't call DPL to avoid runtime issues. A possible fix would be update repair_timing to avoid placing buffers in the macro halos.

Pre repair_timing
[image]

Post repair_timing
[image]

eder-matheus, 2024-05-11:
> @eder-matheus Sounds like you have this under control. I suppose this might be tracked elsewhere?
> 
> 
> 
> This issue is not affecting me at the moment, please feel free to close this issue for my part or keep it open if that is helpful.

Sounds good, I'll close this issue and create a new one specifically for the repair_timing issue. I'll let you know when it's fixed.

closed 2024-05-11 by eder-matheus