The-OpenROAD-Project/OpenROAD #5093: Add rebuild button in GUI

oharboe, 2024-05-13:
### Description

When iterating on a stage in ORFS flow, it would be nice to simply hit "refresh" in the GUI to rebuild the requisite stage.

Current workflow:

1. I run `rm -f results/asap7/mock-array_Element/base/2_2_floorplan_io.odb && make DESIGN_CONFIG=designs/asap7/mock-array/Element/config.mk results/asap7/mock-array_Element/base/2_2_floorplan_io.odb && make DESIGN_CONFIG=designs/asap7/mock-array/Element/config.mk gui_2_2_floorplan_io.odb`
2. Modify IO pin constraints
3. Rinse & repeat

### Suggested Solution

Add "Rebuild" button in GUI that I can hit when having updated IO pin constraints.

I don't have the exact mechanics of this feature clear to me yet.

rovinski, 2024-05-13:
You could use `create_toolbar_button` to make your own, possibly

e.g.
```
create_toolbar_button -text "Run Floorplan" -script ./scripts/floorplan.tcl
```
I'm not sure if this pulls in the environment variables properly but I think it should.

oharboe, 2024-05-13:
@maliberty That only works for one hardcoded case, I would like it to work for all ORFS scripts without configuration.

maliberty, 2024-05-13:
I don't follow exactly what is requested here.  Are you trying to rebuild a flow step externally and reload the result or do some change in a session and see the gui update?  The later should happen automatically (we had the recent fix for ppl updating).

oharboe, 2024-07-14:
This would be nice, but closing issue as it is too far off and a lot can happen between now and when the path to such a feature is clear...

closed 2024-07-15 by maliberty