The-OpenROAD-Project/OpenROAD #5102: Add "rats nest" display prior to routing

oharboe, 2024-05-14:
### Description

It would be easier to do floorplaning(macro placement9 and io constraints if it was possible to see wires prior to them being routed.

Example from KiCad

[image]

### Suggested Solution

Add "rats nest" routing layer prior to routing.

This already exists for clock tree prior to routing.

```
make cts
make gui_cts
```

[image]

oharboe, 2024-05-14:
Thanks! We will give it a go.

closed 2024-05-14 by oharboe

rovinski, 2024-05-14:
Also I believe the more standard term for this is "fly lines" 🙂