The-OpenROAD-Project/OpenROAD #5105: `openroad -gui -no_init` crashes when clicking Timing Report->Update

oharboe, 2024-05-14:
### Describe the bug

```
$ openroad -gui -no_init
OpenROAD v2.0-13770-g98c6501d9 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Signal 11 received
Stack trace:
 0# 0x000060AC24E0C6E3 in openroad
 1# 0x00007C369AE42990 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbBlock::getBTerms() in openroad
 3# sta::dbNetwork::pinIterator(sta::Instance const*) const in openroad
 4# sta::Graph::makePinVertices(sta::Instance const*) in openroad
 5# sta::Graph::makeGraph() in openroad
 6# sta::Sta::ensureGraph() in openroad
 7# gui::STAGuiInterface::initSTA() const in openroad
 8# gui::STAGuiInterface::getTimingPaths(std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&, std::vector<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> >, std::allocator<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > > > const&, std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&) const in openroad
 9# gui::TimingPathsModel::populatePaths(std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&, std::vector<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> >, std::allocator<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > > > const&, std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&) in openroad
10# gui::TimingPathsModel::populateModel(std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&, std::vector<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> >, std::allocator<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > > > const&, std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&) in openroad
11# gui::TimingWidget::populateAndSortModels(std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&, std::vector<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> >, std::allocator<std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > > > const&, std::set<sta::Pin const*, std::less<sta::Pin const*>, std::allocator<sta::Pin const*> > const&) in openroad
12# gui::TimingWidget::populatePaths() in openroad
13# 0x00007C369C106312 in /lib/x86_64-linux-gnu/libQt5Core.so.5
14# QAbstractButton::clicked(bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
15# 0x00007C369CE5FE0E in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
16# 0x00007C369CE619EB in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
17# QAbstractButton::mouseReleaseEvent(QMouseEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
19# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
20# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
21# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
22# QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
23# 0x00007C369CDC7DA1 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
24# 0x00007C369CDCB12F in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
25# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
26# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
27# QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
28# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
29# 0x00007C36970F9F7E in /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
30# 0x00007C369B113B2C in /lib/x86_64-linux-gnu/libglib-2.0.so.0
31# 0x00007C369B16F46F in /lib/x86_64-linux-gnu/libglib-2.0.so.0
32# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
33# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
34# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
35# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
36# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
39# main in openroad
40# 0x00007C369AE28150 in /lib/x86_64-linux-gnu/libc.so.6
41# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
42# _start in openroad
Segmentation fault (core dumped)
```

### Expected Behavior

No crash

### Environment

```shell
OpenROAD v2.0-13770-g98c6501d9
```

### To Reproduce

See above

oharboe, 2024-05-17:
Any possibility of a CI unit-test on this one?

maliberty, 2024-05-17:
We don't have any GUI unit tests due to the lack of an OSS unit test framework for Qt.

oharboe, 2024-05-17:
> We don't have any GUI unit tests due to the lack of an OSS unit test framework for Qt.

This is one area where Python & Java(which I have used, other languages can do this too) is better: monkey patching and mocking makes it possible to test just about anything...

oharboe, 2024-05-17:
@QuantamHD Any ideas on how to write a CI unit-test something like this in OpenROAD for this?

OSS C++ monkey patching? Mocking libraries?

closed 2024-05-17 by maliberty

QuantamHD, 2024-05-17:
This looks reasonable. https://github.com/faaxm/spix

At Google we have a tool called scuba, which essentially just triggers actions via the UI handles and generates a png. You commit golden version of the pdf and at test runtime you compare the golden to the current. 

QT can be targeted at offscreen rendering, and I believe triggering actions via the API is possible.

PR #5130 (2024-05-17): grt: simplify the initialization of parallel arrays
The current code tries to resize the arrays parallel to the nets in one step.  However with multiple entry points (estimate_parasitics, rudy, global_route) it is hard to ensure they all do it right.

It is simpler to just resize them as you add nets.  Its slightly less efficient but minor.

Fixes #5105
PR #5131 (2024-05-17): dbSta: fix crash on calls when no dbBlock exists
DbInstancePinIterator and DbInstanceChildIterator

Fixes #5105
commit 001dfbe6 (2024-05-17): dbSta: fix crash on calls when no dbBlock exists

DbInstancePinIterator and DbInstanceChildIterator

Fixes #5105