The-OpenROAD-Project/OpenROAD #5107: Run GUI with TSAN

oharboe, 2024-05-15:
Run GUI with https://clang.llvm.org/docs/ThreadSanitizer.html and see if you can find any other thread issues with https://github.com/The-OpenROAD-Project/OpenROAD/issues/4990

_Originally posted by @QuantamHD in https://github.com/The-OpenROAD-Project/OpenROAD/issues/5106#issuecomment-2111546962_

QuantamHD, 2024-05-15:
Seems like it worked for me.

```

==================
WARNING: ThreadSanitizer: data race (pid=565408)
  Write of size 8 at 0x7b0800023b20 by thread T4:
    #0 free <null> (openroad+0x3642129) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 <null> <null> (libglib-2.0.so.0+0x56f2e) (BuildId: 53ec298e2881f5aca10690982d7e07681d55bfce)

  Previous write of size 8 at 0x7b0800023b20 by main thread (mutexes: write M0):
    #0 malloc <null> (openroad+0x36419c0) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 g_malloc <null> (libglib-2.0.so.0+0x61221) (BuildId: 53ec298e2881f5aca10690982d7e07681d55bfce)
    #2 tclAppInit(int&, char**, char const*, Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:352:5 (openroad+0x3720b76) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #3 ord::tclAppInit(Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:460:10 (openroad+0x37209da) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #4 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0) (BuildId: 563b9048c335ba1acabed5affb5fc9149741e310)
    #5 __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16 (libc.so.6+0x276c9) (BuildId: 28cb81ff244a5410a3db1cfa4f1e8da568627279)

  Mutex M0 (0x7b0c0002ef50) created at:
    #0 pthread_mutex_init <null> (openroad+0x3645104) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 g_rec_mutex_lock <null> (libglib-2.0.so.0+0xb4ed9) (BuildId: 53ec298e2881f5aca10690982d7e07681d55bfce)
    #2 tclAppInit(int&, char**, char const*, Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:352:5 (openroad+0x3720b76) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #3 ord::tclAppInit(Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:460:10 (openroad+0x37209da) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #4 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0) (BuildId: 563b9048c335ba1acabed5affb5fc9149741e310)
    #5 __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16 (libc.so.6+0x276c9) (BuildId: 28cb81ff244a5410a3db1cfa4f1e8da568627279)

  Thread T4 'gdbus' (tid=565413, running) created by main thread at:
    #0 pthread_create <null> (openroad+0x364373f) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 <null> <null> (libglib-2.0.so.0+0xb5749) (BuildId: 53ec298e2881f5aca10690982d7e07681d55bfce)
    #2 tclAppInit(int&, char**, char const*, Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:352:5 (openroad+0x3720b76) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #3 ord::tclAppInit(Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:460:10 (openroad+0x37209da) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #4 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0) (BuildId: 563b9048c335ba1acabed5affb5fc9149741e310)
    #5 __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16 (libc.so.6+0x276c9) (BuildId: 28cb81ff244a5410a3db1cfa4f1e8da568627279)

SUMMARY: ThreadSanitizer: data race (/usr/local/google/home/ethanmoon/OpenROAD/build/src/openroad+0x3642129) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349) in __interceptor_free
==================
==================
WARNING: ThreadSanitizer: data race (pid=565408)
  Write of size 8 at 0x7b0800023ba0 by thread T4:
    #0 free <null> (openroad+0x3642129) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 g_list_remove <null> (libglib-2.0.so.0+0x5562d) (BuildId: 53ec298e2881f5aca10690982d7e07681d55bfce)

  Previous write of size 8 at 0x7b0800023ba0 by main thread (mutexes: write M0):
    #0 malloc <null> (openroad+0x36419c0) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 g_malloc <null> (libglib-2.0.so.0+0x61221) (BuildId: 53ec298e2881f5aca10690982d7e07681d55bfce)
    #2 tclAppInit(int&, char**, char const*, Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:352:5 (openroad+0x3720b76) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #3 ord::tclAppInit(Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:460:10 (openroad+0x37209da) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #4 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0) (BuildId: 563b9048c335ba1acabed5affb5fc9149741e310)
    #5 __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16 (libc.so.6+0x276c9) (BuildId: 28cb81ff244a5410a3db1cfa4f1e8da568627279)

  Mutex M0 (0x7b0c0002ef50) created at:
    #0 pthread_mutex_init <null> (openroad+0x3645104) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 g_rec_mutex_lock <null> (libglib-2.0.so.0+0xb4ed9) (BuildId: 53ec298e2881f5aca10690982d7e07681d55bfce)
    #2 tclAppInit(int&, char**, char const*, Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:352:5 (openroad+0x3720b76) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #3 ord::tclAppInit(Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:460:10 (openroad+0x37209da) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #4 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0) (BuildId: 563b9048c335ba1acabed5affb5fc9149741e310)
    #5 __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16 (libc.so.6+0x276c9) (BuildId: 28cb81ff244a5410a3db1cfa4f1e8da568627279)

  Thread T4 'gdbus' (tid=565413, running) created by main thread at:
    #0 pthread_create <null> (openroad+0x364373f) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
[... cut ...]
    #3 ord::tclAppInit(Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:460:10 (openroad+0x37209da) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #4 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0) (BuildId: 563b9048c335ba1acabed5affb5fc9149741e310)
    #5 __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16 (libc.so.6+0x276c9) (BuildId: 28cb81ff244a5410a3db1cfa4f1e8da568627279)

SUMMARY: ThreadSanitizer: data race (/usr/local/google/home/ethanmoon/OpenROAD/build/src/openroad+0x3642129) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349) in __interceptor_free
```

oharboe, 2024-05-15:
@QuantamHD How did you compile and run it?

maliberty, 2024-05-15:
I can't pick out a clear issue from that.  Does it make sense to you?

QuantamHD, 2024-05-15:
This one seems interesting

```
==================
==================
WARNING: ThreadSanitizer: data race (pid=565408)
  Atomic read of size 1 at 0x7b0c000355b0 by main thread:
    #0 pthread_mutex_lock <null> (openroad+0x364541f) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 dbus_connection_get_is_connected <null> (libdbus-1.so.3+0x1593a) (BuildId: d7f167f1de4cee2abeba5eeb2bb4ec2729c12bfb)
    #2 gui::MainWindow::MainWindow(QWidget*) /usr/local/google/home/ethanmoon/OpenROAD/src/gui/src/mainWindow.cpp:86:21 (openroad+0x4bb73f5) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #3 gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>> const&, bool) /usr/local/google/home/ethanmoon/OpenROAD/src/gui/src/gui.cpp:1295:21 (openroad+0x4992944) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #4 tclAppInit(int&, char**, char const*, Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:352:5 (openroad+0x3720b76) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #5 ord::tclAppInit(Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:460:10 (openroad+0x37209da) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #6 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0) (BuildId: 563b9048c335ba1acabed5affb5fc9149741e310)
    #7 __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16 (libc.so.6+0x276c9) (BuildId: 28cb81ff244a5410a3db1cfa4f1e8da568627279)

  Previous write of size 8 at 0x7b0c000355b0 by thread T5 (mutexes: write M0):
    #0 malloc <null> (openroad+0x36419c0) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 <null> <null> (libdbus-1.so.3+0x3c577) (BuildId: d7f167f1de4cee2abeba5eeb2bb4ec2729c12bfb)

  Location is heap block of size 40 at 0x7b0c000355b0 allocated by thread T5:
    #0 malloc <null> (openroad+0x36419c0) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 <null> <null> (libdbus-1.so.3+0x3c577) (BuildId: d7f167f1de4cee2abeba5eeb2bb4ec2729c12bfb)

  Mutex M0 (0x7b0c000339f0) created at:
    #0 pthread_mutex_init <null> (openroad+0x3645104) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 <null> <null> (libdbus-1.so.3+0x3c5a7) (BuildId: d7f167f1de4cee2abeba5eeb2bb4ec2729c12bfb)
    #2 tclAppInit(int&, char**, char const*, Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:352:5 (openroad+0x3720b76) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #3 ord::tclAppInit(Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:460:10 (openroad+0x37209da) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #4 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0) (BuildId: 563b9048c335ba1acabed5affb5fc9149741e310)
    #5 __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16 (libc.so.6+0x276c9) (BuildId: 28cb81ff244a5410a3db1cfa4f1e8da568627279)

  Thread T5 'QDBusConnection' (tid=565421, running) created by main thread at:
    #0 pthread_create <null> (openroad+0x364373f) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 QThread::start(QThread::Priority) <null> (libQt5Core.so.5+0xda974) (BuildId: 3357a4af98f1c88252f025b9b6fdf26a700b80c2)
    #2 tclAppInit(int&, char**, char const*, Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:352:5 (openroad+0x3720b76) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #3 ord::tclAppInit(Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:460:10 (openroad+0x37209da) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #4 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0) (BuildId: 563b9048c335ba1acabed5affb5fc9149741e310)
    #5 __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16 (libc.so.6+0x276c9) (BuildId: 28cb81ff244a5410a3db1cfa4f1e8da568627279)

```

QuantamHD, 2024-05-15:
@oharboe I modified the ASAN options, and then compiled with `CC=clang CXX=clang++ cmake -DCMAKE_BUILD_TYPE=Debug -DASAN=ON -GNinja .. && ninja`

[image]

QuantamHD, 2024-05-15:
This one occured after reading a db

```
WARNING: ThreadSanitizer: data race (pid=571606)
  Read of size 1 at 0x7b7000051ef0 by thread T9 (mutexes: write M0):
    #0 gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected>> const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected>>, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler>>, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler>>>> const&, double, QColor const&) /usr/local/google/home/ethanmoon/OpenROAD/src/gui/src/renderThread.cpp:207:34 (openroad+0x4b7ad4e) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 gui::RenderThread::run() /usr/local/google/home/ethanmoon/OpenROAD/src/gui/src/renderThread.cpp:126:5 (openroad+0x4b7a5ff) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #2 <null> <null> (libQt5Core.so.5+0xdaeb0) (BuildId: 3357a4af98f1c88252f025b9b6fdf26a700b80c2)

  Previous write of size 1 at 0x7b7000051ef0 by main thread:
    #0 gui::RenderThread::render(QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected>> const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected>>, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler>>, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler>>>> const&) /usr/local/google/home/ethanmoon/OpenROAD/src/gui/src/renderThread.cpp:104:14 (openroad+0x4b7a412) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 gui::LayoutViewer::fullRepaint() /usr/local/google/home/ethanmoon/OpenROAD/src/gui/src/layoutViewer.cpp:1874:20 (openroad+0x4b1011e) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #2 QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (gui::LayoutViewer::*)()>::call(void (gui::LayoutViewer::*)(), gui::LayoutViewer*, void**) /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152:13 (openroad+0x4b5f62f) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #3 void QtPrivate::FunctionPointer<void (gui::LayoutViewer::*)()>::call<QtPrivate::List<>, void>(void (gui::LayoutViewer::*)(), gui::LayoutViewer*, void**) /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185:13 (openroad+0x4b5f555) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #4 QtPrivate::QSlotObject<void (gui::LayoutViewer::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:418:17 (openroad+0x4b5f3c4) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #5 <null> <null> (libQt5Core.so.5+0x30a365) (BuildId: 3357a4af98f1c88252f025b9b6fdf26a700b80c2)
    #6 QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (gui::ScriptWidget::*)()>::call(void (gui::ScriptWidget::*)(), gui::ScriptWidget*, void**) /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152:13 (openroad+0x4bf707f) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #7 void QtPrivate::FunctionPointer<void (gui::ScriptWidget::*)()>::call<QtPrivate::List<>, void>(void (gui::ScriptWidget::*)(), gui::ScriptWidget*, void**) /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185:13 (openroad+0x4bf6fa5) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #8 QtPrivate::QSlotObject<void (gui::ScriptWidget::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:418:17 (openroad+0x4bf6e14) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #9 <null> <null> (libQt5Core.so.5+0x3062b1) (BuildId: 3357a4af98f1c88252f025b9b6fdf26a700b80c2)
    #10 tclAppInit(int&, char**, char const*, Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:352:5 (openroad+0x3720b76) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #11 ord::tclAppInit(Tcl_Interp*) /usr/local/google/home/ethanmoon/OpenROAD/src/Main.cc:460:10 (openroad+0x37209da) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #12 Tcl_MainEx <null> (libtcl8.6.so+0x108aa0) (BuildId: 563b9048c335ba1acabed5affb5fc9149741e310)
    #13 __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16 (libc.so.6+0x276c9) (BuildId: 28cb81ff244a5410a3db1cfa4f1e8da568627279)

  Location is heap block of size 1952 at 0x7b7000051800 allocated by main thread:
    #0 operator new(unsigned long) <null> (openroad+0x36cc22b) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #1 gui::LayoutTabs::blockLoaded(odb::dbBlock*) /usr/local/google/home/ethanmoon/OpenROAD/src/gui/src/layoutTabs.cpp:77:17 (openroad+0x4b70c37) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #2 QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<odb::dbBlock*>, void, void (gui::LayoutTabs::*)(odb::dbBlock*)>::call(void (gui::LayoutTabs::*)(odb::dbBlock*), gui::LayoutTabs*, void**) /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152:13 (openroad+0x4be0366) (BuildId: 3deff8356d73a0bbe4bc685322ea5f6f7ab77349)
    #3 void QtPrivate::FunctionPointer<void (gui::LayoutTabs::*)(odb::dbBlock*)>::call<QtPrivate::List<odb::dbBlock*>, void>(void (gui::LayoutTabs::*)(odb::dbBlock*), gui::LayoutTabs*, void**) /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185:13 (openroad+0x4be0255) (BuildId [... cut ...]
SUMMARY: ThreadSanitizer: data race /usr/local/google/home/ethanmoon/OpenROAD/src/gui/src/renderThread.cpp:207:34 in gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected>> const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected>>, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler>>, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler>>>> const&, double, QColor const&)
==================
```

QuantamHD, 2024-05-15:
Looks like restart_ is potentially not protected by a mutex.

/usr/local/google/home/ethanmoon/OpenROAD/src/gui/src/renderThread.cpp:150:14
```
  if (!isRunning()) {
    start(LowPriority);
  } else {
    restart_ = true;
    condition_.wakeOne();
  }
```

vs

/usr/local/google/home/ethanmoon/OpenROAD/src/gui/src/renderThread.cpp:104:14
```
    mutex_.lock();
    if (!restart_) {
      condition_.wait(&mutex_);
    }
    restart_ = false;
    mutex_.unlock();

```

oharboe, 2024-05-15:
with:

```
$ git diff
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index 62059a8ce..b0bb31005 100755
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -197,13 +197,13 @@ add_compile_options(
   -ffp-contract=off
   # Apple clang 14.0.0 deprecates sprintf, which generates 900 warnings.
   $<$<CXX_COMPILER_ID:AppleClang>:-Wno-deprecated-declarations>
-  $<$<BOOL:${ASAN}>:-fsanitize=address>
+  $<$<BOOL:${ASAN}>:-fsanitize=thread>
   $<$<BOOL:${ASAN}>:-fno-omit-frame-pointer>
   $<$<BOOL:${ASAN}>:-g>
 )
 
 if (ASAN)
-  add_link_options(-fsanitize=address)
+  add_link_options(-fsanitize=thread)
 endif()
 
 ################################################################
$ cd build
$ CC=clang CXX=clang++ cmake -DCMAKE_BUILD_TYPE=Debug -DASAN=ON -GNinja .. && ninja
```

I get:

```
FAILED: src/gpl/test/fft_test src/gpl/test/fft_test[1]_tests.cmake /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gpl/test/fft_test[1]_tests.cmake 
: && /usr/bin/clang++ -g -fsanitize=thread src/gpl/test/CMakeFiles/fft_test.dir/fft_test.cc.o src/gpl/test/CMakeFiles/fft_test.dir/__/src/fft.cpp.o src/gpl/test/CMakeFiles/fft_test.dir/__/src/fftsg.cpp.o src/gpl/test/CMakeFiles/fft_test.dir/__/src/fftsg2d.cpp.o -o src/gpl/test/fft_test  lib/libgtest.a  lib/libgtest_main.a  /home/oyvind/OpenROAD-flow-scripts/dependencies/lib/libspdlog.a  lib/libgtest.a && cd /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gpl/test && /home/oyvind/OpenROAD-flow-scripts/dependencies/bin/cmake -D TEST_TARGET=fft_test -D TEST_EXECUTABLE=/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gpl/test/fft_test -D TEST_EXECUTOR= -D TEST_WORKING_DIR=/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/test -D TEST_EXTRA_ARGS= -D TEST_PROPERTIES= -D TEST_PREFIX= -D TEST_SUFFIX= -D TEST_FILTER= -D NO_PRETTY_TYPES=FALSE -D NO_PRETTY_VALUES=FALSE -D TEST_LIST=fft_test_TESTS -D CTEST_FILE=/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gpl/test/fft_test[1]_tests.cmake -D TEST_DISCOVERY_TIMEOUT=5 -D TEST_XML_OUTPUT_DIR= -P /home/oyvind/OpenROAD-flow-scripts/dependencies/share/cmake-3.24/Modules/GoogleTestAddTests.cmake
/usr/bin/ld: cannot find /usr/lib/llvm-16/lib/clang/16/lib/linux/libclang_rt.tsan-x86_64.a: No such file or directory
/usr/bin/ld: cannot find /usr/lib/llvm-16/lib/clang/16/lib/linux/libclang_rt.tsan_cxx-x86_64.a: No such file or directory
```

Even if I build only the "openroad" target, the build does not complete:

```
ninja openroad
```

```
[deleted]
usr/bin/ld: cannot find /usr/lib/llvm-16/lib/clang/16/lib/linux/libclang_rt.tsan-x86_64.a: No such file or directory
/usr/bin/ld: cannot find /usr/lib/llvm-16/lib/clang/16/lib/linux/libclang_rt.tsan_cxx-x86_64.a: No such file or directory
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

QuantamHD, 2024-05-15:
@oharboe Looks like there are specific packages that provide libclang_rt. You might just need to find where they are on your system and LD_PRELOAD them.

https://packages.debian.org/sid/libclang-rt-16-dev

maliberty, 2024-05-16:
@QuantamHD waiting on your PR for TSAN