The-OpenROAD-Project/OpenROAD #5121: Clicking on estimated routing congestion exist OpenROAD on `[ERROR GRT-0010]`

oharboe, 2024-05-16:
### Describe the bug

I clicked estimated routing congestion out of habit, not intention and then the GUI exited.

It would be nicer with an error and that the GUI remains running.

```
ODB_FILE=results/foo/base/3_2_place_iop.odb /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui  /home/oyvind/OpenROAD-flow-scripts/flow/scripts/gui.tcl
OpenROAD v2.0-13785-g6db1982e4 
Features included (+) or not (-): +Charts +GPU +GUI +Python : DEBUG
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
estimate_parasitics -placement
[WARNING GUI-0066] Heat map "Routing Congestion" has not been populated with data.
[ERROR GRT-0010] Instance controlAndStatusRegisters_6\[63\]$_DFFE_PP_ is not placed.
[WARNING GUI-0076] QPainter::end: Painter ended with 2 saved states
terminate called after throwing an instance of 'std::runtime_error'
  what():  GRT-0010
Signal 6 received
Stack trace:
 0# 0x00005FC93B94E2C4 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x0000720926E42990 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007209272A4F06 in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# 0x00007209272B6E6C in /lib/x86_64-linux-gnu/libstdc++.so.6
 7# 0x00007209272B6ED7 in /lib/x86_64-linux-gnu/libstdc++.so.6
 8# qt_assert(char const*, char const*, int) in /lib/x86_64-linux-gnu/libQt5Core.so.5
 9# 0x0000720927E9203C in /lib/x86_64-linux-gnu/libQt5Core.so.5
10# 0x0000720926E97B5A in /lib/x86_64-linux-gnu/libc.so.6
11# 0x0000720926F285FC in /lib/x86_64-linux-gnu/libc.so.6
make: *** [/home/oyvind/OpenROAD-flow-scripts/flow/Makefile:1059: gui_3_2_place_iop.odb] Aborted (core dumped)
```

### Expected Behavior

Don't exit the GUI on `[ERROR GRT-0010]`

### Environment

```shell
Latest master
OpenROAD v2.0-13785-g6db1982e4
```

### To Reproduce

The design that causes this error is confidential, but I it should be possible to determine this from inspecting the source code?

oharboe, 2024-05-16:
@maliberty Or is this a corrupt .odb file and not a GUI issue? In that case, the problem is that the previous stage generating the corrupt .odb file has to be fixed?

oharboe, 2024-05-16:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/5122

maliberty, 2024-05-16:
An error message shouldn't cause a seg fault or even an normal exit from the GUI.

maliberty, 2024-05-16:
I don't think downgrading to warnings is the right fix.  We need to see why the tool crashes.  Let's discuss privately about your test case.

oharboe, 2024-05-16:
> I don't think downgrading to warnings is the right fix. We need to see why the tool crashes. Let's discuss privately about your test case.

You should be able to modify the source code to inejct a false error here and verify that the error catching code is working, no?

stefanottili, 2024-05-16:
The heat map display setting shouldn’t be persistent.

maliberty, 2024-05-16:
> The heat map display setting shouldn’t be persistent.

Why?  There are times I am working on routing and want to see congestion each time I run.  If you don't want to see it just turn it off and exit and it will be remembered.

stefanottili, 2024-05-16:
A previous run with Heap Map settings left on 'routing congestion' might trigger
'Heat map "Routing Congestion" has not been populated with data.'
It will not run global routing to create the data. 

Rudy 'silently' recomputes it without any user feedback.
Which depending on the design might take up to 30 seconds leaving one wondering what's happening.
Not sure what happens with regard to IR drop.

If the heat map settings are persistent, it would be nice to have a
"computing RUDY" message, preferably with runtime info.

maliberty, 2024-05-16:
Fixed by https://github.com/The-OpenROAD-Project/OpenROAD/pull/5127

closed 2024-05-16 by maliberty