The-OpenROAD-Project/OpenROAD #5137: New "Incremental Routing" broke detailed routing

smunaut, 2024-05-18:
### Describe the bug

I'm feeding into OpenROAD some partially routed design (I have a custom OpenDB script that manually routes some signal).

In the `DEF` at the input you can see for instance :

```
    - gpio_out[29] ( PIN gpio_out[29] ) ( top_I.ctrl_I pad_uo_out[5] ) + USE SIGNAL
      + FIXED met4 ( 1495460 2436620 ) ( * 2838660 ) M3M4_PR ( 97980 * ) M3M4_PR ( * 2867335 ) M3M4_PR
     ( -850 * ) ;
```

With an older OpenROAD ( tested `4eceed4759ce048f57a5164253fcc32e688cd8ff` just before merging PR #5059 ), it works just fine, that routed net is unchanged at the output and the whole process takes like 15 seconds. ( There really isn't much to route in that test project ! ).

Tested a newer OpenROAD ( all revisions after the merge of that PR, I did a git bisect, none work after that merge ), at the output the net has been changed :

```
    - gpio_out[29] ( PIN gpio_out[29] ) ( top_I.ctrl_I pad_uo_out[5] ) + USE SIGNAL
      + ROUTED met4 ( 1495460 2436620 ) ( * 2838660 )
      NEW met3 ( 97980 2838660 ) ( 1495460 * )
      NEW met3 ( 97980 2838660 ) ( * 2867335 )
      NEW met3 ( -850 2867335 ) ( 97980 * )
      NEW met3 ( 1495460 2838660 ) M3M4_PR
      NEW met3 ( 97980 2838660 ) M3M4_PR
      NEW met3 ( 97980 2867335 ) M3M4_PR ;
```

Now, except for the first segment which is on `met4`, all the other segments are on `met3` instead of changing layer after each via !!!

And now the routing process takes around 20 minutes ! That is 80 times slower !  Not 80% slower ... no, 80 times longer, that's just unusable and this is a trivial test project ... 

### Expected Behavior

Well I expect it to not modify the routing explicitely marked as `FIXED` in the input and I also expect it to not take forever to do it.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.10.0-19-cloud-amd64
os: Debian GNU/Linux 11 (bullseye)
cmake version 3.18.4
-- The CXX compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-13718-g4eceed475
-- System name: Linux
-- Compiler: GNU 10.2.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.9 (found version "3.9.2") found components: Interpreter 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
[... 107 lines cut ...]
-- Found BISON: /usr/bin/bison (found version "3.7.5") 
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen missing components: dot
-- STA version: 2.5.0
-- STA git sha: ee8d3d0fa23bfbc69f3e936ff884c3d30f5bfb59
-- System name: Linux
-- Compiler: GNU 10.2.1
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- SSTA: 0
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.2", minimum required is "3.0")  
-- STA executable: /mnt/pdk/OL2/native/OpenROAD/src/sta/app/sta
-- Configuring incomplete, errors occurred!
```

### To Reproduce

[15-openroad-detailedrouting.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/15366307/15-openroad-detailedrouting.zip)

Added Openlane2 reproducible, should run on its own.

smunaut, 2024-06-02:
Any news on this ? Any more info you need from our side ?

maliberty, 2024-06-02:
No more info is needed for now.  This week will be slow as several people at ETH Zurich.

mattvenn, 2024-06-12:
Hi Matt and Osama - just wanted to say that this was the issue I mentioned at Zurich, and any issues opened by @smunaut will most likely be related to our work with Tiny Tapeout.

osamahammad21, 2024-06-14:
Could you try https://github.com/The-OpenROAD-Project/OpenROAD/pull/5245 and tell me if this is sufficient for your test case?

smunaut, 2024-06-14:
Just tried. Doesn't work for me.

It finishes faster that without the PR ( ~ 2.5 minutes ) but still significantly slower than before ( 15 second ).
But most importantly, it just doesn't produce a good result. Now instead of having those FIXED wires corrupted, they're just gone all together from the output.

osamahammad21, 2024-06-17:
It should be fixed now.

smunaut, 2024-06-17:
@osamahammad21 Yes, it works :tada: 
Thanks for looking into this.

closed 2024-06-18 by osamahammad21