The-OpenROAD-Project/OpenROAD #5145: Avoid GUI crash when trying to open hetamap configuration box

gudeh, 2024-05-21:
### Describe the bug

When trying to double click a heatmap on the GUI and no design is loaded we get a crash:

OpenROAD v2.0-13890-g319cce5a8 
Features included (+) or not (-): +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Signal 11 received
Stack trace:
 0# 0x00006047CBEE54AB in openroad
 1# 0x000070B0B1842520 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbBlock::getDbUnitsPerMicron() in openroad
 3# gui::HeatMapDataSource::showSetup() in openroad
 4# gui::DisplayControls::displayItemDblClicked(QModelIndex const&) in openroad
 5# 0x000070B0B28F1793 in /lib/x86_64-linux-gnu/libQt5Core.so.5
 6# QAbstractItemView::doubleClicked(QModelIndex const&) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
 7# QTreeView::mouseDoubleClickEvent(QMouseEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
 8# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
 9# QFrame::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
10# QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
11# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
12# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
13# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
14# QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
15# 0x000070B0B35C8D40 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
16# 0x000070B0B35CBFD5 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
17# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
19# QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
20# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
21# 0x000070B0ADEF8D6E in /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
22# g_main_context_dispatch in /lib/x86_64-linux-gnu/libglib-2.0.so.0
23# 0x000070B0B1B712B8 in /lib/x86_64-linux-gnu/libglib-2.0.so.0
24# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
25# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
26# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
27# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
28# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
29# ord::tclAppInit(Tcl_Interp*) in openroad
30# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
31# main in openroad
32# 0x000070B0B1829D90 in /lib/x86_64-linux-gnu/libc.so.6
33# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
34# _start in openroad
Segmentation fault (core dumped)

### Expected Behavior

Make the tool not to crash if there is no design loaded.

### Environment

```shell
Git commit: 5c702fa8e68b5f258f4eb4cc59ced242b5d3ebb9
kernel: Linux 6.5.0-35-generic
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-13883-g5c702fa8e
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
[... 120 lines cut ...]
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /home/gudeh/.local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /home/gudeh/.local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /home/gudeh/.local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- FouNumber of processor cores: 16
nd Boost: /home/gudeh/.local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- Found Eigen3: /home/gudeh/.local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.Fu4WxryK4F
```

### To Reproduce

openroad -gui
double click any heatmap.

maliberty, 2024-05-21:
@gudeh are you planning to look at this or do you need help from @AcKoucher ?

gudeh, 2024-05-21:
> @gudeh are you planning to look at this or do you need help from @AcKoucher ?

@AcKoucher just told me to create the issue. I showed the problem on a synch we just did.