The-OpenROAD-Project/OpenROAD #5169: PDN does not act as obstruction for routing from some commits ago

OuDret, 2024-05-28:
### Describe the bug

PDN stopped acting as obstruction, so it shorts with internal routing. 

Picture attached below.

### Expected Behavior

It acts as obstruction once again

### Environment

```shell
I have tested for some OpenROAD versions.
It did not work properly for versions:
v2.0-13932-g333fce946 (LATEST)
v2.0-13893-gf0cc0ee1a

it worked for
v2.0-13302-gda0053d7b
v2.0-12371-gf6da0454e
```

### To Reproduce

I cannot provide most of files but it should be easy to reproduce in any techonology by using the same layer for both power grid and routing.

COMMANDS USED:

pdn.tcl
```
add_global_connection -net {VDD} -inst_pattern {.*} -pin_pattern {^VDD$} -power
add_global_connection -net {VSS} -inst_pattern {.*} -pin_pattern {^VSS$} -ground
add_global_connection -net {VBG} -inst_pattern {.*} -pin_pattern {^VBG$} -power

global_connect

set_voltage_domain -name {CORE} -power {VDD} -ground {VSS} -secondary_power {VBG}
define_pdn_grid -name {grid} -voltage_domains {CORE} 

add_pdn_stripe -grid {grid} -width {300} -layer {metal2} -followpins
add_pdn_stripe -grid {grid} -layer {metal1} -pitch {1470} -spacing {0} -offset {1295} -width {50} -nets VBG -extend_to_boundary
add_pdn_stripe -grid {grid} -layer {metal1} -pitch {1470} -spacing {0} -offset {175} -width {50} -nets VBG -extend_to_boundary
```
flow.tcl
```
# Assumes flow_helpers.tcl has been read.
read_libraries

read_verilog $synth_verilog
link_design $top_module
read_sdc $sdc_file

initialize_floorplan -site $site \
  -die_area $die_area \
  -core_area $core_area

insert_tiecells $tielo_cell/$tielo_port

source $tracks_file

place_pins -random -hor_layers $io_placer_hor_layer -ver_layers $io_placer_ver_layer

source $pdn_cfg
pdngen

repair_tie_fanout -separation $tie_separation $tielo_cell/$tielo_port

set_routing_layers -signal $global_routing_layers \
  -clock $global_routing_clock_layers
set_macro_extension 2

global_placement -routability_driven -density $global_place_density \
  -pad_left $global_place_pad -pad_right $global_place_pad -overflow $overflow

place_pins -hor_layers $io_placer_hor_layer -ver_layers $io_placer_ver_layer

source $layer_rc_file
set_wire_rc -signal -layer $wire_rc_layer
set_wire_rc -clock  -layer $wire_rc_layer_clk

estimate_parasitics -placement

set_placement_padding -global -left $detail_place_pad -right $detail_place_pad
detailed_placement

set_propagated_clock [all_clocks]

detailed_placement -disallow_one_site_gaps

pin_access -bottom_routing_layer $min_routing_layer \
           -top_routing_layer $max_routing_layer

set route_guide designs/$design/results/$design.route_guide

global_route -guide_file $route_guide \
  -congestion_iterations 100 -allow_congestion

pin_access -bottom_routing_layer $min_routing_layer \
           -top_routing_layer $max_routing_layer

set_thread_count [exec getconf _NPROCESSORS_ONLN]

detailed_route -output_drc designs/$design/results/$design.rpt \
               -output_maze designs/$design/results/$design.log \
               -no_pin_access \
               -save_guide_updates \
               -bottom_routing_layer $min_routing_layer \
               -top_routing_layer $max_routing_layer \
               -verbose 0
```

### Screenshots

[image]

stefanottili, 2024-05-28:
you run global_route with a route guide, but where does it come from ? 
An older run without power ?
Try global_route without route guide.

gadfort, 2024-05-29:
@maliberty I've seen something similar in skywater130/aes, most of the shorts from iteration 0 are from routing through the followpins in the power grid, which causes the router to spend most of the iterations trying to clear the shorts.

maliberty, 2024-05-29:
That wouldn't imply that it isn't aware of the short, just that it might converge faster with a different strategy.  Without a real test case this remains unactionable.

OuDret, 2024-05-29:
> That wouldn't imply that it isn't aware of the short, just that it might converge faster with a different strategy. Without a real test case this remains unactionable.

I got permision to send you privately through email our test case. If you want to take a look tell which address to send it to.

Thanks

maliberty, 2024-05-30:
info@precisioninno.com can be used for private communication.

closed 2024-07-08 by osamahammad21