The-OpenROAD-Project/OpenROAD #5190: Instance master is not equivalent

lazzy-dog, 2024-05-31:
### Describe the bug

When I execute inst.swapMaster(n_master), I find that some instances' masters cannot be swapped, such as "AOI222", "OAI222", "OAI221", etc. An error is reported:
[ERROR STA-1000] instance inst_6 swap master AOI222_X4 is not equivalent
However, I have checked my library file, and the information for these masters is complete. Moreover, other masters in my design can be replaced normally, such as "AND2_X1". I want to ask why only these masters cannot be swapped, and they all start with "AOI" or "OAI".
Attached is the test Verilog file I wrote.（Since I couldn't upload the Verilog file, I changed its suffix to .txt）
Sincerely

### Expected Behavior

I want these masters to swap normally.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.5.0-35-generic
os: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-12411-gf456b0a1b
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /home/mrlris/anaconda3/bin/python3.11 (found version "3.11.7") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
[... 176 lines cut ...]
 8# 0x0000703A676A2B9E in /lib/x86_64-linux-gnu/libstdc++.so.6
 9# 0x0000703A676AE20C in /lib/x86_64-linux-gnu/libstdc++.so.6
10# 0x0000703A676AE277 in /lib/x86_64-linux-gnu/libstdc++.so.6
11# 0x0000703A676AE4D8 in /lib/x86_64-linux-gnu/libstdc++.so.6
12# sta::dbStaCbk::inDbInstSwapMasterBefore(odb::dbInst*, odb::dbMaster*) in ./openroad
13# odb::dbInst::swapMaster(odb::dbMaster*) in ./openroad
14# 0x0000632C11AB2149 in ./openroad
15# 0x0000703A67B28008 in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
16# _PyObject_MakeTpCall in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
17# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
18# 0x0000703A67BC23AF in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
19# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
20# 0x0000703A67BC23AF in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
21# _PyEval_EvalFrameDefault in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
22# 0x0000703A67BC23AF in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
23# PyEval_EvalCode in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
24# 0x0000703A67C0C4CD in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
25# _PyRun_SimpleFileObject in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
26# _PyRun_AnyFileObject in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
27# Py_RunMain in /lib/x86_64-linux-gnu/libpython3.10.so.1.0
28# main in ./openroad
29# 0x0000703A67229D90 in /lib/x86_64-linux-gnu/libc.so.6
30# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
31# _start in ./openroad
```

rovinski, 2024-05-31:
Masters can only be swapped if they are logically equivalent. The cells you mention are not logically equivalent. See:

https://github.com/The-OpenROAD-Project/OpenROAD/blob/ad54bbe88b561d1c30451d8a3c85ad11c1692905/src/dbSta/src/dbSta.cc#L752-L764

https://github.com/The-OpenROAD-Project/OpenSTA/blob/e01d3f163f483f233db00410b6515a767a6ca03b/liberty/EquivCells.cc#L258-L265

lazzy-dog, 2024-06-02:
Yes, it seems that I overlooked a few conditions which led to my mistake. Once again, thank you for your response.
Sincerely

closed 2024-06-02 by lazzy-dog