The-OpenROAD-Project/OpenROAD #5197: Global Placement Crash

kareefardi, 2024-06-04:
### Describe the bug

Crash when calling `global_placement`. Crash is avoided by passing `-skip_nesterov_place`.

### Expected Behavior

Graceful exit or no crash

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.8.10-zabbly+
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-13997-gad54bbe88
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
[... 123 lines cut ...]
Signal 8 received
Stack trace:
 0# 0x00005578D632FB7B in openroad
 1# 0x00007FCBD1842520 in /lib/x86_64-linux-gnu/libc.so.6
 2# gpl::BinGrid::initBins() in openroad
 3# gpl::NesterovBase::NesterovBase(gpl::NesterovBaseVars, std::shared_ptr<gpl::PlacerBase>, std::shared_ptr<gpl::NesterovBaseCommon>, utl::Logger*) in openroad
 4# gpl::Replace::initNesterovPlace(int) in openroad
 5# gpl::Replace::doNesterovPlace(int, int) in openroad
 6# 0x00005578D63F5D7E in openroad
 7# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 8# 0x00007FCBD5D37028 in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
11# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
12# ord::tclAppInit(Tcl_Interp*) in openroad
13# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# main in openroad
15# 0x00007FCBD1829D90 in /lib/x86_64-linux-gnu/libc.so.6
16# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
17# _start in openroad
```

### Additional Context

Design has multiple regions.

gudeh, 2024-06-04:
Hi @kareefardi, thanks for your report. Right out of the bat I see a 0 for the number of GCells on the log you sent.

Were you able to run your scripts on a previous version of OR before?

gudeh, 2024-06-05:
I created a PR to avoid the crash. Although, I did not yet discovered the root cause of the issue. With the following fix we get another error moving forward in gpl.

kareefardi, 2024-06-05:
@gudeh Thanks for the update. I didn't run this specific script on previous version. 
I tried the branch in the PR. Indeed I ran into error requesting a higher density. However after setting a higher density I am running into 
```
[ERROR GPL-0304] RePlAce diverged at initial iteration with steplength being -nan. Re-run with a smaller init_density_penalty value.
```
I set the smallest possible `init_density_penalty` but I am still getting the same error. Do you have hints on how to proceed?

gudeh, 2024-06-05:
Are you sure the floorplan is correct?

kareefardi, 2024-06-06:
The design has two regions. All the cells are divided into these regions. There are no cells in the area outside these regions. This is my first time dealing with regions, but it seems unlikely that there something illegal about have the design's cells belong only to regions.

gudeh, 2024-06-06:
Hi @kareefardi. It seems GPL support to regions is broken. We do not use `set_domain_area` anywhere on ORFS scripts, so this has not come to light previously, thanks for sharing the error! I will try to fix this.

kareefardi, 2024-06-11:
Hello @gudeh. Thanks for the reply. Since this is untapped territory, there is not telling whether this is something simple to fix or might take a while. Is that correct?

gudeh, 2024-06-11:
Indeed, here is what happens. I have been recently assigned to work on gpl. Although, this is the first time I dive deeper into the initial placer step, and also with regions and groups. This issue is good, because it is an opportunity for me to be more proficient with the algorithm, and OFC solve an actual problem.

The `Replace::doIinitialPlace()` instantiates some important objects for gpl, including a `PlacerBase` object for each existing group, and one for instances without any groups. The test case in `tools/OpenROAD/test/upf_test` also uses two regions, although we have some cells without any group. On your test case all cells are assigned to at least one group. We end up having one `PlacerBase` object without any cells (it is curious that it assigned some "dummy instances", which I do not fully understand yet), consequently further in the algorithm some calculations end up having NANs, because there are no instances in this `PlacerBase` object.

By erasing the PlacerBase object without cells the gpl converges until the end: 
[image]

[image]

gudeh, 2024-06-11:
I am still a little worried, because I would expect the cells to be initially placed on the center of each region. Since, when we don't have regions, they usually start at the center of the core area.

[image]

maliberty, 2024-06-11:
It looks like initial placement doesn't consider the regions and that gets fixed by nesterov.  It would be good if initial placement were are as well. 

I believe they are off to the side as all pins are on one side and that minimizes wire length (when ignoring the regions).

I'm a bit more concerned that at the end of gpl there are objects outside their regions.  dpl is fixing that but it seems wrong.  Do you understand why this happens?

maliberty, 2024-06-11:
I think your idea to remove the empty placer is correct.

gudeh, 2024-06-13:
> I think your idea to remove the empty placer is correct.

I ran a secure-CI and the failures it got are the same happening on master. I will update the PR for this crash.

gudeh, 2024-06-13:
> It looks like initial placement doesn't consider the regions and that gets fixed by nesterov. It would be good if initial placement were are as well.
>
> I believe they are off to the side as all pins [...]

All of the instances completely outside the regions are clock buffers: `sky130_fd_sc_hd__clkbuf_1` and `sky130_fd_sc_hd__clkbuf_2`. Is it common that they have a special treatment? I see we have a MBFF class on gpl, maybe it is related?

maliberty, 2024-06-13:
mbff is for flop clustering and is likely unrelated.  Perhaps those buffers are being added by the resizer.  If you look at the names of the instances it should give a clue.  If they are really in the no region then their placement is fine.

gudeh, 2024-06-13:
I forgot to mention @kareefardi, the prints I sent here were with a `-density 0.7`, I was not able to run with your preconfigured 0.1, although 0.15 did finish with my fix.

[image]

gudeh, 2024-06-13:
This issue raised another problem also, with the TCL interpreter. We have an exit in the middle of the script, before detail_placement, if we leave it there and run the script. It exits after running all of the commands in the script, as if the exit was in the end of the document, not the middle. 

Should I create another issue for that?

gudeh, 2024-06-13:
> mbff is for flop clustering and is likely unrelated. Perhaps those buffers are being added by the resizer. If you look at the names of the instances it should give a clue. If they are really in the no region then their placement is fine.

some of their names are:

- spm_inst_0/input23
- spm_inst_0/input27
- spm_inst_0/input31
- spm_inst_0/input30
- spm_inst_1/input1
- spm_inst_1/input30

maliberty, 2024-06-13:
Those are input buffers so they wouldn't be in the incoming netlist.  Its fine for them to be outside the regions afaik.

maliberty, 2024-06-13:
> Should I create another issue for that?

Yes

gudeh, 2024-06-17:
I am closing this issue. Please, feel free to confirm if your design is working as expected @kareefardi.

closed 2024-06-17 by gudeh

kareefardi, 2024-06-23:
@gudeh Thanks for your effort. The design does finish now. 

However following up on @maliberty comment with respect to these kind of instances `spm_inst_0/input23`, they do belong to the region. The netlist created for `spm_inst_0` used resizer to insert these input buffers. But in the top level design that instantiates the macros, no input buffers are inserted by resizer. Here is a snippet of the generated def file:
```
GROUPS 2 ;
    - spm_inst_0_domain spm_inst_0/output35_y_iso_y_iso_o_isolation spm_inst_0/output35 spm_inst_0/input9
         spm_inst_0/input8 spm_inst_0/input7 spm_inst_0/input6 spm_inst_0/input5
         spm_inst_0/input4 spm_inst_0/input34 spm_inst_0/input33 spm_inst_0/input32
         spm_inst_0/input31 spm_inst_0/input30 spm_inst_0/input3 spm_inst_0/input29
         spm_inst_0/input28 spm_inst_0/input27 spm_inst_0/input26 spm_inst_0/input25
         spm_inst_0/input24 spm_inst_0/input23 spm_inst_0/input22 spm_inst_0/input21
```