The-OpenROAD-Project/OpenROAD #5203: Make info message more actionable `[INFO GPL-0100] worst slack -1.54e-09`

oharboe, 2024-06-05:
### Description

What is OpenROAD trying to tell me here?

Is OpenROAD trying to tell me that removing `-timing_driven` could speed things up?

For architectural exploration, I initially want fast turnaround time with less accurate results.

```
[INFO GPL-0100] worst slack -1.54e-09
[INFO GPL-0103] Weighted 41603 nets.
```

### Suggested Solution

Make the error message more actionable.

gudeh, 2024-06-05:
Hi @oharboe, this message is not an error message actually. This is printing the worst slack after calling repair design with resizer. Maybe we should transform this into a debug message.

oharboe, 2024-06-05:
@gudeh  Understood. However, it may be useful for the user to know that they can relax the optimization target to improve running times. If message can be rearticulated to communicate this, it would be helpful.

maliberty, 2024-06-06:
It seems that having some document or global options on fast analysis is better than sprinkling hints throughout the log

oharboe, 2024-06-06:
@rovinski Had an idea to collect data on running times vs quality of results and creating a tool that recommended options?

gudeh, 2024-07-03:
I am closing this issue. I am currently working on the timing driven mode in global placer, which is what generates the mentioned logs. We should have further modifications on timing driven in the near future.

closed 2024-07-03 by gudeh