The-OpenROAD-Project/OpenROAD #5214: CI for no differences in .ok file

rovinski, 2024-06-08:
### Description

As a simple sanity check, there could be a CI test that any modified `.ok` files do not have differences in them. This will ensure that they don't get missed on review.

### Suggested Solution

A simple script which greps for "Differences found at line" should suffice.

maliberty, 2024-06-09:
@vvbandeira could this be a GH action?

vvbandeira, 2024-06-11:
> @vvbandeira could this be a GH action?

I think Jenkins should be fine. It would be a simple `grep "Differences found at line" **/*.ok` or `for f in **/*.ok; do ...` plus some minor parsing/substitutions just to print the names of the tests in a nice format.

rovinski, 2025-05-22:
Resolved by #7389

closed 2025-05-22 by rovinski