The-OpenROAD-Project/OpenROAD #5218: libortools.so.9: cannot open shared object file: No such file or directory

juliannz, 2024-06-10:
### Describe the bug

I have the same problem as described in issue #2608, but without docker usage.
I installed or-tools manually, as the binarys are not available for all dists.
`ldd build/src/openroad|grep or` gives the correct result, but `ldd <install-prefix>/bin/openroad|grep or` gives `libortools.so.9 => not found`. I already tried to change the runtime [path](https://github.com/The-OpenROAD-Project/OpenROAD/blob/7a8341464731c2a7216878bb4602d1b2405e5179/src/CMakeLists.txt#L39) and set it to `<install-prefix>`, but this doesnt change anything...

Workarounds:
 - just install or-tools to /opt
 - `rm <install-prefix>/bin/openroad && ln -s build/src/openroad <install-prefix>/bin/openroad`

but both are not that satisfying...

### Expected Behavior

`<install-prefix>/bin/openroad` shouldnt fail

### Environment

```shell
doesnt matter imao
```

### To Reproduce

build using or-tools not installed to /opt

maliberty, 2024-06-10:
Have you tried setting your LD_LIBRARY_PATH to find the .so?  What platform are you using that requires manual installation.

juliannz, 2024-06-10:
Thanks for the hint, I googled a bit and adding `set(CMAKE_INSTALL_RPATH_USE_LINK_PATH TRUE)` after my changes to the [build file](https://github.com/The-OpenROAD-Project/OpenROAD/blob/7a8341464731c2a7216878bb4602d1b2405e5179/src/CMakeLists.txt#L39) resolves the issue (:
Its available again for Ubuntu 20, but wasnt in the [last release](https://github.com/google/or-tools/releases/tag/v9.9). So you're right, pretty rare use case that one needs to build or-tools from source^^
Still, if youre interested, I can provide my fix via pr or so (but after I finished my current project)...

vvbandeira, 2025-08-22:
Closing as the problem looks to be resolved. Please reopen if the problem is still happening on the latest version.

closed 2025-08-22 by vvbandeira