The-OpenROAD-Project/OpenROAD #5230: Segfault when clicking "Navigate back"

oharboe, 2024-06-12:
### Describe the bug

To reproduce:

```
make
make gui_final
```

[image]

[image]

Navigate a bit and click "Navigate back" [image]

Should crash after a few tries:

```
Signal 11 received
Stack trace:
 0# 0x00005FA829D99794 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x000070F7EA442990 in /lib/x86_64-linux-gnu/libc.so.6
 2# std::any::has_value() const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# std::any::any(std::any const&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# gui::Selected::Selected(gui::Selected const&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# 0x00005FA82A8DE879 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# 0x00005FA82A8E1E3B in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# 0x00005FA82A8E1937 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# 0x00005FA82A8E12CF in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# std::function<gui::Selected ()>::operator()() const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# gui::Inspector::handleAction(QWidget*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# 0x00005FA82A8DEE28 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# 0x00005FA82A8E2263 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# 0x00005FA82A8E211F in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
14# 0x00005FA82A8E1F26 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
15# 0x000070F7EB706312 in /lib/x86_64-linux-gnu/libQt5Core.so.5
16# 0x000070F7EC45FFE1 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
17# 0x000070F7EC4619D6 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# QAbstractButton::mouseReleaseEvent(QMouseEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
19# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
20# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
21# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
22# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
23# QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
24# 0x000070F7EC3C7DA1 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
25# 0x000070F7EC3CB12F in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
26# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
27# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
28# QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
29# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
30# 0x000070F7E66F9F7E in /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
31# 0x000070F7EA713B2C in /lib/x86_64-linux-gnu/libglib-2.0.so.0
32# 0x000070F7EA76F46F in /lib/x86_64-linux-gnu/libglib-2.0.so.0
33# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
34# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
35# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
36# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
37# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
38# 0x00005FA829D99FB9 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
39# ord::tclAppInit(Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
40# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
41# main in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
42# 0x000070F7EA428150 in /lib/x86_64-linux-gnu/libc.so.6
43# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
44# _start in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
make: *** [Makefile:1067: gui_6_final.odb] Segmentation fault (core dumped)
```

### Expected Behavior

No crash

### Environment

```shell
OpenROAD v2.0-14155-g417d6db18
```

### To Reproduce

See above

maliberty, 2024-06-12:
I can reproduce this

maliberty, 2024-06-12:
@gadfort you implemented the back button so I'd like your insight.  I see that on the selection we make the action
`makeAction({"Navigate back",...` with a lambda and when the back button is clicked we trigger the action.  

When the action is run Inspector::navigateBack() calls Inspector::inspect() -> reload() -> loadActions() which deletes all the actions including the one currently being run.  When we return back up the call stack we arrive at the deallocated lambda from the action.  

I believe that is the source of the crash.  Do you have thoughts on the best fix?

gadfort, 2024-06-12:
@maliberty taking a quick glance at this, I don't see an obvious fix that doesn't require some odd behavior. Possibly the proper solution is to make calls to inspect a queued connection so the Qt event loop would have to finish one thing before attempting to handle the inspect avoiding the call to `reload()` until after the lambda function has returned.

PR #5238 (2024-06-13): gui: fix a crash when hitting the back button in the inspector
Copy the action to avoid it being deleted while executing.

Fixes #5230
commit 9e138d09 (2024-06-13): gui: fix a crash when hitting the back button in the inspector

Copy the action to avoid it being deleted while executing.

Fixes #5230
closed 2024-06-13 by maliberty