The-OpenROAD-Project/OpenROAD #5244: Error in finding logically equivalent cells in OpenSTA using ASAP7

jaemin0502, 2024-06-14:
### Describe the bug

Error detected while finding the equivalent cells using ASAP7 PDK. 

It seems that OpenSTA don't observe that "O2A1O1Ixp5_ASAP7_75t_R" and "O2A1O1Ixp33_ASAP7_75t_R" are logically equivalent.

### Expected Behavior

They should be considered logically equivalent cells.
I can see the same functions as show below.

function : "(!A1 * !A2 * !C) + (!B * !C)";
function : "(!A1 * !A2 * !C) + (!B * !C)";

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 3.10.0-1160.118.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.25.1
-- The CXX compiler identification is GNU 9.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/tool/gcc/gcc-9.3.0/bin/x86_64-pc-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-13992-g38a3b37
-- System name: Linux
-- Compiler: GNU 9.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/tool/gcc/gcc-9.3.0/bin/x86_64-pc-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.6 (found version "3.6.8") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
[... 90 lines cut ...]
-- TCL header: /usr/include/tcl.h
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /home/tool/boost/boost_1_81_0/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: seNumber of processor cores: 8
rialization 
-- Found VTune: /home/tool/intel/vtune_amplifier/lib64/libittnotify.a  
-- Found Boost: /home/tool/boost/boost_1_81_0/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found suitable version "1.81.0", minimum required is "1.78")  
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Boost: /home/tool/boost/boost_1_81_0/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization system thread 
-- Found Boost: /home/tool/boost/boost_1_81_0/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0")  
-- Found Eigen3: /home/tool/ortools/install/CentOS7-gcc9/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline disabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /dev/shm/tmp.IOEm7FkOPl
```

### To Reproduce

After cloning the github repo below, I ran, "cd 2024_ICCAD_Contest_Gate_Sizing_Benchmark/src/evaluation/ & openroad -python evaluation.py --file_path test.size --design_name NV_NVDLA_partition_m"
https://github.com/ASU-VDA-Lab/2024_ICCAD_Contest_Gate_Sizing_Benchmark/tree/main
I uploaded test.tcl file in this issue post. (please change the name of the file into test.size)

maliberty, 2024-06-14:
I don't see any file test.size (or test.tcl) uploaded here.

maliberty, 2024-06-14:
Note: the repo to clone is git@github.com:ASU-VDA-Lab/2024_ICCAD_Contest_Gate_Sizing_Benchmark.git and various files need to be bunzip2'ed.

jaemin0502, 2024-06-14:
[test.txt](https://github.com/user-attachments/files/15838080/test.txt)

I uploaded test.txt file and you can change this to test.size.

maliberty, 2024-06-14:
Now I get
```
Error: Instance "FE_PHC4441_csb2cmac_a_req_pvld" not found.
```

Please make a self-contained test case as this is not efficient.

jaemin0502, 2024-06-16:
I'm sorry but the repo "[git@github.com](mailto:git@github.com):ASU-VDA-Lab/2024_ICCAD_Contest_Gate_Sizing_Benchmark.git" was **committed exactly the day you downloaded the test design** and this caused the error you said. 

When you want to reproduce the error without downloading the previous code, please simply change the line 1530 of "2024_ICCAD_Contest_Gate_Sizing_Benchmark/design/NV_NVDLA_partition_m/NV_NVDLA_partition_m.size" to "FE_RC_7074_0 O2A1O1Ixp**33**_ASAP7_75t_R" and run "cd 2024_ICCAD_Contest_Gate_Sizing_Benchmark/src/evaluation/ & openroad --python evaluation.py --file_path ../../design/NV_NVDLA_partition_m/NV_NVDLA_partition_m.size --design_name NV_NVDLA_partition_m

maliberty, 2024-06-16:
sta checks more than the function to determine equivalence.  O2A1O1Ixp33_ASAP7_75t_R has 12 timing arcs while O2A1O1Ixp5_ASAP7_75t_R has 11.  They are therefore not considered equivalent.

I don't know why they have a different number of arcs.  That seems like a library issue.

maliberty, 2024-06-17:
The library seems ok but https://github.com/parallaxsw/OpenSTA/discussions/42