The-OpenROAD-Project/OpenROAD #5250: Crash with signal 11 while optimizing routing

TheMichaelGionet, 2024-06-16:
### Describe the bug

I am using siliconcompiler to do an asic flow, where OpenROAD is used for the floorplanning, placement, cts, and routing steps. During the optimization parts of the routing steps, there might be a crash with a signal 11 attached to it. When I give a design a larger area then a crash is less likely to occur, but if I give it a smaller area then a crash is more likely. 

[openroad_crash_msg.txt](https://github.com/user-attachments/files/15858704/openroad_crash_msg.txt)

### Expected Behavior

Not crashing

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.0-91-generic
os: Linux Mint 21.2 (Victoria)
cmake version 3.22.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-14057-g818b8886f
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
[... 100 lines cut ...]
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/boostlib/lib/cmake/Boost-1.85.0/BoostConfig.cmake (found suitable version "1.85.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/boostlib/lib/cmake/Boost-1.85.0/BoostConfig.cmake (found version "1.85.Number of processor cores: 4
0") found components: serialization system thread 
-- Found Boost: /usr/local/boostlib/lib/cmake/Boost-1.85.0/BoostConfig.cmake (found version "1.85.0")  
-- Found Eigen3: /usr/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.dkO9ZS394n
```

### To Reproduce

pull the branch used for siliconcompiler (https://github.com/siliconcompiler/siliconcompiler.git). I used commit e54dbf73d3627c65e5e609b989ddeb36e5d3dbb9.
Install siliconcompiler using the tutorial on their guide https://docs.siliconcompiler.com/en/stable/user_guide/installation.html.
Extract the uploaded tar file into some directory and cd into it.
Do "python3 run.py".
It should then fail when it gets to the optimization step during routing. 

[siliconcompiler_decoder.tar.gz](https://github.com/user-attachments/files/15858722/siliconcompiler_decoder.tar.gz)

maliberty, 2024-06-16:
@gadfort can SiC produce a standalone test case like 'make issue'?

gadfort, 2024-06-16:
@TheMichaelGionet @maliberty
You should be able to use:
`sc-issue -cfg build/blah/route/0/inputs/blah.pkg.json (replace with correct path) -generate`
This should generate standalone testcase.

maliberty, 2024-06-16:
I will wait for you to package one before looking at this.

TheMichaelGionet, 2024-06-17:
I apologize for the wonky way this is uploaded, github wanted the file under 25 MB so I split up the tar file, but it also wanted it in a zip format so I zipped each of the 3 parts. To reconstruct it just unzip each of these and cat them together (aa first, then ab, then ac). 

[sc_issue_decode_sc_module_1_job0_route0_20240617-101924.tar.gz_pt_ac.zip](https://github.com/user-attachments/files/15872884/sc_issue_decode_sc_module_1_job0_route0_20240617-101924.tar.gz_pt_ac.zip)
[sc_issue_decode_sc_module_1_job0_route0_20240617-101924.tar.gz_pt_aa.zip](https://github.com/user-attachments/files/15872881/sc_issue_decode_sc_module_1_job0_route0_20240617-101924.tar.gz_pt_aa.zip)
[sc_issue_decode_sc_module_1_job0_route0_20240617-101924.tar.gz_pt_ab.zip](https://github.com/user-attachments/files/15872883/sc_issue_decode_sc_module_1_job0_route0_20240617-101924.tar.gz_pt_ab.zip)

gadfort, 2024-06-17:
@maliberty what do you mean? this is for the single failing task (it's not starting over from scratch)

maliberty, 2024-06-17:
Do you consider fill cells + grt + drt a single task?

gadfort, 2024-06-17:
that is what is run in that task, yes.

maliberty, 2024-06-17:
On what iteration do you see the failure?  This is a large design so it looks like it will take a while to reproduce

TheMichaelGionet, 2024-06-17:
With this particular run it was optimization iteration 1 after it hit the 20% mark. The exact iteration and percent has not been consistent across runs, but hitting a signal 11 has been consistent across runs.

maliberty, 2024-06-17:
My run is already mostly through iteration 2

maliberty, 2024-06-17:
It ran to completion without error for me so I can't reproduce your issue.

TheMichaelGionet, 2024-06-18:
I have a partition on my PC for ubuntu, I'll try compiling openroad for that and let you know how that works. If all works on your end then I'm willing to bet it's a support issue with its dependencies (specifically for Linux Mint, since I did a somewhat hacky thing to get it to work).

TheMichaelGionet, 2024-06-19:
This particular issue seemed to resolve itself when moving to ubuntu.

closed 2024-06-19 by TheMichaelGionet