The-OpenROAD-Project/OpenROAD #5261: Timing Report filter has mysterious escaping behavior

oharboe, 2024-06-18:
### Description

If I want to use a register called something like:

```
foo/bar/abc[44]$_DFFE_PP_/D (DFFHQNx2_ASAP7_75t_R)
```

Then I have to escape it manually before I enter it into "To:", otherwise nothing happens (no error message or indication that the signal does not exist, just zero reaction).

```
{foo/bar/abc\[44\]$_DFFE_PP_/D (DFFHQNx2_ASAP7_75t_R)}
```

Using "*" works as well, but of course then I don't get exactly the signal I want.

### Suggested Solution

Less surprising behavior. If there are some pattern rules, indicate it in the requester, or otherwise add escaping as relevant.

maliberty, 2024-07-02:
Names generally suck and there is no magic bullet.  They differ by format (\a[1] vs a\[1\]) and escaping can be required to disambiguate (a/b vs a\/b are different).  Case sensitivity or not is another one.  Which characters require escaping also varies. 
 Mixing regex or glob syntax in adds further confusion.  There is no way to make it all automagically work, even in proprietary tools.  

The best we can do is document clearly what is happening.   Perhaps a section on names in the docs?

oharboe, 2024-07-02:
The timing filter GUI is not give any "haptic feedback". If you enter some random text string and hit "enter", there is no feedback whatsoever that you hit enter. There is nothing in the GUI to indicate what the user can enter here and whether or not they should hit enter...

I'm not quite sure what to suggest here.

Could a help button be added that links directly to the documentation?