The-OpenROAD-Project/OpenROAD #5276: Powerswitch floorplanning and placement issues

kareefardi, 2024-06-23:
### Describe the bug

I am unable to insert power switches using UPF without causing overlaps. I am using the power swtich abstract view that is present in the test cases of OpenROAD. Here is what I tried. 

Scenario 1:
- Power switch with single height site
- Tapcells overlap power switches
- Standard cells overlaps power switches

Scenario 2:
- Power switch cell with double height site in floorplan
- PDN places overlapping power swtiches
- Tapcells overlaps power switches

To me the logical scenario is scenario 2 as the power switch cell is a double height cell. However it seems like whenever passing `-additional_sites unithddbl` to `initialize_floorplan`, PDN gen insert switches on both rows (the double height row and the single height row)? 

### Expected Behavior

Be able to properly insert power switch cells without overlaps

### Environment

```shell
Git commit: e956dce5da5e738a5c0d9fa0b097b51f4798cd18
kernel: Linux 6.5.0-35-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-14288-ge956dce5d
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
[... 112 lines cut ...]
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0")  
-- Found Eigen3: /usr/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.uvPpVk7bGU
```

### To Reproduce

[powerswitch.zip](https://github.com/user-attachments/files/15943984/powerswitch.zip)

Inside the zip file, `script.tcl` for the first scenario and `script2.tcl` for the second scenario.   

### Screenshots

Scenario 1:
[image]
Scenario 2:
[image]

maliberty, 2024-06-23:
@gadfort any thoughts?

gadfort, 2024-06-24:
@maliberty I think the double height rows were added after this code, so don't think PDN checks (I can fix that).
I'll need to spend some time to figure out why the tapcells are getting ignored by PDN (it should be honoring them, but I'm not sure what happens if no legal site can be found).

gadfort, 2024-07-12:
@maliberty scenario 1 looks to be an issue with detail placement (running `check_placement` to doesn't report an issue despite there being overlapping cells.). I think detail placement should correctly identify the power switches and avoid the overlaps or report an error because it cannot be legalized.

gadfort, 2024-07-12:
@kareefardi One issue going against these issues is the lack of double height rows. I would think a correct setup would have both since you are adding double height cells. Maybe DPL should generate an error if there are mixed site definitions without rows to support it (@maliberty)?

gadfort, 2024-07-12:
@maliberty in the script provided by @kareefardi (for issue 2nd issue) the tap cells are added after the powergrid is inserted (I think this is wrong order of operations) but this would indicate that the issue is actually in the tapcell tool, since PDN cannot know ahead of time where the tapcells will be placed.

kareefardi, 2024-07-14:
@gadfort In the second scenario, I forgot to add `-additional_site unithddbl` to `initialize_floorplan`. If you add it, you will notice the overlaps between power switch cells themselves *with* double height rows included in the floorplan.

Beside PDN and tapcells, overlap occurs between between power switch cells and standard cells.

gadfort, 2024-07-14:
@kareefardi #5378 fixes the overlapping switch instances, but not any of the other issues, those will need to be corrected by others.

maliberty, 2025-02-21:
@eder-matheus why dpl/dpo are fresh in your mind

maliberty, 2025-05-02:
@osamahammad21 please look at the placement issues here when you can

osamahammad21, 2025-05-03:
Are you sure this is a dpl issue?
I ran check_placement in the first scenario and it seems to identify the power switch cells overlap correctly. And dpl doesn't place any cell that overlaps with the powerswitch cells. DPL itself is unable to resolve the powerswitch cells overlap because their Placement status is Firm.

[image]

From running the second scenario, I can't identify anything wrong with dpl as well. it is able to identify tap cell overlap correctly but It is unable to resolve because also the tap cells placement status is LOCKED.

[image]

osamahammad21, 2025-05-03:
I created a [PR](https://github.com/The-OpenROAD-Project-private/OpenROAD/pull/2557) that should fix tapcell ignoring the power switches but still waiting for the secure CI results.

osamahammad21, 2025-05-07:
@maliberty Should this issue be closed now?

maliberty, 2025-05-07:
If you've fixed the user's test case then close it.

closed 2025-05-07 by osamahammad21